labgrid-project/labgrid #529: Watchdog reboot

nandra, 2019-11-14:
HI,

I'm using labgrid in my project to test target device from various aspects. I'm not sure if this question belong to labgrid or to other ;). I plan to test if watchdog reboot my device when stop pinging HW watchdog. I have device connected over serial port (for login username + pass needs to be entered). Does anybody have experience with this how to do it? I can execute atchdog reboot but then how to wait and re-login and check if board was rebooted? Thanks for any pointers. Marek

Emantor, 2019-11-19:
Sorry for the late response, I was sick last week and did not get to it yesterday. You'll probably want a state within your strategy which invalidates the current state, disables all the drivers and lets your strategy start again from scratch. This way you can stop the watchdog ping program (via systemd, command, …), invalidate the current strategy state and than try to activate your bootloader/shell driver again. If this probes successfully, your watchdog correctly triggers a reboot. The next test would than restart from scratch and resetup the board using the strategy.

nandra, 2019-11-20:
@Emantor thanks for reply. I'm bit new to labgrid and just use it to write some small test over serial line. I can stop watchdog with echo 1 > /dev/watchdog and board will be restarted in 30 secs. So my plan was to start shell issue this command and then somehow wait for reboot command and try to login again. But logging is done with yml so I'm bit confused how to achieve this ;). Thanks.

Emantor, 2019-11-20:
the shelldriver will autodetect whether there is a usuable prompt or login prompt already, if you can't stop in the bootloader to verify that your board rebootet because of the watchdog timeout you'll have to extract this information by other means.
Deactivating and reactivating the ShellDriver will try to detect the prompt during the activation.

nandra, 2019-11-22:
I tried to simulate watchdog reboot with this command in test:
```
stdout, stderr, ret = command.run("reboot")
```

but got timeout:
pexpect.exceptions.TIMEOUT: Timeout of 0.88 seconds exceeded or connection closed by peer
 lib/python3.5/site-packages/pexpect/expect.py:82: TIMEOUT

board was rebooted because when checked on serial console there was login prompt. I added also 
```
command.on_deactivate()
command.on_activate()
```

to re-login but this seems doesn't work. I looked through the code and found only reset via gpio. Is there some example using reboot command. Thanks.

Emantor, 2019-11-22:
`command.run` can't be used here since it uses a pattern to detect a return of the prompt, which might not return in the reboot command. Instead use the ConsoleProtocol and its sendline implementation. Deactivation and activation of drivers is done through the target, so the resulting code would look roughly like this:

```
command = target.get("CommandProtocol")
target.deactivate(command)
console = target.get("ConsoleProtocol")
console.sendline('reboot')
# here the system should reboot, activate the ShellDriver again to login
target.activate(command)
```
`command.on_deactivate` and `command.on_activate` are callbacks which are called by the target on driver activation and deactivation and should usually not require user intervention. Hope this helps.

nandra, 2019-11-22:
@Emantor thanks it works like charm (just need to change s/target.get/target.get_driver/ge) but then I get what I want. Maybe would be good to document that on wiki (I can send PR ;) )

Emantor, 2019-11-22:
On the wiki may not be as useful as within the documentation :smiley: I'll gladly take a PR. Closing this because your question is solved.

closed 2019-11-22 by Emantor