labgrid-project/labgrid #542: Docker image failure when running the coordinator and exporter

manne11, 2019-12-11:
I followed the instructions in "labgrid/docker". When I try to run the labgrid-coordinator. I have the following error message:
```

2019-12-11T13:03:01+0000 [Controller      1] 
2019-12-11T13:03:01+0000 [Controller      1]     :::::::::::::::::
2019-12-11T13:03:01+0000 [Controller      1]           :::::          _____                      __
2019-12-11T13:03:01+0000 [Controller      1]     :::::   :   :::::   / ___/____ ___   ___  ___  / /  ___ _ ____
2019-12-11T13:03:01+0000 [Controller      1]     :::::::   :::::::  / /__ / __// _ \ (_-< (_-< / _ \/ _ `// __/
2019-12-11T13:03:01+0000 [Controller      1]     :::::   :   :::::  \___//_/   \___//___//___//_.__/\_,_//_/
2019-12-11T13:03:01+0000 [Controller      1]           :::::
2019-12-11T13:03:01+0000 [Controller      1]     :::::::::::::::::   Crossbar v19.3.5
2019-12-11T13:03:01+0000 [Controller      1] 
2019-12-11T13:03:01+0000 [Controller      1]     Copyright (c) 2013-2019 Crossbar.io Technologies GmbH, licensed under AGPL 3.0.
2019-12-11T13:03:01+0000 [Controller      1] 
2019-12-11T13:03:01+0000 [Controller      1] Initializing <crossbar.node.node.Node> as node [realm=crossbar, cbdir=/opt/crossbar]
2019-12-11T13:03:01+0000 [Controller      1] Node key files exist and are valid. Node public key is 0xcd77b40048ee11304b6017f607e2f4d80731707ac6cabc6ade949fb1a5a77be5
2019-12-11T13:03:01+0000 [Controller      1] Node key loaded from /opt/crossbar/key.priv
2019-12-11T13:03:01+0000 [Controller      1] Node configuration loaded [config_source=localfile, config_path=/opt/labgrid/.crossbar/config.yaml]
2019-12-11T13:03:01+0000 [Controller      1] Entering event reactor ...
2019-12-11T13:03:01+0000 [Controller      1] Starting standalone node <crossbar.node.node.Node.start>
2019-12-11T13:03:01+0000 [Controller      1] Node ID 5c63d418caa8 set from hostname
2019-12-11T13:03:01+0000 [Controller      1] No extra node router roles
2019-12-11T13:03:01+0000 [Controller      1] RouterServiceAgent ready (realm_name="crossbar", on_ready=None)
2019-12-11T13:03:01+0000 [Controller      1] Registered 23 procedures
2019-12-11T13:03:01+0000 [Controller      1] Signal handler installed on process 1 thread 140451520390976
2019-12-11T13:03:01+0000 [Controller      1] Using default node shutdown triggers ['shutdown_on_worker_exit']
2019-12-11T13:03:01+0000 [Controller      1] Booting node <crossbar.node.node.Node.boot>
2019-12-11T13:03:01+0000 [Controller      1] Booting node from local configuration [parallel_worker_start=False] .. <crossbar.node.node.Node.boot_from_config>
2019-12-11T13:03:01+0000 [Controller      1] Will start 2 workers ..
2019-12-11T13:03:01+0000 [Controller      1] Order node to start Router worker001
2019-12-11T13:03:01+0000 [Controller      1] Starting router worker worker001 <crossbar.node.controller.NodeController.start_worker>

2019-12-11T13:03:02+0000 [Router         17] Starting worker "worker001" for node "5c63d418caa8" on realm "crossbar" with personality "standalone" <crossbar.worker.router.RouterController>
2019-12-11T13:03:02+0000 [Router         17] Running as PID 17 on CPython-EPollReactor
2019-12-11T13:03:02+0000 [Router         17] Entering event reactor ...
2019-12-11T13:03:02+0000 [Router         17] Router worker session for "worker001" joined realm "crossbar" on node router <crossbar.worker.router.RouterController.onJoin>
2019-12-11T13:03:02+0000 [Router         17] Registered 41 procedures
2019-12-11T13:03:02+0000 [Router         17] Router worker session for "worker001" ready
2019-12-11T13:03:02+0000 [Controller      1] Ok, node has started Router worker001
2019-12-11T13:03:02+0000 [Controller      1] Configuring Router worker001 ..
2019-12-11T13:03:02+0000 [Controller      1] Order Router worker001 to start Realm realm001
2019-12-11T13:03:02+0000 [Router         17] Starting router realm realm001 <crossbar.worker.router.RouterController.start_router_realm>
2019-12-11T13:03:02+0000 [Router         17] RouterServiceAgent ready (realm_name="realm1", on_ready=<Deferred at 0x7f1c45dcda20>)
2019-12-11T13:03:02+0000 [Router         17] Realm "realm001" (name="realm1") started
2019-12-11T13:03:02+0000 [Controller      1] Ok, Router worker001 has started Realm realm001
2019-12-11T13:03:02+0000 [Controller      1] Order Realm realm001 to start Role role001
2019-12-11T13:03:02+0000 [Router         17] Starting role "role001" on realm "realm001" <crossbar.worker.router.RouterController.start_router_realm_role>
2019-12-11T13:03:02+0000 [Router         17] role role001 on realm realm001 started
2019-12-11T13:03:02+0000 [Controller      1] Ok, Realm realm001 has started Role role001
2019-12-11T13:03:02+0000 [Controller      1] Order Realm realm001 to start Role role002
2019-12-11T13:03:02+0000 [Router         17] Starting role "role002" on realm "realm001" <crossbar.worker.router.RouterController.start_router_realm_role>
2019-12-11T13:03:02+0000 [Router         17] role role002 on realm realm001 started
2019-12-11T13:03:02+0000 [Controller      1] Ok, Realm realm001 has started Role role002
2019-12-11T13:03:02+0000 [Router         17] started component: labgrid.remote.authenticator.AuthenticatorSession id=4591427529726534
2019-12-11T13:03:02+0000 [Controller      1] Router worker001: component 'component001' started
2019-12-11T13:03:02+0000 [Controller      1] Order Router worker001 to start Transport transport001
2019-12-11T13:03:02+0000 [Router         17] Starting router transport "transport001" <crossbar.worker.router.RouterController.start_router_transport>
2019-12-11T13:03:02+0000 [Router         17] Creating router transport for "transport001" <crossbar.worker.transport.create_router_transport>
2019-12-11T13:03:02+0000 [Router         17] Router transport created for "transport001" <crossbar.worker.transport.RouterWebTransport>
2019-12-11T13:03:02+0000 [Router         17] Created "static" Web service on root path "/" of Web transport "transport001"
2019-12-11T13:03:02+0000 [Router         17] Site starting on 20408
2019-12-11T13:03:02+0000 [Controller      1] Ok, Router worker001 has started Transport transport001
2019-12-11T13:03:02+0000 [Controller      1] Order Transport transport001 to start Web Service webservice001
2019-12-11T13:03:02+0000 [Router         17] Starting "websocket" Web service on path "ws" of transport "transport001" <crossbar.worker.router.RouterController.start_web_transport_service>
2019-12-11T13:03:02+0000 [Controller      1] Ok, Transport transport001 has started Web Service webservice001
2019-12-11T13:03:02+0000 [Controller      1] Ok, Router worker001 configured
2019-12-11T13:03:02+0000 [Controller      1] Starting guest worker coordinator <crossbar.node.controller.NodeController.start_worker>
2019-12-11T13:03:02+0000 [Controller      1] Could not start worker: could not find and executable for 'python'
2019-12-11T13:03:02+0000 [Controller      1] Could not start node: Traceback (most recent call last):
--- <exception caught here> ---
[... 23 lines cut ...]
crossbar URL: ws://192.168.1.42:20408/ws
crossbar realm: realm1
exporter name: 3191676d4900
exporter hostname: 3191676d4900
resource config file: /opt/conf/exporter.yaml
Traceback (most recent call last):
  File "/usr/local/bin/labgrid-exporter", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.2.1.dev320+g0a0661d.d20191211-py3.7.egg/labgrid/remote/exporter.py", line 579, in main
    runner.run(ExporterSession, log_level=level)
  File "/usr/local/lib/python3.7/dist-packages/autobahn/asyncio/wamp.py", line 261, in run
    (transport, protocol) = loop.run_until_complete(coro)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 959, in create_connection
    raise exceptions[0]
  File "/usr/lib/python3.7/asyncio/base_events.py", line 946, in create_connection
    await self.sock_connect(sock, address)
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 464, in sock_connect
    return await fut
  File "/usr/lib/python3.7/asyncio/selector_events.py", line 494, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('192.168.1.42', 20408)
```
Any suggestions on how to solve it?

krevsbech, 2019-12-11:
Yes that's an unfortunate error introduced in #525 
I have an open PR #540 that will fix this issue among others. I hope to have it merged soon.
If you need a quick fix consider cherry-picking 140fcab0de225c7106eea3f0d6edd183bc9389d3

manne11, 2019-12-11:
Thanks, the commit solves the problem with the coordinator. Running the exporter still gives the same error though. Hoping the merge would resolve it.

krevsbech, 2019-12-12:
It might. However, if you are following the README in the docker folder bear in mind that the examples are examples where you need to adjust IP's etc. Hence yes docker-compose in staging in the merge request open should work as an example. But you still need to adjust stuff to make it work in you setup.
Are you creating a seperate docker network ?
Please post the docker run commands you use.

manne11, 2019-12-12:
>  if you are following the README in the docker folder bear in mind that the examples are examples where you need to adjust IP's etc

It's my mistake, I just realized I am using the wrong IP's and forgot to provide a ".yaml" when initializing the container for the exporter. After correcting them the exporter works fine. 

> But you still need to adjust stuff to make it work in yoru setup.
> Are you creating a separate docker network ?

I have no intention to create a separate docker network. I'm trying to understand how to use the labgrid setup in containers and had the idea of spawning all of them through "docker compose" but I see that this [commit](https://github.com/labgrid-project/labgrid/pull/540/commits/2eb6df531b9ba085571baea7db89b3b84e606e1b#diff-c3c5d1a0565f78531baa5b8804b21c4f) also attempts something similar. Thank you for the help

closed 2019-12-12 by manne11