labgrid-project/labgrid #543: Cleaning up the getting started page

TheMeaningfulEngineer, 2019-12-14:
Am just going through the docs with fresh eyes and have a few questions/directions to put on the table just to see what the reception might be. Background wise, I have written proprietary solutions to serve as "uart drivers" to be executed via Avocado/pytest.

I will assume the following `goal proposition`: 

> The goal proposition of the getting started document is to attract new contributors/users to lab-grid. It aims to show the most functionality in the simplest possible manner for new users so they are motivated to stay and explore further.

### [Installation](https://labgrid.readthedocs.io/en/latest/getting_started.html#installation)

Reaches the aim reasonably well. Installing labgrid is a matter of copy pasting and the basic example works with minimal changes needed. :+1: 

The notes could perhaps live in some troubleshooting section not to slow down the installation [happy path](https://en.wikipedia.org/wiki/Happy_path).

### [Optional requirements](https://github.com/labgrid-project/labgrid/blob/master/doc/getting_started.rst#optional-requirements)

This feels like it comes out of nowhere. If you agree with the `goal proposition` this doesn't help achieving it. It this part of the docs the new user hasn't been introduced with a concept of `drivers` and is being bombarded with optional requirements for more complex drivers. 

Would suggest moving this to someplace else and focus just on what is needed for the first working example.

### [Running Your First Test](https://github.com/labgrid-project/labgrid/blob/master/doc/getting_started.rst#running-your-first-test)

It's impossible to do a workable example without some forward referencing. Here the concepts of `target`, `driver` and `resource` are forward referenced. This would be a good place for a note pointing to a link to the [terminology](https://github.com/labgrid-project/labgrid/blob/master/doc/overview.rst).

In the final part of running the first test lay the instructions on how to build the docs. This seems quite unrelated an should likely live elsewhere.

 ### [Setting Up the Distributed Infrastructure](https://github.com/labgrid-project/labgrid/blob/master/doc/getting_started.rst#setting-up-the-distributed-infrastructure)

Now it is possible this is a valid use case for a new user to encounter, but what seem to be missing is an answer to **why?**.  I have just successfully ran an simple example and am ready to know more about the concepts of `drivers` and `protocols` etc... or perhaps try some more copy pastable working examples... what does the description of the distributed infrastructure component bring to the table?

Perhaps it would be more clear to have something like:

> The above cli example was just for show. The real power of labgrid is in it's distributed infrastructure which provides what the cli doesn't and that is.....bla bla

---

Hope you don't consider this as an ungrateful rant :) 
I am looking forward to seeing lab-grid thrive because it addresses a lot of painpoints I've been encountering in the last few years. Keep up the good work and am looking forward to my first PR (perhaps even for this issue :) )

Emantor, 2019-12-17:
Thanks for your in depth feedback, I'll gladly accept a pull request which fixes the critiqued points.

TheMeaningfulEngineer, 2019-12-17:
Sure thing, 
I do need some help on the conceptual overview of the distributed architecture, mostly on what's the value gained for the introduced complexity over the cli?

And if it makes sense to try to pursue an copy-pastable example from it in the "Getting started" page?

Emantor, 2020-01-10:
The distributed architecture allows remote access to the boards and is integrated with the CLI. This allows the user to store the boards in a central location and use them from local workstations (think many users in a corporate setting.)

TheMeaningfulEngineer, 2020-01-10:
Got it. So the idea is to outgrow being just the testing framework but be a way to interact with a board farm remotely. Sounds great, thanks.

Bastian-Krause, 2020-01-13:
That's basically the whole idea, labgrid is not a testing framework, but a remote control library containing a pytest plugin ;)

TheMeaningfulEngineer, 2020-02-10:
I've started a PR for this. https://github.com/labgrid-project/labgrid/pull/576
Would like to hear your comments on the general path taken and some help in describing the reasoning on when resources get separated from driver properties.

The assumption I have at the moment is:

> Resources are split from Drivers to accommodate a technical issue for a more complex use case. By doing this, the complexity spills down to the core concepts where it's hard to see the motivation behind it. This makes the core concepts difficult to explain.

I how you take the above quote openheartedly :) 
It's phrased in that manner because that seems to be the clearest definition of the problem I'm facing at the moment. I would appreciate if you challenge it, and am looking forward to having my assumption proven wrong.

Emantor, 2024-03-20:
Docs have been overhauled since this was opened, closing.