labgrid-project/labgrid #545: Detecting prompt timeout

nandra, 2019-12-19:
Hi,

I have orangepi neo plus2 and I have issue with login to board. I setup shell driver as in example but when board is rebooted as suggested here: https://github.com/labgrid-project/labgrid/issues/529#issuecomment-557468415
I cannot login anymore and I got timeout like:
```
self = <pexpect.expect.Expecter object at 0x7fb5f7ca5518>, err = TIMEOUT('Timeout of 1.99 seconds exceeded or connection closed by peer',)

    def timeout(self, err=None):
        spawn = self.spawn
    
        spawn.before = spawn.buffer
        spawn.after = TIMEOUT
        index = self.searcher.timeout_index
        if index >= 0:
            spawn.match = TIMEOUT
            spawn.match_index = index
            return index
        else:
            spawn.match = None
            spawn.match_index = None
            msg = str(spawn)
            msg += '\nsearcher: %s' % self.searcher
            if err is not None:
                msg = str(err) + '\n' + msg
>           raise TIMEOUT(msg)
E           pexpect.exceptions.TIMEOUT: Timeout of 1.99 seconds exceeded or connection closed by peer
E           <labgrid.util.expect.PtxExpect object at 0x7fb5f759f320>
E           command: None
E           args: None
E           buffer (last 100 chars): b'nsor login: run() { echo -n "$MARKER"; sh -c "$@"; echo "$MARKER $?"; }\r\n \'EAWA\'\'VSEXIB\'\r\nPassword: '
E           before (last 100 chars): b'nsor login: run() { echo -n "$MARKER"; sh -c "$@"; echo "$MARKER $?"; }\r\n \'EAWA\'\'VSEXIB\'\r\nPassword: '
E           after: <class 'pexpect.exceptions.TIMEOUT'>
E           match: None
E           match_index: None
E           exitstatus: None
E           pid: None
E           child_fd: -1
E           closed: True
E           timeout: 30
E           delimiter: <class 'pexpect.exceptions.EOF'>
E           logfile: None
E           logfile_read: None
E           logfile_send: None
E           maxread: 1
E           ignorecase: False
E           searchwindowsize: None
E           delaybeforesend: 0.05
E           delayafterclose: 0.1
E           delayafterterminate: 0.1
E           searcher: searcher_re:
E               0: re.compile(b'EAWAVSEXIB')

../labgrid/labgrid-venv/lib/python3.5/site-packages/pexpect/expect.py:82: TIMEOUT

```

when tested on other project for rpi3 same code works perfectly fine. Any ideas?
Could be that login is disturbed somehow (I also hit this issue and from log it seems it stops in u-boot) and I just workarounded it with time.sleep(30). Thanks.

Bastian-Krause, 2019-12-19:
> buffer (last 100 chars): b'nsor login: run() { echo -n "$MARKER"; sh -c "$@"; echo "$MARKER $?"; }\r\n \'EAWA\'\'VSEXIB\'\r\nPassword: '

The ShellDriver finished its [_await_login()](https://github.com/labgrid-project/labgrid/blob/3c4de08893502da1b4976a5b14d42bbd64623922/labgrid/driver/shelldriver.py#L109) and already [injects the run function](https://github.com/labgrid-project/labgrid/blob/3c4de08893502da1b4976a5b14d42bbd64623922/labgrid/driver/shelldriver.py#L61). This is wrong as the login prompt is still active. Please share your config and the complete logs.

nandra, 2019-12-19:
Log can be found here: https://pastebin.com/LTrQyStT

and config is here:
```
targets:
  main:
    resources:
      RawSerialPort:
        port: "/dev/ttyUSB0"
    drivers:
      ManualPowerDriver:
        name: "example"
      SerialDriver: {}
      ShellDriver:
        prompt: '#'
        login_prompt: 'figured-sensor login:'
        login_timeout: 75
        username: 'root'
        password: '1234'
```

Bastian-Krause, 2019-12-19:
Could you run labgrid-pytest with --lg-log -s -vv and share the logs? I'd like to see the first login attempt.

nandra, 2019-12-19:
log:

```
=== Log starts here ===
NOTICE:  BL31: v2.0(release):t0.0.11
NOTICE:  BL31: Built : 11:40:16, Oct 31 2019
NOTICE:  BL31: Detected Allwinner H5 SoC (1718)
NOTICE:  BL31: STUB PMIC setup code called

U-Boot 2018.09 (Oct 31 2019 - 11:40:36 +0100) Allwinner Technology
          
DRAM:  1 GiB
MMC:   SUNXI SD/MMC: 0, SUNXI SD/MMC: 1
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
USB2:   USB EHCI 1.00
USB3:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 2 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
447 bytes read in 1 ms (436.5 KiB/s)
## Executing script at 4fc00000
19374592 bytes read in 852 ms (21.7 MiB/s)
```

and exception: https://pastebin.com/kj5rgNCk

To me it looks like it's not waiting for prompt and read somethign from u-boot and stop.

I just start board and test. Also seems login_timeout is not taken into consideration as I changed it to 75 secs but exception happened before timeout. Thanks.

Emantor, 2019-12-19:
Your prompt is matched during the startup procedure, as it is set to a single `#`.  I suggest to reconfigure your prompt to something more unique and longer than a single character, which can be matched anywhere during the bootup procedure. This is unfortunately a limitation using the regex matching and requires changes in the dut configuration.

nandra, 2019-12-19:
@Emantor ok thanks a lot. Never think of such thing, you saved my day :).

closed 2019-12-19 by nandra