labgrid-project/labgrid #546: Fails to login when login prompt overlayed with initscript messages

nandra, 2019-12-20:
Hi,

I have situation where login prompt is there but labgrid report login timeout and with debug I see that login prompt cannot be parsed directly because other startup scripts run in background like:

```
unconfigured-sensor login: Found WiFi ID0 = 0
Found WiFi ID1 = 3
Sleep 10..
Sleep done..
Device setup complete
run() { echo -n "$MARKER"; sh -c "$@"; echo "$MARKER $?"; }
Password: hci0: Type: Primary  Bus: UART
        BD Address: AC:83:F3:48:FD:5D  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:1224 acl:0 sco:0 events:60 errors:0
        TX bytes:796 acl:0 sco:0 commands:60 errors:0
```

I think sending Enter should resolve an issue (at least resolve when attached on serial directly). Would be such change valid or there exists way how to workaround this? Thanks.

jluebbe, 2019-12-20:
labgrid tries to similar issues with kernel output by disabling kernel log to the console, but in general you need to have an undisturbed and reliable console.
I your case it looks like the device wants a password, but labgrid thinks that it already got a prompt (as it's sending the run helper function). Maybe your prompt pattern is not specific enough?

Bastian-Krause, 2019-12-20:
This looks a lot like #545. Could you share your logs (`labgrid-pytest -s -vv`) and config again?

nandra, 2019-12-23:
@Bastian-Krause #545 is resolved by changing shell prompt to something unique and not just #
@jluebbe those messages are not from kernel but from from init scripts

jluebbe, 2020-01-10:
Just sending a newline wouldn't be reliable as there is no way to know when background processes stop interfering with the console. The could mix their output with any output from later commands, so this won't be reliable.

The proper solution is to start your `getty` only after the system has started completely and/or disconnecting background processes from the console. This also avoids confusing human user. ;)

As I see it now, I wouldn't like to add an unreliable workaround for this problem, sorry.

nandra, 2020-01-14:
Looks like target.deactivate(console) helps in this case.

closed 2020-01-14 by nandra