labgrid-project/labgrid #559: Cannot login to board

nandra, 2020-01-14:
I'm using labgrid to test presence of accelerometer on my board.
Test is simple just check if path exists and then reboot board.

my yaml looks like:
```
targets:
  main:
    resources:
      RawSerialPort:
        port: "/dev/ttyUSB0"
    drivers:
      ManualPowerDriver:
        name: "example"
      SerialDriver: {}
      ShellDriver:
        prompt: 'root@MYCAM-1234:~#'
        login_prompt: ' login: '
        username: 'root'
        password: ''
```

and debug log shows:
```
Labgrid Console Logfile for main
Logfile started at 2020-01-14 10:13:37.078249
=== Log starts here ===

Poky (Yocto Project Reference Distro) 2.4.4 MYCAM-1234 ttyAMA0

MYCAM-1234 login:

Poky (Yocto Project Reference Distro) 2.4.4 MYCAM-1234 ttyAMA0

MYCAM-1234 login:

```

also trace is pushed to here: https://pastebin.com/6Ur1tdfd

Any ideas why it cannot login to board? I suspect again some regexp issue but cannot spotted it by myself. Thanks.

Emantor, 2020-01-14:
Pasing this into [regexr](regexr.com/4s5v8), I suspect that you want to lose the space after `login:`.

nandra, 2020-01-14:
@Emantor edited

```
login_prompt: ' login: '
```

to :
```
login_prompt: ' login:'
```

but issue is the same.

Bastian-Krause, 2020-01-14:
The trailing space is not the problem, it is actually there, see the pastebin logs.
The problem seems to be that the SerialDriver cannot write the password. Are you able to login manually? Try to raise the log level to debug. This makes the ConsoleExpectMixin log writes.

Emantor, 2020-01-14:
Ah, I looked at the trace, the shelldriver does not see the initial login prompt and instead presses enter at this prompt. Since the username is not set the prompt is redisplayed and the password is never matched.
You'll want to activate the `SerialDriver` before doing the power cycle, this way there should not be any los characters.

nandra, 2020-01-14:
Username is set in yaml file as root. And I can login with root and empty password (well I used passwd and instead of new password type Enter only). Maybe password is wrongly set in yaml file then? When use "normal" password it works fine. Thanks.

Bastian-Krause, 2020-01-14:
> Username is set in yaml file as root.

Yes, and I can't see anything wrong with that.

> And I can login with root and empty password (well I used passwd and instead of new password type Enter only). Maybe password is wrongly set in yaml file then? When use "normal" password it works fine.

You don't even reach the password prompt, so this is not the problem. The problem is that we can;t see the login attempt, the username is apparently not sent to the login prompt. As said above, try raising the log level to debug (`-vv`) and provide us the complete output of the pytest call.

Emantor, 2020-01-27:
@nandra is this still an issue?

nandra, 2020-01-27:
@Emantor nope resolved. Thanks.

closed 2020-01-27 by nandra