labgrid-project/labgrid #569: NetworkUSBStorageDriver: ManagedFile breaks bmaptool

Bastian-Krause, 2020-01-31:
[man bmaptool](https://github.com/intel/bmap-tools/blob/master/docs/man1/bmaptool.1) reads:
> Unless the bmap file is explicitly specified with the "--bmap" option, bmaptool automatically discovers it by looking for a file with the same basename as IMAGE but with the ".bmap" extension. The bmap file is only looked for in IMAGE's directory (or base URL, in case IMAGE was specified as an URL). If the bmap file is not found, bmaptool fails.

If the file given to NetworkUSBStorageDriver.write_image() is not available on the remote side the source file will be copied, but not the bmap file. This results in bmaptool failing.

Emantor, 2020-01-31:
Thats unfortunate. We should check for a similarly named bmap file, however we can't simply transfer the bmap file using the ManagedFile since it will be uploaded to a different location.

Bastian-Krause, 2020-01-31:
Correct, we'll should specify the bmap file explicitly with `--bmap`.

Emantor, 2021-01-22:
Fixed by 42ffd066cbd730e59460b151208ebf5265606335.

closed 2021-01-22 by Emantor