labgrid-project/labgrid #582: multiple USBSerialPort for one board

egrigore, 2020-02-26:
Is there any way to set multiple USBSerialPort for one board? 

Our use case: we have boards that are exposing multiple consoles on the same uart port and we would like to be able to access all the consoles in the same time.

How the UART port shows when connected to desktop (on which it is the exporter):

> [ 2314.080100] usb 3-3: new full-speed USB device number 8 using xhci_hcd
[ 2314.209545] usb 3-3: New USB device found, idVendor=xxxx, idProduct=yyyy
[ 2314.209552] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2314.209556] usb 3-3: Product: Dual RS232
[ 2314.209559] usb 3-3: Manufacturer: FTDI
[ 2314.210107] ftdi_sio 3-3:1.0: FTDI USB Serial Device converter detected
[ 2314.210149] usb 3-3: Detected FT2232C
[ 2314.210382] usb 3-3: FTDI USB Serial Device converter now attached to ttyUSB0
[ 2314.210754] ftdi_sio 3-3:1.1: FTDI USB Serial Device converter detected
[ 2314.210813] usb 3-3: Detected FT2232C
[ 2314.211048] usb 3-3: FTDI USB Serial Device converter now attached to ttyUSB1

How it is configured in the exporter now:

> myplacename:
>     location: myplacename
>     USBSerialPort:
>         match:
>             'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.1'

and now 
labgrid-client -p myplacename console
access the /dev/ttyUSB1 console. How do I configure the exporter to access also the /dev/ttyUSB0 one.

I have seen #487 but I wasn't able to implement it.

egrigore, 2020-02-26:
As far as I understand from #487 you create a different place for each uart interface. Is there any other way? Like adding a parameter to console ? labgrid-client -p <place_name> console [<number of console>]

egrigore, 2020-02-26:
I managed to access both consoles like this:

exporter yaml:

```
DESK--MX8MM_ddr4-evk--1-console-main:
  location: "DESK--MX8MM_ddr4-evk--1"
  USBSerialPort:
    match:
        'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.1'

DESK--MX8MM_ddr4-evk--1-console-sec:
  location: "DESK--MX8MM_ddr4-evk--1"
  USBSerialPort:
    match:
        'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.0'

DESK--MX8MM_ddr4-evk--1:
  location: "DESK--MX8MM_ddr4-evk--1"
  UniversalUpdateUtility:
    match:
        'DEVPATH': '/devices/pci0000:00/0000:00:14.0/usb3/3-2'
  NetworkPowerPort:
    model: 'simplehttp'
    host: '<exporter-ip>:8081/3-2'
    index: 1
  NetworkService:
    address: '<exporter-ip>'
    username: 'testlabgrid'
	
```

creating the place and matching resources:

```
labgrid-client -p DESK--MX8MM_ddr4-evk--1 create
labgrid-client -p DESK--MX8MM_ddr4-evk--1 add-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/*
labgrid-client -p DESK--MX8MM_ddr4-evk--1 add-named-match <exporter-host>/DESK--MX8MM_ddr4-evk--1-console-main/NetworkSerialPort main
labgrid-client -p DESK--MX8MM_ddr4-evk--1 add-named-match <exporter-host>/DESK--MX8MM_ddr4-evk--1-console-sec/NetworkSerialPort sec
```

in the end i can access both consoles like this:
```
labgrid-client -p DESK--MX8MM_ddr4-evk--1 console main
labgrid-client -p DESK--MX8MM_ddr4-evk--1 console sec
```

It would help if someone can confirm that this is the right way of doing this, please

Emantor, 2020-02-26:
> It would help if someone can confirm that this is the right way of doing this, please

It is, sorry I couldn't answer earlier, I am at Embedded World at the moment.

Bastian-Krause, 2020-02-26:
The exporter.yaml can be simplified:
```
DESK--MX8MM_ddr4-evk--1:
  location: "DESK--MX8MM_ddr4-evk--1"
  UniversalUpdateUtility:
    match:
      'DEVPATH': '/devices/pci0000:00/0000:00:14.0/usb3/3-2'
  NetworkPowerPort:
    model: 'simplehttp'
    host: '<exporter-ip>:8081/3-2'
    index: 1
  NetworkService:
    address: '<exporter-ip>'
    username: 'testlabgrid'
  USBSerialPort1:
    cls: USBSerialPort
    match:
      'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.1'
  USBSerialPort2:
    cls: USBSerialPort
    match:
      'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.0'
```
Then:
```
export LG_PLACE=DESK--MX8MM_ddr4-evk--1
labgrid-client create
labgrid-client add-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/UniversalUpdateUtility
labgrid-client add-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/NetworkPowerPort
labgrid-client add-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/NetworkService
labgrid-client add-named-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/USBSerialPort/USBSerialPort1 main
labgrid-client add-named-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/USBSerialPort/USBSerialPort2 sec
```
I hope I did not mix up anything.

egrigore, 2020-02-27:
I have tried this way: 
```
DESK--MX8MM_ddr4-evk--1-M4:
  USBSerialPort:
    match:
        'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.0'
DESK--MX8MM_ddr4-evk--1:
  location: "DESK--MX8MM_ddr4-evk--1"
  USBSerialPort:
    match:
        'ID_PATH': 'pci-0000:00:14.0-usb-0:3:1.1'
  UniversalUpdateUtility:
    match:
        'DEVPATH': '/devices/pci0000:00/0000:00:14.0/usb3/3-2'
  NetworkPowerPort:
    model: 'simplehttp'
    host: '<exporter-ip>:8081/3-2'
    index: 1
  NetworkService:
    address: '<exporter-ip>'
    username: 'testlabgrid'
```
and then created the place like this:
```
labgrid-client -p DESK--MX8MM_ddr4-evk--1 create
labgrid-client -p DESK--MX8MM_ddr4-evk--1 add-match <exporter-host>/DESK--MX8MM_ddr4-evk--1/*
labgrid-client -p DESK--MX8MM_ddr4-evk--1 add-named-match <exporter-host>/DESK--MX8MM_ddr4-evk--1-M4/NetworkSerialPort M4
```
I was expecting to be able to access the main console simple without parameter - but that doesn't work:

```
(venv) testlabgrid@<exporter-host>:~/labgrid-tests$ labgrid-client -p DESK--MX8MM_ddr4-evk--1 console
labgrid-client: error: multiple resources matching <class 'labgrid.resource.serialport.NetworkSerialPort'> found in target Target(name='DESK--MX8MM_ddr4-evk--1', env=None)
This may be caused by disconnected exporter or wrong match entries.
You can use the 'show' command to review all matching resources.
```

the M4 one works:
```
(venv) testlabgrid@fsr-ub1664-152:~/labgrid-tests$ labgrid-client -p DESK--MX8MM_ddr4-evk--1 console M4
connecting to NetworkSerialPort(target=Target(name='DESK--MX8MM_ddr4-evk--1', env=None), name='M4', state=<BindingState.bound: 1>, avail=True, host='<exporter-host>', port=46592, speed=115200, protocol='rfc2217') calling microcom -s 115200 -t <exporter-host>:46592
connected to <exporter-ip> (port 46592)
```

@Emantor  is there a way to set multiple consoles with one as the default one that can be accessed with labgrid-client -p <place> console - without parameters ? - i was expecting to be able to do that from the above example.

Bastian-Krause, 2020-02-27:
No, there is no such thing as a "default" resource. If you have multiple resources of the same class you need to specify all of them by name.

egrigore, 2020-02-27:
Ok, can't that be added? We will try to see how to handle that internally , since some of the boards will have multiple consoles and some not - and the code will need to be changed to address both cases.

egrigore, 2020-02-28:
Hi, it seems that the functionality is implemented and can be improved in target.py -> get_resource.

Anyway, for now we just call that method from client.py with USBSerialPort in case no name was given:
```
diff --git a/labgrid/remote/client.py b/labgrid/remote/client.py
index 25ff877..163cde6 100755
--- a/labgrid/remote/client.py
+++ b/labgrid/remote/client.py
@@ -779,6 +779,8 @@ class ClientSession(ApplicationSession):
     async def _console(self, place, target):
         name = self.args.name
         from ..resource import NetworkSerialPort
+        if not name:
+            name =  "USBSerialPort"
         resource = target.get_resource(NetworkSerialPort, name=name)
         host, port = proxymanager.get_host_and_port(resource)

```

Maybe it's not the best solution, but it helps for now. If you have a different suggestion let me know

atline, 2022-12-01:
> No, there is no such thing as a "default" resource. If you have multiple resources of the same class in one place you need to address all of them by name.

So, is it now possible to define default like next just as the OP mentioned? I really feel that useful!

A core:  labgrid-client -p xxx console
M core: labgrid-client -p xxx console M4

atline, 2023-03-06:
> So, is it now possible to define default like next just as the OP mentioned? I really feel that useful!
>
> A core: labgrid-client -p xxx console M core: labgrid-client -p xxx console M4

We really feel this feature useful to give same experience to user, so submit a PR: https://github.com/labgrid-project/labgrid/pull/1098

PR #1112 (2023-03-29): target: make get_resource() prioritize resources named "default" on ambiguous results
**Description**
If a resource named "default" is found and either none or too many resources passed the name check, return the resource named "default".

This is a slightly simpler approach than #1026 (the other part is covered by #1111) and #1098.

**Checklist**
- [x] Documentation for the feature
- [x] Tests for the feature 
- [x] PR has been tested
- [x] Man pages have been regenerated

Closes #1098
Closes #582
closed 2023-03-29 by Emantor