labgrid-project/labgrid #584: getaddrinfo: Name or service not known

atline, 2020-03-05:
For some reasons, some pc in internal network could not ping my hostname, just could ping by ip, so when use console it will show next:

```
(crossbar_venv) root@pie:~# labgrid-client -p example-place console
connecting to NetworkSerialPort(target=Target(name='example-place', env=None), name='RawSerialPort', state=<BindingState.bound: 1>, avail=True, host='NXL60705', port=54595, speed=115200, protocol='rfc2217') calling microcom -s 115200 -t nxl60705:54595
getaddrinfo: Name or service not known
connection lost

```

The error is same with using microcom:

    root@pie:~# microcom -s 115200 -t nxl60705:54595
    getaddrinfo: Name or service not known

But use ip, its ok:

```
(crossbar_venv) root@pie:~# microcom -s 115200 -t 192.168.1.101:54595
connected to 192.168.1.101 (port 54595)
Escape character: Ctrl-\
Type the escape character followed by c to get to the menu or q to quit
```

My question: if possible we set something in any configure, then we could let labgrid console use ip to open serial? 
(PS: I don't want to add map to /etc/hosts, too born to do so many maps, thanks)

closed 2020-03-08 by atline

initBasti, 2021-07-11:
@jluebbe @atline 
Hey,

for some reason, I have the same problem, that the Name or service is not known. This is especially strange because two days ago I was able to enter the serial console just fine with the following export configuration:
```
example:
  RawSerialPort:
      port: "/dev/ttyUSB0"
```

My workflow looks like this:
- Open two terminals on the server device, run `crossbar start &` on the first and `labgrid-exporter exporter_conf.yaml` on the second.
- On my laptop I run the following chain of commands:
```
(labgrid-venv) basti:~/$ labgrid-client -x ws://192.168.178.70:20408/ws -p rpi_example create
(labgrid-venv) basti:~/$ labgrid-client -x ws://192.168.178.70:20408/ws resources
nanopct4/example/NetworkSerialPort
(labgrid-venv) basti:~/$ labgrid-client -x ws://192.168.178.70:20408/ws -p rpi_example add-match nanopct4/example/NetworkSerialPort
(labgrid-venv) basti:~/$ labgrid-client -x ws://192.168.178.70:20408/ws -p rpi_example acquire
acquired place rpi_example
(labgrid-venv) basti:~/$ labgrid-client -x ws://192.168.178.70:20408/ws -p rpi_example con
connecting to NetworkSerialPort(target=Target(name='rpi_example', env=None), name='RawSerialPort', state=<BindingState.bound: 1>, avail=True, host='nanopct4', port=56251, speed=115200, protocol='rfc2217') calling microcom -s 115200 -t nanopct4:56251
getaddrinfo: Name or service not known
connection lost
```

When sending an interrupt signal to the running exporter I get a warning about a pending job here is the full log:
https://paste.debian.net/1204072/

Is there something obvious I can do to access the device via serial console? Accessing it via ssh works without problems, but this stopped working for some reason and I am not aware of major changes performed by myself.

Greetings,
Sebastian

Emantor, 2021-07-11:
Your dns configuration is broken, labgrid tries to run `microcom -s 115200 -t nanopct4:56251` but nanopct4 does not resolve on your network.