labgrid-project/labgrid #585: Regular Labgrid Releases

jgoetzinger, 2020-03-12:
It is currently quite hard to track changes and decide when to update to a newer Labrid version. In addition it seems like there is no simple way to detect which Labgrid version is installed on a machine. Regular releases would make it easier to track versions on different machines and keep them consistent. v0.2.0 is from 4 Jan 2019

ejoerns, 2020-03-12:
As there is no reasonable API guaranty that one could give at the moment, I would agree that having releases is quite useful for tracking changes, but, it would only make sense on a non-semantic versioning scheme.
I'd vote for time-based releases (e.g. v2020.03.0), such as we have for barebox or ptxdist. 

This also slightly touches #166

Emantor, 2020-10-26:
We should track this in #166.

closed 2020-10-26 by Emantor