labgrid-project/labgrid #590: About LAVA/Labgrid bridge.

atline, 2020-03-25:
Hi, labgrid guys,

I'd saw next from: https://elinux.org/Automated_Testing_Summit_2019

> build a prototype to move boards between LAVA and labgrid (Jan)

Half one year passed, so if any detail we could have a reference?

**Let me explain my problem here:**

We have several sites use LAVA, while several sites use LABGRID for different test scope, different purpose. But we want to share devices. There are 2 situations here:

**1. LABGRID users want to use the devices which owned by LAVA users:**

Then, labgrid could send request to LAVA to change the status of device to maintenance state or something like a fake request to make LAVA think the device is busy, then LAVA will not dispatch new request to this device. Thus, labgrid could safely use this device.

**2. LAVA users want to use the devices which owned by LABGRID users:**

Then, LAVA will have to use `labgrid lock` to get the serial port of that labgrid remote device. Then, will have issue which I mentioned in https://git.lavasoftware.org/lava/lava/issues/385, I submit a PR in https://git.lavasoftware.org/lava/lava/-/merge_requests/1045, but not sure if it's the only correct way.

So, I want to know the main idea of your prototype to move boards between LAVA and labgrid to be align with community to avoid duplicate effort. Could you kindly suggest?

jluebbe, 2020-03-25:
Sadly I didn't find time to work on implementing this. :/

The idea at ATS 2019 was to have a separate component which handles the locking. The benefits would be:

- avoiding having a test job waiting deep in the other framework
- existing LAVA or labgrid client don't need any changes
- just one implementation per framework (instead of n*n)

The new daemon would talk to the labgrid coordinator (probably via the reservations API) and to the LAVA scheduler (to the job queue and board status). So it knows which boards are in which state and which jobs are waiting. When it detects that a reservation/job is pending on one side, it would start to "disable" it on the other (by reserving it for itself in labgrid, or using maintenance mode in lava). Then it can enable it on the side where the reservation/job is waiting.

This would mean that the actual access in labgrid and LAVA would still be stand alone, as the locking would make sure that there is no conflict.

Adding support for further frameworks (like fuego or tbot) would just need changes in this daemon, and not for each "peer" in every framework.

atline, 2020-03-25:
Sounds great. If I not misunderstand, this separate component is something like a monitor.

Then, if one use labgrid to lock a device, then the monitor will query the status from labgrid and set the lava device into maintenance state. If one send request to lava, this monitor will query from lava and lock the device automatically with labgrid lock.

Just one question, you mentioned no changes for lava & labgrid, then the work mode of this separate component surely be working in a polling way, in another word, not works in a event-driven way. So how you make sure during the polling gap there is no conflict? Don't know if I express it clear, I mean something like next example:

**Time second 1**: labgrid submit a device lock

**Time second 3**: the separate component polling the status of labgrid, found this device used by labgrid, so set this device in LAVA as maintenance. (Yes, no job will be dispatched to this device in LAVA)

But, how about **Time second 2**? If at this time point, LAVA receive a task, what will happen? There surely a gap in polling mode!

Additional question is about `If one send request to lava, this monitor will query from lava and lock the device automatically with labgrid lock.` (If I not misunderstand it):

The separate component lock the device for lava, but lava no need to do any changes, then what will happen when lava use `{% set connection_commands = {'uart0': 'labgrid-client -p + console'} %` to get the console of the device. You know "the lock in separate component" & "lava use connection_commands to open serial", the two logic are totally asynchronous, we can't suppose one is surely before another.

atline, 2020-03-26:
@jluebbe I update the issue in [lava site](https://git.lavasoftware.org/lava/lava/-/issues/385#note_14862), wish you could finally give us a solution which no need `n*n` patch. And could you leave this question open so we could track it, thanks.

atline, 2022-01-25:
Close this topic as we internal have a similar solution now, thanks for the inspire.

closed 2022-01-25 by atline