labgrid-project/labgrid #593: Best practice to deploy labgrid compoments?

atline, 2020-03-28:
1. Coordinator(Machine A), Exporter(Machine B), Machine which linked the device(Machine C)
2. Coordinator(Machine A), Exporter(Machine A), Machine which linked the device(Machine C)
3. Coordinator(Machine A), Exporter(Machine C), Machine which linked the device(Machine C)

I wonder for about three physical topological, what's the suggested way?

In fact I ask this question because I'm worrying if use E.g. Option 2, after lock the device with the help of coordinator, every time when we use `labgrid-client console`, the traffic will first go to `Machine A`, then `Machine C`, the path is too long which may be result in slow?

But if use `Option 3`, the network traffic will just go to `Machine C`? 

Is my doubt correct? What's the suggest topological for exporter?  Please advise, thanks!

Emantor, 2020-03-28:
>     1. Coordinator(Machine A), Exporter(Machine B), Machine which linked the device(Machine C)
>     2. Coordinator(Machine A), Exporter(Machine A), Machine which linked the device(Machine C)

These two are not supported. The exporter marks device availability based on udev and therefore needs to be connected to the device.

>     3. Coordinator(Machine A), Exporter(Machine C), Machine which linked the device(Machine C)

Here the exporter is directly connected to the target and can mark the resources based on the devices discovered via udev.

> I wonder for about three physical topological, what's the suggested way?
>
> In fact I ask this question because I'm worrying if use E.g. Option 2, after lock the device with the help of coordinator, [...]

Let me give another example:
Client Machine: Machine which runs the tests or labgrid-client
Coordinator Machine: Runs the coordinator
Exporter Machine: Runs the exporter

Place reservation, locking and resource status is reported by "lightweight" calls to and from client machine to the coordinator machine. As soon as data needs to be transferred to the device, the client machine will connect directly to the exporter machine to upload files, communicate via serial, etc. 
> But if use `Option 3`, the network traffic will just go to `Machine C`?

Yes, clients will only fetch the resource information from the coordinator, everything which communicates with the target (image upload, serial console, fastboot,…) is done directly from client to exporter.

atline, 2020-03-29:
Thanks @Emantor for detail explanation, some question: 

**(I)**
> Q1: But if use Option 3, the network traffic will just go to Machine C?

> A1: Yes, clients will only fetch the resource information from the coordinator, everything which communicates with the target (image upload, serial console, fastboot,…) is done directly from client to exporter.

**YES, this answers my question.**

**(II)**
> Q2: 1. Coordinator(Machine A), Exporter(Machine B), Machine which linked the device(Machine C)<br/>
>      2. Coordinator(Machine A), Exporter(Machine A), Machine which linked the device(Machine C)

> A2: These two are not supported. The exporter marks device availability based on udev and therefore needs to be connected to the device.

**Not sure if udev is the only way for exporter to make device availability? What I tried is next:**

1. Coordinator(Machine A)

2. Exporter(Machine A) with `labgrid-exporter remote.yaml`:
    **remote.yaml:**
    ```
    example-group:
      location: example-location
      NetworkSerialPort:
        host: "192.168.1.5"
        port: 7001
    ```

3. Machine which linked the device(Machine C), the IP is `192.168.1.5`:
    **ser2net.conf**
    ```
    7001:telnet:0:/dev/ttyS3:115200 8DATABITS NONE 1STOPBIT local banner
    ```

With above I could successfully visit the device console using `labgrid-client -p example-place acquire` & `labgrid-client -p example-place console` after I add the resource to example-place.

So from your answer of (I), I guess in situation (II), the `labgrid-client` will not parse the "192.168.1.5:7001" from exporter & directly communicate to the machine C? Still the exporter will be responsible for connect to Machine C?  So this is not suggested by best practice?

Emantor, 2020-03-29:
`NetworkSerialPorts` are a special case for devices which export these plainly, like the moxa serial switches. In this case you are correct, the client will connect directly to machine C. Note that this won't work for most other Resource types, which expect the connection to be done via the USB bus. OTOH
`NetworkSerialPorts` and `PDUDaemonPort` don't require a connection to the exporter and will instead connect directly to the host.

atline, 2020-03-30:
> `NetworkSerialPorts` are a special case for devices which export these plainly, like the moxa serial switches. In this case you are correct, the client will connect directly to machine C. Note that [...]

That's very helpful, thanks!

closed 2020-03-30 by atline