labgrid-project/labgrid #597: stdout and stderr not available in the exceptions from ssh driver

TheMeaningfulEngineer, 2020-04-08:
If the ssh subprocess throws an error for whatever the reason (the example will show a permission error), the text of the error isn't accessible in the `ExecutionError` that gets raised.

```
from labgrid import Target
from labgrid.resource import NetworkService
from labgrid.driver import SSHDriver

lg_target = Target("Device")
lg_network_service = NetworkService(lg_target,
                                    name='network_service',
                                    address='192.168.3.11',
                                    username='root')

lg_ssh_driver = SSHDriver(lg_target, name=None)
try:
    lg_target.activate(lg_ssh_driver)
except Exception as e:
    print(e.msg)
    print(e.stdout)
    print(e.stderr)
```

**Current behavior**

```
root@192.168.3.11: Permission denied (publickey).
failed to connect to 192.168.3.11 with ['ssh', '-f', '-o', 'LogLevel=ERROR', '-o', 'PasswordAuthentication=no', '-x', '-o', 'ConnectTimeout=30', '-o', 'ControlPersist=300', '-o', 'UserKnownHostsFile=/dev/null', '-o', 'StrictHostKeyChecking=no', '-o', 'ServerAliveInterval=15', '-MN', '-S', '/tmp/labgrid-ssh-tmp-njzanbrm/control-192.168.3.11', '-p', '22', 'root@192.168.3.11'] and 255
None
None
```

**Expected behavior**

```
failed to connect to 192.168.3.11 with ['ssh', '-f', '-o', 'LogLevel=ERROR', '-o', 'PasswordAuthentication=no', '-x', '-o', 'ConnectTimeout=30', '-o', 'ControlPersist=300', '-o', 'UserKnownHostsFile=/dev/null', '-o', 'StrictHostKeyChecking=no', '-o', 'ServerAliveInterval=15', '-MN', '-S', '/tmp/labgrid-ssh-tmp-njzanbrm/control-192.168.3.11', '-p', '22', 'root@192.168.3.11'] and 255
None
root@192.168.3.11: Permission denied (publickey).
```

TheMeaningfulEngineer, 2020-04-08:
The reason for this is likely the way how the `stdout` and `stderr` are piped.
Here is a filtered out example showing the gist of the way [this happens in the driver](https://github.com/labgrid-project/labgrid/blob/master/labgrid/driver/sshdriver.py#L75):

```
# labgrid_issue.py
import subprocess
import shlex

# An ssh command that will fail
cmd = "ssh root@12345"

process = subprocess.Popen(shlex.split(cmd))
```

```
$ python labgrid_issue.py   
ssh: connect to host 0.0.48.57 port 22: Invalid argument          
```

TheMeaningfulEngineer, 2020-04-08:
The flow in the driver code does seem to not use `Popen` in an asynchronous way.
 https://github.com/labgrid-project/labgrid/blob/b0cc54dc2e2c9cf94c9c98e22f00a150ce475597/labgrid/driver/sshdriver.py#L75-L78

It call `Popen` (which is an async call) and in the next line waits for the returns. I guess in that case I don't see the reason not to use the higher level and blocking [subprocess.run](https://docs.python.org/3/library/subprocess.html#subprocess.run).

```
# labgrid_issue.py
import subprocess    
import shlex    
    
# An ssh command that will fail    
cmd = "ssh root@12345"    
    
process = subprocess.run(shlex.split(cmd), capture_output=True)    
    
print("I am printed:      " + process.stderr.decode())    
```

```
$ python labgrid_issue.py            
I am printed:      ssh: connect to host 0.0.48.57 port 22: Invalid argument
```

Emantor, 2020-04-08:
Sounds sensible, can you submit this as a PR? Don't forget to add the timeout to `subprocess.run`.

closed 2020-04-17 by Emantor