labgrid-project/labgrid #607: how to configure setuptools_scm to ignore tags from labgrid repo

egrigore, 2020-05-21:
Hello,

We have put a tag to labgrid repo - which is not related to labgrid itself or the packages that it installs, it is an internal tag we use for all the repos. After that, labgrid installation failed - because it doesn't find that version to install (which is normal):

```
/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/version.py:92: UserWarning: tag 'rel_imx_5.4.24-2.1.0_rc1' no version found
  warnings.warn("tag {!r} no version found".format(tag))
Traceback (most recent call last):
  File "setup.py", line 74, in <module>
    "Programming Language :: Python :: 3.7",
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/venv/lib/python3.5/site-packages/setuptools/__init__.py", line 144, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.5/distutils/core.py", line 108, in setup
    _setup_distribution = dist = klass(attrs)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/venv/lib/python3.5/site-packages/setuptools/dist.py", line 426, in __init__
    k: v for k, v in attrs.items()
  File "/usr/lib/python3.5/distutils/dist.py", line 281, in __init__
    self.finalize_options()
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/venv/lib/python3.5/site-packages/setuptools/dist.py", line 717, in finalize_options
    ep(self)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/venv/lib/python3.5/site-packages/setuptools/dist.py", line 724, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/integration.py", line 17, in version_keyword
    dist.metadata.version = _get_version(config)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/__init__.py", line 147, in _get_version
    parsed_version = _do_parse(config)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/__init__.py", line 103, in _do_parse
    version = _version_from_entrypoints(config) or _version_from_entrypoints(
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/__init__.py", line 63, in _version_from_entrypoints
    version = _call_entrypoint_fn(root, config, ep.load())
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/__init__.py", line 44, in _call_entrypoint_fn
    return fn(root, config=config)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/git.py", line 138, in parse
    return meta(tag, config=config, node=node, dirty=dirty, branch=branch)
  File "/work/bamboo2/agent-2/IMXTEST-CORETEST-CHECK/labgrid/.eggs/setuptools_scm-3.5.0-py3.5.egg/setuptools_scm/version.py", line 205, in meta
    assert parsed_version is not None, "cant parse version %s" % tag
AssertionError: cant parse version rel_imx_5.4.24-2.1.0_rc1

```
How can we make setup.py to ignore that tag ?

We temporary changed in setup.py use_scm_version to False , but I don't know if that is the right solution: 
`use_scm_version=False`

Can you help with an advice?

Thank you

Emantor, 2020-05-21:
IMO I would like to get rid of `setuptools_scm` altogether, since this interferes with reproducible packaging for NixOS. For `setuptools_scm`, `.git` is required to be inside the repo while building, which makes the download not-reproducable. @jluebbe, what do you think?

Emantor, 2020-05-25:
Further investigation reveals that `export SETUPTOOLS_SCM_PRETEND_VERSION="$version"` can be used to setup a fixed version, you should be able to integrate this into an installation script for your custom version.

egrigore, 2020-06-05:
Hi, I have tried using export SETUPTOOLS_SCM_PRETEND_VERSION="$version" but I don't know what value to give to version? 

I have tried to leave it blank but it doesn't work.

Emantor, 2020-10-26:
My preferred version choice is to add another `.1` add the end of the version and than to enforce this specific version during packaging.
I'll close this because this is answered, version number formats are enforced by the python ecosystem in this case.

closed 2020-10-26 by Emantor