labgrid-project/labgrid #609: adb driver supporting TCP connection

deeagle001, 2020-05-26:
Hi Guys,

I see that the labgrid offers AndroidFastboot driver. I am curious, that is there any ongoing development to implement adb protocol for android devices (maybe using adb executable as an external tool) that supports TCP connection instead of USB?
This will be extremely useful for Android emulators such as QEMU, which does not create a virtual USB device in order to be accessible via adb, but make itself available through TCP.

Thank you in advance!

jluebbe, 2020-05-26:
I don't know of anyone using labgrid for Android development, so as far as I know there is no ongoing work on ADB support. It should probably look similar to the existing NetworkService/SSHDriver (NetworkService should really be renamed to SSHService).

So maybe you'd have a ADBService, perhaps automatically created like `resource/docker.py` does for NetworkService.

Emantor, 2020-10-26:
Barebox AFAIR now supports fastboot over network. I'd gladly take a PR for this feature, otherwise we'll implement this when needed.

jluebbe, 2020-10-26:
This would probably needed a separate Resource to describe how to connect, as it's not related to USB anymore.

Emantor, 2022-10-24:
This was implemented in #959, closing this issue.

closed 2022-10-24 by Emantor