labgrid-project/labgrid #610: first_test-Project doesn't work

mprestel, 2020-05-28:
Hello,
i tried the first_test example in labgrid. I boot up and connected my embedded board via UART and get the following error :

(labgrid-venv) matthias@matthias-OptiPlex-980:~/first_test$ pytest --lg-env local.yaml test_shell.py 
============================= test session starts ==============================
platform linux -- Python 3.5.2, pytest-5.3.1, py-1.8.1, pluggy-0.13.1
rootdir: /home/matthias/first_test
plugins: labgrid-0.2.1.dev447+gd11a5ec
collected 1 item                                                               

test_shell.py E                                                          [100%]

==================================== ERRORS ====================================
_________________________ ERROR at setup of test_shell _________________________

target = Target(name='main', env=Environment(config_file='local.yaml'))

    @pytest.fixture(scope='session')
    def command(target):
>       shell = target.get_driver('CommandProtocol')

conftest.py:6: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/target.py:214: in get_driver
    return self._get_driver(cls, name=name, activate=activate)
../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/target.py:190: in _get_driver
    self.activate(found[0])
../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/target.py:423: in activate
    client.on_activate()
../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/driver/shelldriver.py:60: in on_activate
    self._await_login()
../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/step.py:219: in wrapper
    _result = func(*_args, **_kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = ShellDriver(target=Target(name='main', env=Environment(config_file='local.yaml')), name=None, state=<BindingState.boun...prompt=' login: ', username='root', password='', keyfile='', login_timeout=60, console_ready='', await_login_timeout=2)

    @step()
    def _await_login(self):
        """Awaits the login prompt and logs the user in"""
    
        start = time.time()
    
        expectations = [self.prompt, self.login_prompt, TIMEOUT]
        if self.console_ready != "":
            expectations.append(self.console_ready)
    
        # We call console.expect with a short timeout here to detect if the
        # console is idle, which results in a timeout without any changes to
        # the before property. So we store the last before value we've seen.
        # Because pexpect keeps any read data in it's buffer when a timeout
        # occours, we can't lose any data this way.
        last_before = b''
    
        while True:
            index, before, _, _ = self.console.expect(
                expectations,
                timeout=self.await_login_timeout
            )
    
            if index == 0:
                # we got a promt. no need for any further action to activate
                # this driver.
                self.status = 1
                break
    
            elif index == 1:
                # we need to login
                self.console.sendline(self.username)
                index, _, _, _ = self.console.expect([self.prompt, "Password: "], timeout=10)
                if index == 1:
                    if self.password:
                        self.console.sendline(self.password)
                        remaining_time = (start + self.login_timeout) - time.time()
                        self.console.expect(self.prompt, timeout=remaining_time)
                    else:
                        raise Exception("Password entry needed but no password set")
                self._check_prompt()
                break
    
            elif index == 2:
                # expect hit a timeout while waiting for a match
                if before == last_before:
                    # we did not receive anything during
                    # self.await_login_timeout.
                    # let's assume the target is idle and we can safely issue a
                    # newline to check the state
                    self.console.sendline("")
    
            elif index == 3:
                # we have just activated a console here
                # lets start over again and see if login or prompt will appear
                # now.
                self.console.sendline("")
    
            last_before = before
    
            if time.time() > start + self.login_timeout:
>               raise TIMEOUT("Timeout of {} seconds exceeded during waiting for login".format(self.login_timeout))  # pylint: disable=line-too-long
E               pexpect.exceptions.TIMEOUT: Timeout of 60 seconds exceeded during waiting for login

../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/driver/shelldriver.py:169: TIMEOUT
========================= 1 error in 60.21s (0:01:00) ==========================

jluebbe, 2020-05-28:
Are you sure that your prompt pattern matches what your device sends? If you run pytest -vvs and attach the result, maybe I could say more.

mprestel, 2020-05-28:
(labgrid-venv) matthias@matthias-OptiPlex-980:~/first_test$ pytest -vvs
============================= test session starts ==============================
platform linux -- Python 3.5.2, pytest-5.3.1, py-1.8.1, pluggy-0.13.1 -- /home/matthias/labgrid-venv/bin/python3
cachedir: .pytest_cache
rootdir: /home/matthias/first_test
plugins: labgrid-0.2.1.dev447+gd11a5ec
collected 7 items                                                              

test_hwclock.py::test_hwclock_rate 
SKIPPED
test_hwclock.py::test_hwclock_value 
SKIPPED
test_memory.py::test_memory_mbw 
SKIPPED
test_memory.py::test_memory_memtester_short 
SKIPPED
test_rt.py::test_rt_cyclictest_short 
SKIPPED
test_rt.py::test_rt_hackbench_short 
SKIPPED
test_shell.py::test_shell 
SKIPPED

============================== 7 skipped in 0.04s ==============================

jluebbe, 2020-05-28:
Ah, sorry. Please use `-vvs` in addition to your existing arguments (otherwise the target configuration is not loaded and the tests are skipped).

mprestel, 2020-05-28:
Never mind. At first, thanks for your help. The new output is :
(labgrid-venv) matthias@matthias-OptiPlex-980:~/first_test$ pytest --lg-env local.yaml test_shell.py -vvs
============================= test session starts ==============================
platform linux -- Python 3.5.2, pytest-5.3.1, py-1.8.1, pluggy-0.13.1 -- /home/matthias/labgrid-venv/bin/python3
cachedir: .pytest_cache
rootdir: /home/matthias/first_test
plugins: labgrid-0.2.1.dev447+gd11a5ec
collected 1 item                                                               

test_shell.py::test_shell 
   ShellDriver._await_login state='start'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.001'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.003'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop' duration='2.002'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
[... cut ...]
            if time.time() > start + self.login_timeout:
>               raise TIMEOUT("Timeout of {} seconds exceeded during waiting for login".format(self.login_timeout))  # pylint: disable=line-too-long
E               pexpect.exceptions.TIMEOUT: Timeout of 60 seconds exceeded during waiting for login

../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/driver/shelldriver.py:169: TIMEOUT
========================= 1 error in 60.23s (0:01:00) ==========================

jluebbe, 2020-05-28:
This looks like your target is not sending anything on the console. Labgrid sends a \n to trigger some reaction from the target. You could also add `--lg-log`, which should produce a console_main file with the data received from the target.

mprestel, 2020-05-28:
My target is a x86-board with Ubuntu 18.04 installed and connected via ttyUSB0 with my host. When i am logged in with picocom in this port, i can see the whole debug-messages during boot up.

mprestel, 2020-05-28:
Sorry, but i am complete new with labgrid. The output with --lg-log :

(labgrid-venv) matthias@matthias-OptiPlex-980:~/first_test$ pytest --lg-env local.yaml test_shell.py -vvs --lg-log
============================= test session starts ==============================
platform linux -- Python 3.5.2, pytest-5.3.1, py-1.8.1, pluggy-0.13.1 -- /home/matthias/labgrid-venv/bin/python3
cachedir: .pytest_cache
rootdir: /home/matthias/first_test
plugins: labgrid-0.2.1.dev447+gd11a5ec
collected 1 item                                                               

test_shell.py::test_shell 
   ShellDriver._await_login state='start'
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect state='stop' duration='2.003' result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>)
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]}
     SerialDriver.expect state='stop' duration='0.027' result=(1, b'\r\r\nUbuntu 18.04.4 LTS christ-laptop ttyS0\r\n\r\nchrist-laptop', <_sre.SRE_Match object; span=(58, 66), match=b' login: '>, b' login: ')
     SerialDriver.expect state='start' args={'pattern': ['christ@\\w+:[^ ]+ ', 'Password: ']}
     SerialDriver.expect state='stop' duration='10.011' exception=TIMEOUT("Timeout of 9.97 seconds exceeded or connection closed by peer\n<labgrid.util.expect.PtxExpect object at 0x7f703736f4e0>\ncommand: None\nargs: None\nbuffer (last 100 chars): b'christ\\r\\r\\nPasswort: '\nbefore (last 100 chars): b'christ\\r\\r\\nPasswort: '\nafter: <class 'pexpect.exceptions.TIMEOUT'>\nmatch: None\nmatch_index: None\nexitstatus: None\npid: None\nchild_fd: -1\nclosed: True\ntimeout: 30\ndelimiter: <class 'pexpect.exceptions.EOF'>\nlogfile: None\nlogfile_read: None\nlogfile_send: None\nmaxread: 1\nignorecase: False\nsearchwindowsize: None\ndelaybeforesend: 0.05\ndelayafterclose: 0.1\ndelayafterterminate: 0.1\nsearcher: searcher_re:\n    0: re.compile(b'christ@\\\\w+:[^ ]+ ')\n    1: re.compile(b'Password: ')",)
   ShellDriver._await_login state='stop' duration='12.045' exception=TIMEOUT("Timeout of 9.97 seconds exceeded or connection closed by peer\n<labgrid.util.expect.PtxExpect object at 0x7f703736f4e0>\ncommand: None\nargs: None\nbuffer (last 100 chars): b'christ\\r\\r\\nPasswort: '\nbefore (last 100 chars): b'christ\\r\\r\\nPasswort: '\nafter: <class 'pexpect.exceptions.TIMEOUT'>\nmatch: None\nmatch_index: None\nexitstatus: None\npid: None\nchild_fd: -1\nclosed: True\ntimeout: 30\ndelimiter: <class 'pexpect.exceptions.EOF'>\nlogfile: None\nlogfile_read: None\nlogfile_send: None\nmaxread: 1\nignorecase: False\nsearchwindowsize: None\ndelaybeforesend: 0.05\ndelayafterclose: 0.1\ndelayafterterminate: 0.1\nsearcher: searcher_re:\n    0: re.compile(b'christ@\\\\w+:[^ ]+ ')\n    1: re.compile(b'Password: ')",)
ERROR

==================================== ERRORS ====================================
_________________________ ERROR at setup of test_shell _________________________

self = <pexpect.expect.Expecter object at 0x7f703571e710>
timeout = 9.970213651657104

    def expect_loop(self, timeout=-1):
        """Blocking expect"""
        spawn = self.spawn
    
        if timeout is not None:
            end_time = time.time() + timeout
    
        try:
            incoming = spawn.buffer
            spawn._buffer = spawn.buffer_type()
            spawn._before = spawn.buffer_type()
            while True:
                idx = self.new_data(incoming)
                # Keep reading until exception or return.
                if idx is not None:
                    return idx
                # No match at this point
                if (timeout is not None) and (timeout < 0):
                    return self.timeout()
                # Still have time left, so read more data
>               incoming = spawn.read_nonblocking(spawn.maxread, timeout)

../labgrid-venv/lib/python3.5/site-packages/pexpect/expect.py:111: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <labgrid.util.expect.PtxExpect object at 0x7f703736f4e0>, size = 1
timeout = 9.970213651657104

    def read_nonblocking(self, size=1, timeout=-1):
        """Pexpects needs a nonblocking read function, simply use pyserial with
        a timeout of 0."""
        assert timeout is not None
        if timeout == -1:
            timeout = self.timeout
>       return self.driver.read(size=size, timeout=timeout)

../labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev447+gd11a5ec-py3.5.egg/labgrid/util/expect.py:35: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = SerialDriver(target=Target(name='main', env=Environment(config_file='local.yaml')), name=None, state=<BindingState.active: 2>, txdelay=0.0)
_args = (), _kwargs = {'size': 1, 'timeout': 9.970213651657104}

    @wraps(func)
[... 127 lines cut ...]
E           after: <class 'pexpect.exceptions.TIMEOUT'>
E           match: None
E           match_index: None
E           exitstatus: None
E           pid: None
E           child_fd: -1
E           closed: True
E           timeout: 30
E           delimiter: <class 'pexpect.exceptions.EOF'>
E           logfile: None
E           logfile_read: None
E           logfile_send: None
E           maxread: 1
E           ignorecase: False
E           searchwindowsize: None
E           delaybeforesend: 0.05
E           delayafterclose: 0.1
E           delayafterterminate: 0.1
E           searcher: searcher_re:
E               0: re.compile(b'christ@\\w+:[^ ]+ ')
E               1: re.compile(b'Password: ')

../labgrid-venv/lib/python3.5/site-packages/pexpect/expect.py:82: TIMEOUT
============================== 1 error in 12.23s ===============================
(

mprestel, 2020-05-28:
(labgrid-venv) matthias@matthias-OptiPlex-980:~/first_test$ cat console_main
Labgrid Console Logfile for main
Logfile started at 2020-05-28 16:03:06.068974
=== Log starts here ===

Ubuntu 18.04.4 LTS christ-laptop ttyS0

christ-laptop login: christ
Passwort: Labgrid Console Logfile for main
Logfile started at 2020-05-28 16:05:45.833212
=== Log starts here ===

Ubuntu 18.04.4 LTS christ-laptop ttyS0

christ-laptop login: christ

jluebbe, 2020-05-28:
> E before (last 100 chars): b'christ\r\r\nPasswort: '
…
> E 0: re.compile(b'christ@\w+:[^ ]+ ')
> E 1: re.compile(b'Password: ')

Your target sends 'Passwort:', but labgrid expects 'Password:' ('t' vs. 'd'). This is not configurable so far. For testing, it's probably easiest to disable the password or to change your language configuration to English (which may be useful anyway, as other parts probably expect English output as well).

mprestel, 2020-05-28:
I have now removed the password-line in local.yaml. But this seems to change nothing. The TIMEOUT occurs.

By the way : It seems, that it works, because the console_main contains the text which i have written to the Serial-Console on the target. 

I don't understand. How can the console_main contains the text which was send, even labgrid says that a TIMEOUT occurs ?

jluebbe, 2020-05-28:
> I have now removed the password-line in local.yaml. But this seems to change nothing. The TIMEOUT occurs.

You target probably still requests a password, but labgrid doesn't detect that because of language difference. So it waits forever for 'Password:' or the shell promt (which never appears).

> By the way : It seems, that it works, because the console_main contains the text which i have written to the Serial-Console on the target.

Please make sure to not use the serial port from two programs (i.e. labgrid and your terminal program) at the same time.

> I don't understand. How can the console_main contains the text which was send, even labgrid says that a TIMEOUT occurs ?

Because it is waiting for something that the target is not sending. The console_main contains only the received data (which includes echoed characters).

mprestel, 2020-05-28:
Thanks, i understand. So i should reinstall Ubuntu on the target with english-language. 
Is there still another way ?

jluebbe, 2020-05-28:
You can disable the password on the target (usually by setting an empty one).

mprestel, 2020-05-28:
Now i removed the target-pw with sudo passwd -d christ
But the timout still occurs

jluebbe, 2020-05-28:
You should see similar lines in the error output as before:
> E before (last 100 chars): b'christ\r\r\nPasswort: '
> …
> E 0: re.compile(b'christ@\w+:[^ ]+ ')
> E 1: re.compile(b'Password: ')

That should give you the information what labgrid is waiting for (the re.compile patterns) and what it has received so far (the before line).

Emantor, 2020-05-28:
Due to #411 we currently can't handle empty passwords where a password prompt is still displayed, which seems to be the case here.

mprestel, 2020-05-29:
Thanks. I have now changed the targets systems language to english and setting password same as username. So imy local.yaml looks like the following :
`targets:
  main:
    resources:
      RawSerialPort:
        port: "/dev/ttyUSB0"
    drivers:
      ManualPowerDriver:
        name: "example"
      SerialDriver: {}
      ShellDriver:
        prompt: 'christ@\w+:[^ ]+ '
        login_prompt: ' login: '
        username: 'christ'
        password: 'christ'
`

The output of console_main looks ok :
=== Log starts here ===

Ubuntu 18.04.4 LTS christ-desktop ttyS0

christ-desktop login: christ
Password: 
Last login: Fr Mai 29 08:25:13 CEST 2020 on ttyS0
Welcome to Ubuntu 18.04.4 LTS (GNU/Linux 4.18.0-17-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

 * Canonical Livepatch is available for installation.
   - Reduce system reboots and improve kernel security. Activate at:
     https://ubuntu.com/livepatch

20 Software-Pakete können aktualisiert werden.
11 Aktualisierungen sind Sicherheitsaktualisierungen.

Your Hardware Enablement Stack (HWE) is supported until April 2023.

**But the output of labgrid seems nevertheless run into an timeout :**

`(labgrid-venv) matthias@matthias-OptiPlex-980:~/first_test$ pytest --lg-env local.yaml test_shell.py -vvs --lg-log
=================================================================================== test session starts ===================================================================================
platform linux -- Python 3.5.2, pytest-5.3.1, py-1.8.1, pluggy-0.13.1 -- /home/matthias/labgrid-venv/bin/python3
cachedir: .pytest_cache
rootdir: /home/matthias/first_test
plugins: labgrid-0.2.1.dev447+gd11a5ec
collected 1 item                                                                                                                                                                          

test_shell.py::test_shell 
   ShellDriver._await_login state='start'
     SerialDriver.expect args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]} state='start'
     SerialDriver.expect duration='2.003' result=(2, b'', <class 'pexpect.exceptions.TIMEOUT'>, <class 'pexpect.exceptions.TIMEOUT'>) state='stop'
     SerialDriver.expect args={'pattern': ['christ@\\w+:[^ ]+ ', ' login: ', <class 'pexpect.exceptions.TIMEOUT'>]} state='start'
     SerialDriver.expect duration='0.040' result=(1, b'\r\r\nUbuntu 18.04.4 LTS christ-desktop ttyS0\r\n\r\nchrist-desktop', <_sre.SRE_Match object; span=(60, 68), match=b' login: '>, b' login: ') state='stop'
     SerialDriver.expect args={'pattern': ['christ@\\w+:[^ ]+ ', 'Password: ']} state='start'
     SerialDriver.expect duration='0.031' result=(1, b'christ\r\r\n', <_sre.SRE_Match object; span=(9, 19), match=b'Password: '>, b'Password: ') state='stop'
     SerialDriver.expect args={'pattern': 'christ@\\w+:[^ ]+ '} state='start'
     SerialDriver.expect exception=TIMEOUT("Timeout of 57.15 seconds exceeded or connection closed by peer\n<labgrid.util.expect.PtxExpect object at 0x7f5bc82804a8>\ncommand: None\nargs: None\nbuffer (last 100 chars): b'n.\\r\\n\\r\\nYour Hardware Enablement Stack (HWE) is supported until April 2023.\\r\\nchrist@christ-desktop:~$ '\nbefore (last 100 chars): b'n.\\r\\n\\r\\nYour Hardware Enablement Stack (HWE) is supported until April 2023.\\r\\nchrist@christ-desktop:~$ '\nafter: <class 'pexpect.exceptions.TIMEOUT'>\nmatch: None\nmatch_index: None\nexitstatus: None\npid: None\nchild_fd: -1\nclosed: True\ntimeout: 30\ndelimiter: <class 'pexpect.exceptions.EOF'>\nlogfile: None\nlogfile_read: None\nlogfile_send: None\nmaxread: 1\nignorecase: False\nsearchwindowsize: None\ndelaybeforesend: 0.05\ndelayafterclose: 0.1\ndelayafterterminate: 0.1\nsearcher: searcher_re:\n    0: re.compile(b'christ@\\\\w+:[^ ]+ ')",) duration='57.982' state='stop'
   ShellDriver._await_login exception=TIMEOUT("Timeout of 57.15 seconds exceeded or connection closed by peer\n<labgrid.util.expect.PtxExpect object at 0x7f5bc82804a8>\ncommand: None\nargs: None\nbuffer (last 100 chars): b'n.\\r\\n\\r\\nYour Hardware Enablement Stack (HWE) is supported until April 2023.\\r\\nchrist@christ-desktop:~$ '\nbefore (last 100 chars): b'n.\\r\\n\\r\\nYour Hardware Enablement Stack (HWE) is supported until April 2023.\\r\\nchrist@christ-desktop:~$ '\nafter: <class 'pexpect.exceptions.TIMEOUT'>\nmatch: None\nmatch_index: None\nexitstatus: None\npid: None\nchild_fd: -1\nclosed: True\ntimeout: 30\ndelimiter: <class 'pexpect.exceptions.EOF'>\nlogfile: None\nlogfile_read: None\nlogfile_send: None\nmaxread: 1\nignorecase: False\nsearchwindowsize: None\ndelaybeforesend: 0.05\ndelayafterclose: 0.1\ndelayafterterminate: 0.1\nsearcher: searcher_re:\n    0: re.compile(b'christ@\\\\w+:[^ ]+ ')",) duration='60.060' state='stop'
ERROR

========================================================================================= ERRORS ==========================================================================================
______________________________________________________________________________ ERROR at setup of test_shell _______________________________________________________________________________

self = <pexpect.expect.Expecter object at 0x7f5bc663f908>, timeout = 57.14773631095886

    def expect_loop(self, timeout=-1):
        """Blocking expect"""
[... 163 lines cut ...]
[... cut ...]
E           ignorecase: False
E           searchwindowsize: None
E           delaybeforesend: 0.05
E           delayafterclose: 0.1
E           delayafterterminate: 0.1
E           searcher: searcher_re:
E               0: re.compile(b'christ@\\w+:[^ ]+ ')

../labgrid-venv/lib/python3.5/site-packages/pexpect/expect.py:82: TIMEOUT
=============================================================================== 1 error in 60.26s (0:01:00) ===============================================================================
(`

Emantor, 2020-05-29:
Your prompt contains a `-`, so you'll have to adjust the regex to match the `-` as well. I.e.
```
christ@[\w-]+:[^ ]+ 
```
 for the prompt, I frequently use [regexr.com](regexr.com/55j57) as a nice visual aid to create the regular expressions.

mprestel, 2020-05-29:
It passed ! Great ! Many thanks for all your help !

Many thanks and keep healthy !

closed 2020-05-29 by mprestel