labgrid-project/labgrid #613: sshdriver not returning stdout output

TheMeaningfulEngineer, 2020-06-04:
The code which provoked was a command to read a file on a target. i.e. `cat /etc/file.json`.

I have traced down the issue in the driver and the cause is in [this particular line](https://github.com/labgrid-project/labgrid/blob/master/labgrid/driver/sshdriver.py#L165).

The return from the executed command happens correctly but is than just discarded.
Would be great to know the reasoning behind the `stdout.pop()`

jluebbe, 2020-06-04:
Probably to be compatible to the ShellDriver, which also doesn't have an empty final line. SSHDriver should only remove the last line if it is empty.

TheMeaningfulEngineer, 2020-06-04:
@jluebbe 
Thanks, between the lines of your comment I have read this:

> This is a bug. It should be addressed by ensuring the compatibility happens in a smarted way.

Correct?

jluebbe, 2020-06-04:
Yes. Although the run methods are already doing some processing of the output, so it's not really intended to cover all corner cases of "downloading" files. In the medium term, we should split off a way to run commands and get the raw, unprocessed output.

Bastian-Krause, 2023-07-23:
Should be solved by #654.

closed 2023-07-23 by Bastian-Krause