labgrid-project/labgrid #619: Serial-Login without local.yaml

mprestel, 2020-06-08:
Hello,
I have to login and logout via Serial to my target during runtime in my pytest-File. Well, connecting to the serial port is very easy with : 
`ser = serial.Serial('/dev/ttyUSB0', 115200, timeout=1)`
But how can I do a login with username and password via Serial without writing a seperate function who reads the complete Serial-Input ?
Is there a "smarter way" like a library or function i could use ?

Many thanks !

jluebbe, 2020-06-08:
Login in done automatically by the ShellDriver (using the configured username and password). Logout is not implemented yet.
Why do you need to login again during a test?

mprestel, 2020-06-08:
I have to write a test, which reboots a target many times (>60 times). At first, I switch a relay which powers on my target. After the target boots up, i have to login with username and password via Serial, send a shutdown command and switch the relay to power off the target after the shutdown is being done. 
Sounds easy, but the serial part makes me crazy.

Bastian-Krause, 2020-06-12:
That's exactly what labgrid is for. Have a look at ["Using a Strategy"](https://labgrid.readthedocs.io/en/latest/getting_started.html#using-a-strategy).
The [BareboxRebootStrategy](https://github.com/labgrid-project/labgrid/blob/master/examples/strategy/bareboxrebootstrategy.py) does pretty much what you want, simply remove the barebox part if you want to skip the bootloader part altogether.
It is generally a good idea to invest some time reading about labgrid's driver model in the docs.

Emantor, 2020-10-26:
@mprestel Did this answer your question? If so, please close this issue.

Bastian-Krause, 2023-05-02:
I'll assume that this question is answered. If not, please reopen.

closed 2023-05-02 by Bastian-Krause