labgrid-project/labgrid #626: Is there a way to log on exporter side which user and for how long did he acquired a board?

loutre2412, 2020-07-15:
Hi, 
I'm interested in having a log file on exporter side where I can have details about board's usage (like who acquired it/for how long/ when the board was released etc). Is there an already existing way to do this? I'm open at any suggestion that may help me.
Many thanks!

jluebbe, 2020-07-15:
The exporter doesn't really have this information, but the coordinator could log it. I'd be willing to merge a PR for that.

loutre2412, 2020-07-16:
Actually this is what we want (i should write coordinator instead of exporter). It will be very useful to have this PR.

Bastian-Krause, 2022-09-29:
For the record: #983 added coordinator logging of who acquired which place and when it was released.

Bastian-Krause, 2022-11-17:
See also #1041.