labgrid-project/labgrid #634: labgrid pytest plugin is not working anymore on Ubuntu16.04/python3.5

egrigore, 2020-09-01:
Hi,

Starting from 30 August on all the exporters that have Ubuntu 16.04 and python 3.5 the pytest plugin fails to install labgrid options.  
Is there any deprecation that reached the limit?

Not working example:
```
(venv) bamboo@ubuntu-16-workstation:~/just-for-test/labgrid-tests$ pytest --version
This is pytest version 5.3.1, imported from /home/bamboo/just-for-test/labgrid-tests/venv/lib/python3.5/site-packages/pytest.py
setuptools registered plugins:
  pytest-dependency-0.5.1 at /home/bamboo/just-for-test/labgrid-tests/venv/lib/python3.5/site-packages/pytest_dependency.py
  pytest-html-1.22.1 at /home/bamboo/just-for-test/labgrid-tests/venv/lib/python3.5/site-packages/pytest_html/plugin.py
  pytest-rerunfailures-9.1 at /home/bamboo/just-for-test/labgrid-tests/venv/lib/python3.5/site-packages/pytest_rerunfailures.py
  pytest-metadata-1.8.0 at /home/bamboo/just-for-test/labgrid-tests/venv/lib/python3.5/site-packages/pytest_metadata/plugin.py
(venv) bamboo@ubuntu-16-workstation:~/just-for-test/labgrid-tests$ pytest -h | grep labgrid
(venv) bamboo@ubuntu-16-workstation:~/just-for-test/labgrid-tests$

```

Working example:

```
(venv) bamboo@ubuntu-18-workstation:~/test-env/labgrid-tests$ pytest --version
This is pytest version 5.3.1, imported from /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/pytest.py
setuptools registered plugins:
  pytest-timeout-1.4.2 at /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/pytest_timeout.py
  pytest-rerunfailures-9.1 at /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/pytest_rerunfailures.py
  pytest-html-2.1.1 at /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/pytest_html/plugin.py
  pytest-metadata-1.10.0 at /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/pytest_metadata/plugin.py
  pytest-dependency-0.5.1 at /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/pytest_dependency.py
  labgrid-0.0.0 at /home/bamboo/test-env/labgrid-tests/venv/lib/python3.6/site-packages/labgrid-0.0.0-py3.6.egg/labgrid/pytestplugin/__init__.py
(venv) bamboo@ubuntu-18-workstation:~/test-env/labgrid-tests$ pytest -h | grep labgrid
labgrid:
                        labgrid environment config file (deprecated).
  --lg-env=LG_ENV       labgrid environment config file.
                        labgrid coordinator websocket URL.

```

egrigore, 2020-09-01:
As suggested on [IRC channel](https://riot.im/app/#/room/#labgrid:matrix.org) , it seems that the problem is from the latest version of setuptools: https://github.com/pypa/setuptools/issues/2350 

After adding  SETUPTOOLS_USE_DISTUTILS=stdlib before python setup.py install it seems that it works:

```
(venv) bamboo@ubuntu-16-workstation:~/labgrid-tests$ pytest -h | grep labgrid
/home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/OpenSSL/crypto.py:12: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release ofcryptography. Please upgrade your Python.
  from cryptography import x509
labgrid:
                        labgrid environment config file (deprecated).
  --lg-env=LG_ENV       labgrid environment config file.
                        labgrid coordinator websocket URL.
/usr/lib/python3.5/asyncio/base_events.py:429: ResourceWarning: unclosed event loop <_UnixSelectorEventLoop running=False closed=False debug=False>
(venv) bamboo@ubuntu-16-workstation:~/labgrid-tests$ pytest --version
/home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/OpenSSL/crypto.py:12: CryptographyDeprecationWarning: Python 3.5 support will be dropped in the next release ofcryptography. Please upgrade your Python.
  from cryptography import x509
This is pytest version 5.3.1, imported from /home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/pytest.py
setuptools registered plugins:
  pytest-html-1.22.1 at /home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/pytest_html/plugin.py
  pytest-dependency-0.5.1 at /home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/pytest_dependency.py
  pytest-metadata-1.8.0 at /home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/pytest_metadata/plugin.py
  pytest-rerunfailures-9.1 at /home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/pytest_rerunfailures.py
  labgrid-0.0.0 at /home/bamboo/labgrid-tests/venv/lib/python3.5/site-packages/labgrid-0.0.0-py3.5.egg/labgrid/pytestplugin/__init__.py
```

Some notes:
- upgrading to python >3.5 seems inevitable , so I guess labgrid/setup.py will be changed accordingly right?
- if someone is using the pytest-timeout plugin for the tests , this doesn't seem to work anymore with the latest setuptools

Emantor, 2020-09-28:
>     * upgrading to python >3.5 seems inevitable , so I guess labgrid/setup.py will be changed accordingly right?

Yes. Unless we pin all dependencies for python 3.5, we'll start to see errors in the tests and on running platforms.
At the moment PRs are failing because of python3.5 incompatible changes in six, AFAICS.
 
>     * if someone is using the pytest-timeout plugin for the tests , this doesn't seem to work anymore with the latest setuptools

Thanks for the info.