labgrid-project/labgrid #641: SSHDriver not working reliably but sometimes asking for users password

sessl3r, 2020-10-06:
Hi there,

just using labgrid (v0.2.1 on Ubuntu 18.04) within pytest. I defined some basic local.yaml and some simple test which just executes the hostname command on a remote and checks if the hostname is correct (minimized example for this issue). This is working but not always. When I execute the labgrid-pytest command multiple times it sometimes asks for the password and sometimes it just works as expected and passes.

```
local.yaml
targets:
  c1:
    resources:
      NetworkService:
        address: 'pc1'
        username: 'user'
        password: 'pass'
    drivers:
      SSHDriver: {}
```

```
test_hostname.py
import pytest
from labgrid.protocol import CommandProtocol

@pytest.fixture(scope="module")
def client1(env):
  t = env.get_target("c1")
  return t.get_driver(CommandProtocol)

def test_hostname(client1):
  assert client1.run("echo $(hostname)") == (["pc1"], [], 0)
```

Commandline:
`$ labgrid-client --lg-env=local.yaml`

Result:
Sometimes it passes as one would expect.
Sometimes the following is printed:

```
[...]
collected 2 items

test_hostname.py user@pc1 password:
```

Here I can enter nothing and ssh asks me again. If the correct password is entered it is also not working interestingly. However it should never ask me, and of course not randomly.

With -vvv some more info is given:
```
client1 = SSHDriver(target=Target(name='client1', env=Environment(config_file='local.yaml')), name=None, state=<BindingState.active: 2>, keyfile='', stderr_merge=False)

    def test_hostname(client1):
>       assert client1.run("echo $(hostname)") == (["pc1"], [], 0)
E    AssertionError: assert ([],\n ['Control socket connect(/tmp/labgrid-ssh-tmp-w_vfgo72/control-pc1): '\n  'Connection refused\r',\n  'Permission denied, please try again.\r',\n  'Permission denied, please try again.\r',\n  'user@pc1: Permission denied (publickey,password).\r'],\n 255) == (['pc1'], [], 0)
E      At index 0 diff: [] != ['pc1']
E      Full diff:
E        (
E      +  ['pc1'],
E         [],
E      +  0,
E      -  ['Control socket connect(/tmp/labgrid-ssh-tmp-w_vfgo72/control-pc1): '
E      -   'Connection refused\r',
E      -   'Permission denied, please try again.\r',
E      -   'Permission denied, please try again.\r',
E      -   'user@pc1: Permission denied (publickey,password).\r'],
E      -  255,
E        )
```

This output is gained if I hit enter three times without giving a password. So this output sounds like no password was found in the environment.

Already tried some things which seem to change how often the password request happens (just an feeling, not really tested it long enough):
* Initally the scope of the fixture was session, tested module and also none==function
* Check if normal ssh to same pc with same credentials is reliable (yes it is)
* Executed in a clean session (no tmux, special bash configs etc.)

Emantor, 2020-10-06:
We are using the [sshpass](https://sourceforge.net/projects/sshpass/) program to enter the password at the prompt. This however seems to be failing with a connection refused, either because the Control Socket is not set up correctly or there is a real network connection error.
My general recommendation is to switch to a public key based authentication if possible.
However since we support this use case I'll try to reproduce this locally to investigate why the socket sometimes is not setup correctly.

sessl3r, 2020-10-06:
Hi @Emantor,
thanks for the very quick response and thanks for looking into it.
In the meanwhile I switched (at least parts) to key based logins and that is working without any issues! Thanks for this pointer.
Let me know if we can verify anything on our side regarding this issue.
Regards

PR #715 (2021-02-03): driver/sshdriver: add retry for fast failures
**Description**

Implement retry loop around start of ssh controlmaster connection,
avoiding "connection refused" errors caused by targets starting sshd after network interface is brought up.

Fixes #641
closed 2021-02-03 by Emantor