labgrid-project/labgrid #643: Sshpass subprocess exits with returncode 0

githubchriste, 2020-10-12:
I use SshDriver to connect to a target and execute a bunch of commands. The authentication is done with password, not with keyfile.
Sometimes, I need to restart the target and so, the connection to it lost. After it reboots, I try to reconnect to it, by deactivating and reactivating the SSHDriver. 
However, the subprocess which is started to run sshpass has a returncode of 0, after reactivating the driver, so the user password is required to be entered on the command line.

Emantor, 2020-10-12:
This is probably a duplicate of #641. I have not had the time to reproduce this yet.

Emantor, 2021-09-15:
We have since added a retry loop and switched to SSH's own env variable, so this should be fixed, closing.

closed 2021-09-15 by Emantor