labgrid-project/labgrid #658: SSHDriver Keepalive no longer running if connecting to mulitple SSH sessions on the same host with different ports (containers)

sessl3r, 2020-11-03:
Hi there,

when using the SSHDriver in some setup in which we have one host serving multiple lxc containers which map there ssh ports to eg 10022, 11022, 12022, ... the SSHDriver (sometimes not always) reports the following error message:

```
self = SSHDriver(target=Target(name='s2', env=Environment(config_file='remote.yaml')), name=None, state=<BindingState.active: 2>, keyfile='id_rsa_test', stderr_merge=False)
cmd = 'hostname', codec = 'utf-8', decodeerrors = 'strict', timeout = 30

    def _run(self, cmd, codec="utf-8", decodeerrors="strict", timeout=None): # pylint: disable=unused-argument
        """Execute `cmd` on the target.
    
        This method runs the specified `cmd` as a command on its target.
        It uses the ssh shell command to run the command and parses the exitcode.
        cmd - command to be run on the target
    
        returns:
        (stdout, stderr, returncode)
        """
        if not self._check_keepalive():
>           raise ExecutionError("Keepalive no longer running")
E           labgrid.driver.exception.ExecutionError: Keepalive no longer running

venv/lib/python3.6/site-packages/labgrid-0.2.1.dev497+g1d52675-py3.6.egg/labgrid/driver/sshdriver.py:173: ExecutionError
```

When we reduce our code to only use one SSHDriver per host (so one single mapped port, lets say 10022) the same is working without any issues.

Below I added some minimal example which we used to reproduce the issue without our complete code base.
Also this minimal example most of the times leads to this error.

For us internally we just built a hacky [workaround](https://github.com/sessl3r/labgrid/commit/4118825592a5ad64652cb2ab0a0f613eb61ba077) in which we just restart the keepalive in case it is not reacting. This is sufficient for us to get this whole setup working.
The LXC containers are nothing special and are only mapping some ports (ssh, vnc, ...).

Are we missing something here? Do you need any more input from me?

Thanks
Tobias

remote.yaml:

``` yaml
targets:
  s1:
    resources:
      RemotePlace:
        name: host1-test1
    drivers:
      SSHDriver:
        keyfile: id_rsa_test
  s2:
    resources:
      RemotePlace:
        name: host1-test2
    drivers:
      SSHDriver:
        keyfile: id_rsa_test
```

exporter.yaml
``` yaml
host1-test1:
  ssh:
    cls: NetworkService
    address: host1
    port: 41022
    username: root
    password: root
host1-test2:
  ssh:
    cls: NetworkService
    address: host1
    port: 42022
    username: root
    password: root
```

conftest.py
``` python
import pytest
@pytest.fixture(scope="module")
def s1(env):
        t = env.get_target("s1")
        c = t.get_driver("SSHDriver")
        t.activate(c)
        yield c
        t.deactivate(c)
@pytest.fixture(scope="module")
def s2(env):
        t = env.get_target("s2")
        c = t.get_driver("SSHDriver")
        t.activate(c)
        yield c
        t.deactivate(c)
```

test_keepalive.py
``` python
import pytest
def test_keepalive(s1, s2):
        s1.run_check("hostname")
        s2.run_check("hostname")
```

Emantor, 2020-11-11:
I added a branch with a test which opens an SSH connection to the same host (we use localhost for the tests), but could not reproduce this. See [here](https://github.com/Emantor/labgrid/commit/65f3264ddb1b39f8ef550aba055ad832ec719a27). Am I missing something from you issue description?

sessl3r, 2020-11-11:
Hi. Thanks for adding the test. The thing missing here indeed is the different port. In your test of course we have two sshdriver instances but both are connecting to the same ssh server. This was also working in my tests. The problem started when having multiple ssh servers on the same hostname with different ports.

Emantor, 2020-11-11:
> Hi. Thanks for adding the test. The thing missing here indeed is the different port. In your test of course we have two sshdriver instances but both are connecting to the same ssh server. This was [...]

Emantor, 2020-11-11:
> Hi. Thanks for adding the test. The thing missing here indeed is the different port. In your test of course we have two sshdriver instances but both are connecting to the same ssh server. This was [...]

I see, We probably need to include the port into the control socket name (from memory, have not looked at the code yet). Sorry for the close and reopen, misclick in the github interface…

Emantor, 2020-11-11:
Please try #666. (the PR of the beast! :laughing:)

sessl3r, 2020-11-11:
Thanks. That was quick. Solves the problem in the setup where it was failing before. Great!

Emantor, 2020-11-11:
Now I am thorougly confused, since the ControlMaster instances should be created in different temp directories.
I'll get to this on friday and setup a proper test for it. I'll keep this open since we don't have a fix in master yet.

Emantor, 2021-01-22:
@sessl3r Do you still experience this issue on latest master without the PR that supposedly fixes this?

sessl3r, 2021-01-25:
Can only tell that with this PR it was working fine for me. Need to check with current master - will take some days to get a slot on this setup.

Emantor, 2021-01-25:
> Can only tell that with this PR it was working fine for me. Need to check with current master - will take some days to get a slot on this setup.

Sure, thanks for testing again 👍

sessl3r, 2021-02-01:
Hi again,
just gave it a try with current [master](https://github.com/labgrid-project/labgrid/commit/1d47b1256f49f5f4b4e843539320c4a373a75b6d) and tag v0.3.0. each 50 iterations. both are ok and no keepalive errors occur anymore.

Emantor, 2021-02-01:
Thanks, than I will close this as fixed.

closed 2021-02-01 by Emantor