labgrid-project/labgrid #660: client subcommands scp/rsync return 0 on command error

Bastian-Krause, 2020-11-04:
The return code of the executed commands (scp/rsync) is lost.
```
bst@erbse$ labgrid-client -c conf.yaml scp foo :bar
Selected role main from configuration file
   INFO: Connected to 192.168.23.39
   INFO: Running command: ['scp', '-o', 'ControlPath=/tmp/labgrid-ssh-tmp-ag484isf/control-192.168.23.39', 'foo', '_:bar']
foo: No such file or directory
bst@erbse$ echo $?
0
```

```
bst@erbse$ labgrid-client -c conf.yaml rsync foo :bar
Selected role main from configuration file
   INFO: Connected to 192.168.23.39
   INFO: Running command: ['rsync', '-v', '--rsh=ssh -o ControlPath=/tmp/labgrid-ssh-tmp-2vhi3t7a/control-192.168.23.39', '-rlpt', '--one-file-system', '--progress', 'foo', '_:bar']
sh: rsync: not found
rsync: connection unexpectedly closed (0 bytes received so far) [sender]
rsync error: remote command not found (code 127) at io.c(235) [sender=3.1.3]
bst@erbse$ echo $?
0
```

Emantor, 2020-11-04:
Please try #662.

Bastian-Krause, 2020-11-04:
Every subcommand method that does not raise an exception on error is affected by this. Other than scp/rsync, at least console.

Bastian-Krause, 2023-01-13:
Added ssh remote execution case.

Bastian-Krause, 2023-01-13:
See discussion in #662 for a way forward to fix this.

PR #1137 (2023-04-19): remote/client: return exit code for ssh/scp/rsync/telnet/video/audio/console
**Description**
Correctly set the exitcode for these subcommands. For this we introduce an InteractiveCommandError, which only sets the exitcode to the one received from the program.

**Checklist**
- [x] PR has been tested

Fixes #660
Supersedes #662
closed 2023-04-19 by Emantor