labgrid-project/labgrid #674: Extendable Strategies shipped with labgrid

Bastian-Krause, 2020-11-24:
labgrid ships some Strategies in `labgrid/strategy/`. In a lot of cases these Strategies cover 90% of a use case, e.g. only one additional command must be executed in a certain state or an additional state must be added.
The shipped Strategies do not allow extension, so most parts of an already existing Strategy are duplicated . This could be changed by turning the Strategies shipped with labgrid into GraphStrategies (or into a new type of less complex linear Strategies with the same "one state per method" approach). This makes it possible to use inheritance. With this a custom Strategy could be as simple as:

```python
import attr
from labgrid.factory import target_factory
from labgrid.strategy import GraphStrategy, UBootStrategy

@target_factory.reg_driver
@attr.s(eq=False)
class CustomStrategy(UBootStrategy):
    @GraphStrategy.depends("off")
    def state_uboot(self):
        super().state_uboot()
        self.uboot.run("foo")
```

jluebbe, 2021-01-28:
I'm not convinced that this actually leads to easier-to-understand strategy implementations. There are many benefits to having the full state machine in a single file.

For example, the fix I did in https://github.com/labgrid-project/labgrid/commit/1d47b1256f49f5f4b4e843539320c4a373a75b6d would easily break derived strategies.

Bastian-Krause, 2021-01-29:
I see, what do you think about the
> new type of less complex linear Strategies with the same "one state per method" approach

?

jluebbe, 2021-01-29:
> I see, what do you think about the
> 
> > new type of less complex linear Strategies with the same "one state per method" approach
> 
> ?

So that would mainly avoid having to type "self.transition(Status.barebox)", by handling that in a base class. I'm not sure if that's actually easier to understand. Do you see other benefits?

Bastian-Krause, 2021-02-01:
We had some discussion internally:
- extending upstream strategies is considered error-prone
- GraphStrategies are considered less intuitive
- thus there will probably be no "linear GraphStrategy" (as described above)

closed 2021-02-01 by Bastian-Krause