labgrid-project/labgrid #677: Question: Multi Device Testing - Step Reporter

sessl3r, 2020-11-26:
Hi,

we are building up a testbed which already uses labgrid together with pytest.
From the other bug you might got that this setup consist of multiple 'nodes' (pcs, duts, ...)

The controlling and testing itself works for us. Now we are focusing on logging all the execution steps (just everything).
We found the StepReporter which can be activated to log such things onto console which we would copy to do it into our logging format.

The question here is how can we get more information about those steps. The main missing part is which resource/driver issued some step. I have seen there is a member called resource in the StepEvent but is never set.

Or is this the wrong place to look into? We really want to just log all raw interactions with every resource.

Thanks for your help & Regards,
Tobias

Bastian-Krause, 2020-11-26:
With:
`pytest --lg-coordinator=ws://labgrid:20408/ws --lg-log -vv --lg-env qemu-v7a.yaml tests/test_barebox.py::test_barebox_md5sum`

You get output similar to:                                                                                                                                                                                                                                                  

```
tests/test_barebox.py::test_barebox_md5sum 
   BareboxStrategy.transition state='start' args={'status': 'barebox'}
     BareboxStrategy.transition state='start' args={'status': <Status.off: 1>}
       QEMUDriver.off state='start'
       QEMUDriver.off state='stop'
     BareboxStrategy.transition state='stop' duration='0.001'
     QEMUDriver.off state='start'
     QEMUDriver.off state='stop'
     QEMUDriver.on state='start'
       QEMUDriver.monitor_command state='start' args={'command': 'cont'}
       QEMUDriver.monitor_command state='stop'
     QEMUDriver.on state='stop' duration='0.092'
     BareboxDriver._await_prompt state='start'
       QEMUDriver.expect state='start' args={'pattern': ['barebox@[^:]+:[^ ]+ ', 'stop autoboot', 'Password: ', <class 'pexpect.exceptions.TIMEOUT'>]}
       QEMUDriver.expect state='stop' result=(1, b'\r\n\r\nbarebox 2020.09.0 #1 2020-09-01T00:00:00+00:00\r\n\r\n\r\nBoard: V2P-CA9\r\n\x1b[31mERROR:\x1b[0m smc911x 4e000000.ethernet@3,02000000.of: Device not READY in 100ms aborting\r\n\x1b[31mERROR:\x1b[0m smc911x 4e000000.ethernet@3,02000000.of: probe failed: No such device\r\ncfi_flash 40000000.flash@0,00000000.of: found cfi flash at 0x40000000, size 64 MiB\r\ncfi_flash 40000000.flash@0,00000000.of: found cfi flash at 0x44000000, size 64 MiB\r\nConcatenating MTD devices:\r\n(0): "40000000.flash@0,00000000.of"\r\n(1): "40000000.flash@0,00000000.of"\r\ninto device "nor"\r\n\x1b[31mERROR:\x1b[0m <NULL>: Failed to get \'vmmc\' regulator.\r\nmmci-pl18x 10005000.mmci@5000.of: registered as mci0\r\nmci0: detected SD card version 2.0\r\nmci0: registered disk0\r\nstate: New state registered \'state\'\r\nstate: Using bucket 0@0x00000000\r\nmalloc space: 0x6fefe1c0 -> 0x7fdfc37f (size 255 MiB)\r\nfound force-builtin environment, using defaultenv\r\n\x1b[?25h\r\nHit m for menu or any to ', <re.Match object; span=(922, 935), match=b'stop autoboot'>, b'stop autoboot') duration='0.210'
         QEMUDriver._write state='start' args={'data': b'\n'}
         QEMUDriver._write state='stop'
       QEMUDriver.expect state='start' args={'pattern': ['barebox@[^:]+:[^ ]+ ', 'stop autoboot', 'Password: ', <class 'pexpect.exceptions.TIMEOUT'>]}
       QEMUDriver.expect state='stop' result=(0, b':    3\r\n\x1b[1;32m', <re.Match object; span=(15, 44), match=b'barebox@\x1b[1;36mV2P-CA9:/\x1b[0m '>, b'barebox@\x1b[1;36mV2P-CA9:/\x1b[0m ') duration='0.011'
         QEMUDriver._write state='start' args={'data': b'echo "WYGB""HCTOVQ"\n'}
         QEMUDriver._write state='stop'
       QEMUDriver.expect state='start' args={'pattern': 'WYGBHCTOVQ'}
       QEMUDriver.expect state='stop' result=(0, b'echo "WYGB""HCTOVQ"\r\n', <re.Match object; span=(21, 31), match=b'WYGBHCTOVQ'>, b'WYGBHCTOVQ') duration='0.011'
       QEMUDriver.expect state='start' args={'pattern': 'barebox@[^:]+:[^ ]+ '}
       QEMUDriver.expect state='stop' result=(0, b'\r\n\x1b[1;32m', <re.Match object; span=(9, 38), match=b'barebox@\x1b[1;36mV2P-CA9:/\x1b[0m '>, b'barebox@\x1b[1;36mV2P-CA9:/\x1b[0m ')
     BareboxDriver._await_prompt state='stop' duration='0.233'
   BareboxStrategy.transition state='stop' duration='0.327'
   BareboxDriver.run_check state='start' args={'cmd': 'echo -o md5 test'}
       QEMUDriver._write state='start' args={'data': b'echo -o /cmd \'echo -o md5 test\'; echo "XGFS""MIPIWP"; sh /cmd; echo "XGFS""MIPIWP" $?;\n'}
       QEMUDriver._write state='stop'
     QEMUDriver.expect state='start' args={'pattern': 'XGFSMIPIWP(.*)XGFSMIPIWP\\s+(\\d+)\\s+.*barebox@[^:]+:[^ ]+ '}
     QEMUDriver.expect state='stop' result=(0, b'echo -o /cmd \'echo -o md5 test\'; echo "XGFS""MIPIWP"; sh /cmd; echo "XGFS""MIPIWP" $?;\r\n', <re.Match object; span=(88, 150), match=b'XGFSMIPIWP\r\nXGFSMIPIWP 0\r\n\x1b[1;32mbarebox@>, b'XGFSMIPIWP\r\nXGFSMIPIWP 0\r\n\x1b[1;32mbarebox@\x1b[1;36mV2P-CA9:/\x1b[0m ') duration='0.024'
   BareboxDriver.run_check state='stop' result=[] duration='0.024'
   BareboxDriver.run_check state='start' args={'cmd': 'md5sum md5'}
       QEMUDriver._write state='start' args={'data': b'echo -o /cmd \'md5sum md5\'; echo "GPOY""IYKBPX"; sh /cmd; echo "GPOY""IYKBPX" $?;\n'}
       QEMUDriver._write state='stop'
     QEMUDriver.expect state='start' args={'pattern': 'GPOYIYKBPX(.*)GPOYIYKBPX\\s+(\\d+)\\s+.*barebox@[^:]+:[^ ]+ '}
     QEMUDriver.expect state='stop' result=(0, b'echo -o /cmd \'md5sum md5\'; echo "GPOY""IYKBPX"; sh /cmd; echo "GPOY""IYKBPX" $?;\r\n', <re.Match object; span=(82, 183), match=b'GPOYIYKBPX\r\nd8e8fca2dc0f896fd7cb4cb0031ba249  >, b'GPOYIYKBPX\r\nd8e8fca2dc0f896fd7cb4cb0031ba249  md5\r\nGPOYIYKBPX 0\r\n\x1b[1;32mbarebox@\x1b[1;36mV2P-CA9:/\x1b[0m ') duration='0.023'
   BareboxDriver.run_check state='stop' result=['d8e8fca2dc0f896fd7cb4cb0031ba249  md5'] duration='0.024'
PASSED                                                                                                                                                                                                                                                                    [100%]   QEMUDriver.off state='start'
     QEMUDriver.monitor_command state='start' args={'command': 'quit'}
     QEMUDriver.monitor_command state='stop'
   QEMUDriver.off state='stop' duration='0.009'
```
Every step reporter line starts with the acting driver/strategy. Also try `--lg-colored-steps` for better readability.

sessl3r, 2020-11-26:
Hi,

thanks for the very quick response. This feature i known and very helpful in a single board setup.
In our setup we have - for example the following yaml file:

``` yaml
  test1:
    resources:
      NetworkService:
        address: test1
        username: root
        password: root
    drivers:
      SSHDriver:
        keyfile: keyfile 

   test2:
    resources:
      NetworkService:
        address: test2
        username: root
        password: root
    drivers:
      SSHDriver:
        keyfile: keyfile
   [...] 
```

The output using verbose mode from the stepreporter looks like this:

```
INFO     SSHDriver(target=Target(name='test1', env=Environment(config_file='remote.yaml')), name=None, state=<BindingState.bound: 1>, keyfile='keyfile', stderr_merge=False)(Target(name='test1', env=Environment(config_file='remote.yaml'))):sshdriver.py:138 Conn
ected to test1
INFO     SSHDriver(target=Target(name='test2', env=Environment(config_file='remote.yaml')), name=None, state=<BindingState.bound: 1>, keyfile='keyfile', stderr_merge=False)(Target(name='test2', env=Environment(config_file='remote.yaml'))):sshdriver.py:138 Conn
ected to test2
   SSHDriver.run_check state='start' args={'cmd': 'whoami'}
   SSHDriver.run_check state='stop' result=['root'] duration='0.155'
   SSHDriver.run_check state='start' args={'cmd': 'whoami'}
   SSHDriver.run_check state='stop' result=['root'] duration='0.150'
PASSED
```

Our test executed two times whoami. But to which SSHDriver? We would need the information about the resource name used. Without there is no possibility to match commands to Nodes.

Am I missing something here?

Thanks & Regards,
Tobias

Bastian-Krause, 2020-11-26:
So you want to know the target name per step, right?

You could create a custom StepReporter which inherits either from `StepReporter` or from `ColoredStepReporter`. It should override `_line_format()` and use `event.step.source.target.name` to add the target's name to the output line.

Maybe @Emantor or @jluebbe have a better idea?

sessl3r, 2020-11-26:
Ah that is perfect. We are in any case writing a custom one. Just overseen that the member source is the target. So everything is in place. Thanks for the very fast and good responses!

Bastian-Krause, 2020-11-26:
Glad I could help.

closed 2020-11-26 by Bastian-Krause

Emantor, 2020-11-26:
> So you want to know the target name per step, right?
>
> You could create a custom StepReporter which inherits either from `StepReporter` or from `ColoredStepReporter`. It should override [...]

The new logging reporter I am working should have the target name in the logs. Working on it again tomorrow, hopefully I have something to PR than.