labgrid-project/labgrid #684: ProcessWrapper changes breaks pytester plugin and testdir fixture

jremmet, 2020-12-03:
Since
commit: 002b90dcbdab ("util/helper: make ProcessWrapper callbacks non-nested static methods")
the usage of pytester plugin is broken.

Testcase: `pytest test_tmpdir.py`

test_tmpdir.py:
```
import pytest

pytest_plugins = ["pytester"]

def test_testdir(testdir):
    testdir.parseconfigure("-h")
```

Will result in

```
____________________________________________________________ test_testdir ____________________________________________________________

testdir = <Testdir local('/tmp/pytest-of-jremmet/pytest-2/test_testdir0')>

    def test_testdir(testdir):
>       testdir.parseconfigure("-h")

/home/jremmet/tmp/labgrid/test_tmpdir.py:6: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
                buf += b'\n'
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/_pytest/pytester.py:1130: in parseconfigure
    config._do_configure()
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/_pytest/config/__init__.py:982: in _do_configure
    self.hook.pytest_configure.call_historic(kwargs=dict(config=self))
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/pluggy/hooks.py:308: in call_historic
    res = self._hookexec(self, self.get_hookimpls(), kwargs)
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/pluggy/manager.py:93: in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/pluggy/manager.py:87: in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/labgrid/pytestplugin/hooks.py:50: in pytest_configure
    processwrapper.enable_logging()
/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/labgrid/util/helper.py:128: in enable_logging
    self.register(ProcessWrapper.log_callback)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = ProcessWrapper(callbacks=[<function ProcessWrapper.log_callback at 0x7efeb3370560>])
callback = <function ProcessWrapper.log_callback at 0x7efeb3370560>

    def register(self, callback):
        """Register a callback with the ProcessWrapper"""
>       assert callback not in self.callbacks
E       assert <function ProcessWrapper.log_callback at 0x7efeb3370560> not in [<function ProcessWrapper.log_callback at 0x7efeb3370560>]
E        +  where [<function ProcessWrapper.log_callback at 0x7efeb3370560>] = ProcessWrapper(callbacks=[<function ProcessWrapper.log_callback at 0x7efeb3370560>]).callbacks

/home/jremmet/tmp/labgrid/venv-only/lib/python3.7/site-packages/labgrid/util/helper.py:104: AssertionError
====================================================== short test summary info =======================================================
FAILED test_tmpdir.py::test_testdir - assert <function ProcessWrapper.log_callback at 0x7efeb3370560> not in [<function ProcessWrap...
========================================================= 1 failed in 0.12s ==========================================================
```

Emantor, 2020-12-03:
Why is our labgrid client hook called multiple times? According to the docs it should only be called for the labgrid pytestplugin, [see here](https://docs.pytest.org/en/stable/reference.html#pytest.hookspec.pytest_configure).

jremmet, 2020-12-03:
> Why is our labgrid client hook called multiple times? According to the docs it should only be called for the labgrid pytestplugin, [see here](https://docs.pytest.org/en/stable/reference.html#pytest.hookspec.pytest_configure).

I think it is rerun by [testdir](https://docs.pytest.org/en/stable/reference.html?highlight=pytester#testdir) when its setup the Testdir instance.

Emantor, 2020-12-04:
Opened a discussion at https://github.com/pytest-dev/pytest/discussions/8099.

Bastian-Krause, 2020-12-04:
@jremmet Can you elaborate on your use case for pytester/testdir?

RonnyPfannschmidt, 2020-12-04:
the testdir fixture is a utility for doing integration tests against pytest, it runs multiple pytest sessions in a process

so the hook happens for each of those pytest sessions

jremmet, 2020-12-07:
> @jremmet Can you elaborate on your use case for pytester/testdir?

We have own hooks for ordering and configuring of our board tests. The tests use labgrid to run the tests on embedded boards. Our hooks provide also labgrid environment (like --lg-env).
To tests this hooks we use testdir. Due to the usage of labgrid functionality the hooks are not easy to test without the labgrid plugin.

closed 2020-12-07 by Emantor