labgrid-project/labgrid #688: run pytests on target

HerrMuellerluedenscheid, 2020-12-05:
Hey,
We have some python software that we deploy on the target that come with pytests. I was experimenting with labgrid pytest test such as:
```
import pytest

def test_py(command):
    """Test memcopy bandwidth"""

    result = command.run_check('cd fake-test;pytest')
    assert "passed" in result.decode()
```
I was wondering if there is a more direct way to execute pytests on the target, instead of parsing output of tests and interpreting the result.
Thanks in advance!

Emantor, 2020-12-07:
No, we support simple executions as shown on in your example, but there is no further support for parsing the output of pytest or connecting both python processes on the target and test runner to retrieve test output information. If you want to implement parsing of pytest output and contribute that back to the community, https://github.com/labgrid-project/labgridhelper is the correct project for this.
Another option is to start the remote process with junit-xml and parsing the output of this file, this should be more stable than parsing the CLI output.

HerrMuellerluedenscheid, 2020-12-07:
@Emantor Ok, I'll try to come up with a solution for that. Thanks for info!

closed 2020-12-07 by HerrMuellerluedenscheid