labgrid-project/labgrid #692: First example fails

HerrMuellerluedenscheid, 2020-12-10:
Hi,
I followed the gettings started guide and updated the `local.yaml`. When I run:

     pytest --lg-env local.yaml  --lg-log=logdir test_shell.py

I get the following error stack:

> 
> test_shell.py E                                                                                  [100%]
> 
> ================================================ ERRORS ================================================
> _____________________________________ ERROR at setup of test_shell _____________________________________
> 
> target = Target(name='main', env=Environment(config_file='local.yaml'))
> 
>     @pytest.fixture(scope='session')
>     def command(target):
> >       shell = target.get_driver('CommandProtocol')
> 
> conftest.py:6: 
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/target.py:218: in get_driver
>     return self._get_driver(cls, name=name, activate=activate)
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/target.py:194: in _get_driver
>     self.activate(found[0])
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/target.py:433: in activate
>     client.on_activate()
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/driver/shelldriver.py:71: in on_activate
>     self._run("dmesg -n 1")
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/driver/shelldriver.py:84: in _run
>     self._check_prompt()
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/driver/shelldriver.py:188: in _check_prompt
>     timeout=5
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/binding.py:100: in wrapper
>     return func(self, *_args, **_kwargs)
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/step.py:219: in wrapper
>     _result = func(*_args, **_kwargs)
> ../../../../.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974-py3.7.egg/labgrid/driver/consoleexpectmixin.py:54: in expect
>     index = self._expect.expect(pattern, timeout=timeout)
> ../../../../.local/lib/python3.7/site-packages/pexpect/spawnbase.py:344: in expect
>     timeout, searchwindowsize, async_)
> ../../../../.local/lib/python3.7/site-packages/pexpect/spawnbase.py:372: in expect_list
>     return exp.expect_loop(timeout)
> ../../../../.local/lib/python3.7/site-packages/pexpect/expect.py:181: in expect_loop
>     return self.timeout(e)
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
> 
> self = <pexpect.expect.Expecter object at 0x7f6649531748>
> err = TIMEOUT('Timeout of 4.94 seconds exceeded or connection closed by peer')
> 
>     def timeout(self, err=None):
>         spawn = self.spawn
>     
>         spawn.before = spawn._before.getvalue()
>         spawn.after = TIMEOUT
>         index = self.searcher.timeout_index
>         if index >= 0:
>             spawn.match = TIMEOUT
>             spawn.match_index = index
>             return index
>         else:
>             spawn.match = None
>             spawn.match_index = None
>             msg = str(spawn)
>             msg += '\nsearcher: %s' % self.searcher
>             if err is not None:
>                 msg = str(err) + '\n' + msg
>     
>             exc = TIMEOUT(msg)
>             exc.__cause__ = None    # in Python 3.x we can use "raise exc from None"
> >           raise exc
> E           pexpect.exceptions.TIMEOUT: Timeout of 4.94 seconds exceeded or connection closed by peer
> E           <labgrid.util.expect.PtxExpect object at 0x7f6649578cc0>
> E           command: None
> E           args: None
> E           buffer (last 100 chars): b" echo 'UMZI''XPKULO'\r\nUMZIXPKULO\r\nroot@2B8A57EF:~# "
> E           before (last 100 chars): b" echo 'UMZI''XPKULO'\r\nUMZIXPKULO\r\nroot@2B8A57EF:~# "
> E           after: <class 'pexpect.exceptions.TIMEOUT'>
> E           match: None
> E           match_index: None
> E           exitstatus: None
> E           pid: None
> E           child_fd: -1
> E           closed: True
> E           timeout: 30
> E           delimiter: <class 'pexpect.exceptions.EOF'>
> E           logfile: None
> E           logfile_read: None
> E           logfile_send: None
> E           maxread: 1
> E           ignorecase: False
> E           searchwindowsize: None
> E           delaybeforesend: 0.05
> E           delayafterclose: 0.1
> E           delayafterterminate: 0.1
> E           searcher: searcher_re:
> E               0: re.compile(b'UMZIXPKULO\\s+[A-Z0-9]{4,8}:\\~#')
> 
> ../../../../.local/lib/python3.7/site-packages/pexpect/expect.py:144: TIMEOUT
> 

When I check the log output I actually see communication taking place:

> root@2B8A57EF:~# Labgrid Console Logfile for main
> Logfile started at 2020-12-11 00:18:55.593252
> === Log starts here ===
> 
> root@2B8A57EF:~# run() { echo -n "$MARKER"; sh -c "$@"; echo "$MARKER $?"; }
> root@2B8A57EF:~# echo 'UMZI''XPKULO'
> UMZIXPKULO

HerrMuellerluedenscheid, 2020-12-10:
The connection is actually pretty responsive, thus a little strange it is running into a timeout. Is there maybe some return character slipping through?

Emantor, 2020-12-11:
`[A-Z0-9]{4,8}:\\~#` does not match your prompt, did you add the `\\` or are they added by python? Please post your `local.yaml` file as well. Unless your logfile is truncated your device does not appear to display a prompt after the echo command.

HerrMuellerluedenscheid, 2020-12-11:
For some reason I thought I need to escape the  `~`. Anyways, after removing that (single) backslash, the result and behaviour stays the same. This is my local.yaml:
```
targets:
  main:
    resources:
      RawSerialPort:
        port: "/dev/ttyUSB0"
    drivers:
      ManualPowerDriver:
        name: "example"
      SerialDriver: {}
      ShellDriver:
        prompt: '[A-Z0-9]{4,8}:~#'
        login_prompt: '.* login: '
        username: 'root'
```

HerrMuellerluedenscheid, 2020-12-11:
Ah, damn. I was on another platform where the `root@` was missing before the hostname on the prompt. Sorry, I was blind! Now, it's working!

closed 2020-12-11 by HerrMuellerluedenscheid