labgrid-project/labgrid #696: Which is the right ID_PATH to use in usbpower cycle example

HerrMuellerluedenscheid, 2020-12-16:
Hey,

I'm trying to get the `usbpower/cycle.py` example to work but I'm failling so far. I'm using the AmazonBasics HU9003V1EBL hub. From user space I turn a testing port on/off:

    uhubctl -a on -l 1-1  -p 3 

Do I understand correctly that `-p 3` would coincide with `index=3` in the example?

I attempted to find the correct `PATH_ID` with `udevadm info /sys/bus/usb/devices/* | less`. From the vendor ID and PID (`2109:2817`) I identified this block to be the hub (not sure if this is correct, though):
```
P: /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.4
N: bus/usb/001/004
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.4
E: DEVNAME=/dev/bus/usb/001/004
E: DEVTYPE=usb_device
E: DRIVER=usb
E: PRODUCT=2109/2817/9023
E: TYPE=9/0/2
E: BUSNUM=001
E: DEVNUM=004
E: MAJOR=189
E: MINOR=3
E: SUBSYSTEM=usb
E: USEC_INITIALIZED=3085613
E: ID_VENDOR=VIA_Labs__Inc.
E: ID_VENDOR_ENC=VIA\x20Labs\x2c\x20Inc.\x20\x20\x20\x20\x20\x20\x20\x20\x20
E: ID_VENDOR_ID=2109
E: ID_MODEL=USB2.0_Hub
E: ID_MODEL_ENC=USB2.0\x20Hub\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20
E: ID_MODEL_ID=2817
E: ID_REVISION=9023
E: ID_SERIAL=VIA_Labs__Inc._USB2.0_Hub
E: ID_BUS=usb
E: ID_USB_INTERFACES=:090001:090002:
E: ID_VENDOR_FROM_DATABASE=VIA Labs, Inc.
E: ID_PATH=pci-0000:00:14.0-usb-0:1.4
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_1_4
E: ID_FOR_SEAT=usb-pci-0000_00_14_0-usb-0_1_4
E: TAGS=:seat:
```
Thus, in the example I set:

    match={'ID_PATH': 'pci-0000:00:14.0-usb-0:1.4:1.0'}, index=3)

Unfortunately, I'm getting the following exception:

```
Traceback (most recent call last):
  File "cycle.py", line 14, in <module>
    t.activate(upd)
  File "/home/marius/.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974.d20201210-py3.7.egg/labgrid/target.py", line 424, in activate
    self.await_resources(resources)
  File "/home/marius/.local/lib/python3.7/site-packages/labgrid-0.2.1.dev541+g1613974.d20201210-py3.7.egg/labgrid/target.py", line 95, in await_resources
    filter=waiting
labgrid.exceptions.NoResourceFoundError: Not all resources are available: {USBPowerPort(target=Target(name='main', env=None), name=None, state=<BindingState.bound: 1>, avail=False, timeout=5.0, match={'ID_PATH': 'pci-0000:00:14.0-usb-0:1:1.0', 'DEVTYPE': 'usb_interface', 'DRIVER': 'hub', 'SUBSYSTEM': 'usb'}, device=None, index=3)}
```

I also tried other `ID_PATH`s and `index`es but so far without luck. What am I missing here?

Regards

Marius

HerrMuellerluedenscheid, 2020-12-26:
I tried to same on a different platform but with the same result, unfortunately. Is there a way to debug this through `uhubctl` maybe?

HerrMuellerluedenscheid, 2020-12-26:
I'm also a little confused by that 'Note' box in the USBPowerPort section https://labgrid.readthedocs.io/en/latest/configuration.html
None of the devices that are listed when running `udevadm info /sys/bus/usb/devices/* | less` has an `ID_PATH` that is ending ending with `:1.0`.

Emantor, 2020-12-26:
On my devel lab machine with a switchable USB HUB, `1-3` identifies itself as a generic USB device:
```
root@dollysisters:~# udevadm info /sys/bus/usb/devices/1-3
P: /devices/pci0000:00/0000:00:14.0/usb1/1-3
N: bus/usb/001/003
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3
E: DEVNAME=/dev/bus/usb/001/003
E: DEVTYPE=usb_device
E: DRIVER=usb
E: PRODUCT=5e3/608/6052
E: TYPE=9/0/1
E: BUSNUM=001
E: DEVNUM=003
E: MAJOR=189
E: MINOR=2
E: SUBSYSTEM=usb
E: USEC_INITIALIZED=3771202
E: ID_VENDOR=05e3
E: ID_VENDOR_ENC=05e3
E: ID_VENDOR_ID=05e3
E: ID_MODEL=USB2.0_Hub
E: ID_MODEL_ENC=USB2.0\x20Hub
E: ID_MODEL_ID=0608
E: ID_REVISION=6052
E: ID_SERIAL=05e3_USB2.0_Hub
E: ID_BUS=usb
E: ID_USB_INTERFACES=:090000:
E: ID_VENDOR_FROM_DATABASE=Genesys Logic, Inc.
E: ID_MODEL_FROM_DATABASE=Hub
E: ID_PATH=pci-0000:00:14.0-usb-0:3
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_3
E: ID_FOR_SEAT=usb-pci-0000_00_14_0-usb-0_3
E: TAGS=:seat:
```
While the corresponding interface `:1.0` is bound to the `hub` driver:
```
root@dollysisters:~# udevadm info /sys/bus/usb/devices/1-3:1.0
P: /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0
E: DEVTYPE=usb_interface
E: DRIVER=hub
E: PRODUCT=5e3/608/6052
E: TYPE=9/0/1
E: INTERFACE=9/0/0
E: MODALIAS=usb:v05E3p0608d6052dc09dsc00dp01ic09isc00ip00in00
E: SUBSYSTEM=usb
E: USEC_INITIALIZED=3771889
E: ID_USB_CLASS_FROM_DATABASE=Hub
E: ID_USB_PROTOCOL_FROM_DATABASE=Single TT
E: ID_VENDOR_FROM_DATABASE=Genesys Logic, Inc.
E: ID_MODEL_FROM_DATABASE=Hub
E: ID_PATH=pci-0000:00:14.0-usb-0:3:1.0
E: ID_PATH_TAG=pci-0000_00_14_0-usb-0_3_1_0
```
and the device with the `hub` driver identifies itself with `:1.0` in the `ID_PATH`. Can you add your udevadm info output?

Ref https://github.com/labgrid-project/labgrid/pull/291

HerrMuellerluedenscheid, 2020-12-26:
Hey,
sure
```
xxx@ci-coordinator:~ $ sudo  udevadm info /sys/bus/usb/devices/1-1.3
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3
N: bus/usb/001/004
L: 0
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3
E: DEVNAME=/dev/bus/usb/001/004
E: DEVTYPE=usb_device
E: DRIVER=usb
E: PRODUCT=2109/2817/364
E: TYPE=9/0/2
E: BUSNUM=001
E: DEVNUM=004
E: MAJOR=189
E: MINOR=3
E: SUBSYSTEM=usb
E: USEC_INITIALIZED=7488349
E: ID_VENDOR=VIA_Labs__Inc.
E: ID_VENDOR_ENC=VIA\x20Labs\x2c\x20Inc.\x20\x20\x20\x20\x20\x20\x20\x20\x20
E: ID_VENDOR_ID=2109
E: ID_MODEL=USB2.0_Hub
E: ID_MODEL_ENC=USB2.0\x20Hub\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20\x20
E: ID_MODEL_ID=2817
E: ID_REVISION=0364
E: ID_SERIAL=VIA_Labs__Inc._USB2.0_Hub
E: ID_BUS=usb
E: ID_USB_INTERFACES=:090001:090002:
E: ID_VENDOR_FROM_DATABASE=VIA Labs, Inc.
E: ID_PATH=platform-3f980000.usb-usb-0:1.3
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_3
E: ID_FOR_SEAT=usb-platform-3f980000_usb-usb-0_1_3
E: TAGS=:seat:
```

Compared to your output from the hub driver mine does not expose a `ID_PATH`. Strange.

```
ansible@ci-coordinator:~ $ sudo  udevadm info /sys/bus/usb/devices/1-1.3:1.0
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0
L: 0
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0
E: DEVTYPE=usb_interface
E: DRIVER=hub
E: PRODUCT=2109/2817/364
E: TYPE=9/0/2
E: INTERFACE=9/0/2
E: MODALIAS=usb:v2109p2817d0364dc09dsc00dp02ic09isc00ip02in00
E: SUBSYSTEM=usb
E: USEC_INITIALIZED=7647123
E: ID_USB_CLASS_FROM_DATABASE=Hub
E: ID_USB_PROTOCOL_FROM_DATABASE=TT per port
E: ID_VENDOR_FROM_DATABASE=VIA Labs, Inc.
```

Emantor, 2020-12-26:
`DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3:1.0` should work equaly well as `ID_PATH`, we support all udev properties for device matching. Still strange that the HUB has no `ID_PATH`, my example were from an x86_64 intel NUC with connected USB Hub.

HerrMuellerluedenscheid, 2020-12-26:
Nice! Matching against `DEVPATH` works flawless! I would guess that quite a number of people might start with a RaspberryPi running raspbian. Maybe it would be legit to switch the documentation to match `DEVPATH` rather than `ID_PATH`? I'm not really that experienced, though.
But awesome that it works now. Thanks a lot!! And Merry Christmas :christmas_tree:

closed 2020-12-26 by HerrMuellerluedenscheid

Emantor, 2020-12-27:
> Maybe it would be legit to switch the documentation to match `DEVPATH` rather than `ID_PATH`? I'm not really that experienced, though.

We should at least mention that all udev properties are supported and add `DEVPATH` as another example.