labgrid-project/labgrid #697: sshdriver: sshpass: fails when connecting through remote exporter

alegsan, 2020-12-17:
Hi,

our exporter configuration looks like that:
```
750-8216:                                                                           
  USBSerialPortService:                                                             
    cls: USBSerialPort                                                              
    match:                                                                          
      '@ID_SERIAL_SHORT': '1284744278'                                              
                                                                                    
  USBSerialPortSerial:                                                              
    cls: USBSerialPort                                                              
    match:                                                                          
      '@sys_name': '3-1.3.2.4.3:1.0'                                                
                                                                                    
  USBSDMuxDevice:                                                                   
    match:                                                                          
      ID_SERIAL_SHORT: '000000000563'                                               
                                                                                    
  NetworkService:                                                                   
    address: 192.168.1.157                                                          
    username: root                                                                  
    password: root                                                                                                
```

When we want to open a ssh session to the `NetworkService` from a remote client we get the exception:
```
(labgrid-venv) alegsan@nas:~/workspace/git/labgrid$ labgrid-client -vvv -P lc013508 -p 750-8216 ssh
  DEBUG: Creating SSHConnection for lc013508
  DEBUG: No existing control socket found
  DEBUG: ControlSocket: /tmp/labgrid-connection-vcga0abn/control-lc013508
  DEBUG: Master Start command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -n -MN -o ConnectTimeout=30 -o ControlPersist=300 -o ControlMaster=yes -o ControlPath=/tmp/labgrid-connection-vcga0abn/control-lc013508 -o StrictHostKeyChecking=yes lc013508
  DEBUG: Connected to lc013508
   INFO: Created new SSH connection to lc013508
  DEBUG: Started keepalive for lc013508
  DEBUG: Running control command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -o ControlMaster=no -o ControlPath=/tmp/labgrid-connection-vcga0abn/control-lc013508 -O forward -L52732:127.0.0.1:20408 lc013508
  DEBUG: expanded remote resources for place 750-8216: [NetworkService(target=Target(name='750-8216', env=None), name='NetworkService', state=<BindingState.bound: 1>, avail=True, address='192.168.1.157', username='root', password='root', port=22), NetworkSerialPort(target=Target(name='750-8216', env=None), name='serial', state=<BindingState.bound: 1>, avail=True, host='lc013508', port=44705, speed=115200, protocol='rfc2217'), NetworkUSBSDMuxDevice(target=Target(name='750-8216', env=None), name='USBSDMuxDevice', state=<BindingState.bound: 1>, avail=True, host='lc013508', timeout=10.0, busnum=1, devnum=5, path='/dev/sda', vendor_id=1060, model_id=16449, control_path='/dev/sg0'), NetworkSerialPort(target=Target(name='750-8216', env=None), name='service', state=<BindingState.bound: 1>, avail=True, host='lc013508', port=58375, speed=115200, protocol='rfc2217')]
  DEBUG: Creating SSHConnection for lc013508
  DEBUG: No existing control socket found
  DEBUG: ControlSocket: /tmp/labgrid-connection-8t3vk8gn/control-lc013508
  DEBUG: Master Start command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -n -MN -o ConnectTimeout=30 -o ControlPersist=300 -o ControlMaster=yes -o ControlPath=/tmp/labgrid-connection-8t3vk8gn/control-lc013508 -o StrictHostKeyChecking=yes lc013508
  DEBUG: Connected to lc013508
   INFO: Created new SSH connection to lc013508
  DEBUG: Started keepalive for lc013508
Traceback (most recent call last):
  File "/home/alegsan/workspace/git/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev552+gc96120a-py3.5.egg/labgrid/remote/client.py", line 1621, in main
    args.func(session)
  File "/home/alegsan/workspace/git/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev552+gc96120a-py3.5.egg/labgrid/remote/client.py", line 1005, in ssh
    drv = self._get_ssh()
  File "/home/alegsan/workspace/git/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev552+gc96120a-py3.5.egg/labgrid/remote/client.py", line 1001, in _get_ssh
    target.activate(drv)
  File "/home/alegsan/workspace/git/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev552+gc96120a-py3.5.egg/labgrid/target.py", line 433, in activate
    client.on_activate()
  File "/home/alegsan/workspace/git/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev552+gc96120a-py3.5.egg/labgrid/driver/sshdriver.py", line 44, in on_activate
    self.control = self._start_own_master()
  File "/home/alegsan/workspace/git/labgrid/labgrid-venv/lib/python3.5/site-packages/labgrid-0.2.1.dev552+gc96120a-py3.5.egg/labgrid/driver/sshdriver.py", line 135, in _start_own_master
    "no control socket to {}".format(self.networkservice.address)
labgrid.driver.exception.ExecutionError: no control socket to 192.168.1.157
   INFO: Closing SSH connection to lc013508
  DEBUG: Stopping keepalive for lc013508
  DEBUG: Running control command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -o ControlMaster=no -o ControlPath=/tmp/labgrid-connection-8t3vk8gn/control-lc013508 -O cancel lc013508
  DEBUG: Running control command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -o ControlMaster=no -o ControlPath=/tmp/labgrid-connection-8t3vk8gn/control-lc013508 -O exit lc013508
   INFO: Closing SSH connection to lc013508
  DEBUG: Stopping keepalive for lc013508
  DEBUG: Running control command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -o ControlMaster=no -o ControlPath=/tmp/labgrid-connection-vcga0abn/control-lc013508 -O cancel lc013508
  DEBUG: Running control command: ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -o ControlMaster=no -o ControlPath=/tmp/labgrid-connection-vcga0abn/control-lc013508 -O exit lc013508
```
After that we printed the `args` variable before line https://github.com/labgrid-project/labgrid/blob/c96120adb30ea466929d03f6cecbfd7b123765ea/labgrid/driver/sshdriver.py#L98 and executed the command directly on the shell:
```
(labgrid-venv) alegsan@nas:~/workspace/git/labgrid$ export SSHPASS=root; sshpass -e ssh -f -o LogLevel=ERROR -x -o ConnectTimeout=30 -o ControlPersist=300 -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -o ServerAliveInterval=15 -MN -S /tmp/labgrid-ssh-tmp-nh04nktg/control-192.168.1.157 -p 22 -l root 192.168.1.157 -o 'ProxyCommand=ssh -x -o LogLevel=ERROR -o PasswordAuthentication=no -o ControlMaster=no -o ControlPath=/tmp/labgrid-connection-_j58vyxj/control-lc013508 lc013508 -W 192.168.1.157:22 2>/tmp/labgrid-ssh-tmp-nh04nktg/proxy-stderr'
packet_write_wait: Connection to UNKNOWN port 0: Broken pipe
```
Further investigation shows that if we remove the `-f` (go to background) parameter to the `ssh` command the error `packet_write_wait: Connection to UNKNOWN port 0: Broken pipe` disappears but only in combination with `sshpass`. Without `sshpass` the ssh connection is correctly established in background with `-f` parameter. So the problem here seems to be `sshpass`.

Does anybody has the same problem in combination of using `sshpass`? Does anybody has a hint how to solve this problem? 

We know that its also possible to use ssh-key login instead `sshpass` but in this case we would need to preconfigure our targets after each deployment step. So the use of `sshpass` is very comfortable in our testracks and daily development use cases.

Thanks

Emantor, 2020-12-17:
There seem to be at lot of bugs related to the sshpass support. I'll try to add a test case which captures this edge case. This may also be related to the sshpass version, AFAICS the program has not been updated in some time.

Emantor, 2020-12-17:
Related: https://github.com/clarkwang/passh which has some examples and also includes `sshpass -f` in the broken examples.

jluebbe, 2020-12-18:
We already control a large part of the SSH connection setup (such as starting the control master). Maybe handing password entry ourselves would be more robust.

alegsan, 2020-12-18:
> There seem to be at lot of bugs related to the sshpass support. I'll try to add a test case which captures this edge case. This may also be related to the sshpass version, AFAICS the program has not been updated in some time.

Yes, after some searching there are many posts that are related to the sshpass problem above. There are some suggestions to use `nohup` in combination with `sshpass`. For example `sshpass -p pass nohup ssh -f ...`. This works with the above command on the shell but not when we put this in the `args`. But i think this would only be a temporarly workaround.
By the way, updating to the "newest" version of sshpass did not solved the issue.

> We already control a large part of the SSH connection setup (such as starting the control master). Maybe handing password entry ourselves would be more robust.

With the experience of sshpass i think this would be the best/independently solution. Any hints how and where this should be implemented? Currently we are not the best python programmers, but maybe we could start implementing this? In our case we need this password based possibility.

Are you guys using only the ssh-key possibility?

alegsan, 2021-01-12:
@Emantor @jluebbe any progress/hints here for us? we tried the workaround wich i mented above with `nohup` but without success....

closed 2021-05-07 by alegsan