labgrid-project/labgrid #760: Coordinator broken in v0.3.2

sephalon, 2021-03-30:
When starting Crossbar on Ubuntu 18.04 in a virtual environment, the following error message appears:

```
2021-03-29T17:36:11+0200 [Controller  27287] 
2021-03-29T17:36:11+0200 [Controller  27287]     :::::::::::::::::
2021-03-29T17:36:11+0200 [Controller  27287]           :::::          _____                      __
2021-03-29T17:36:11+0200 [Controller  27287]     :::::   :   :::::   / ___/____ ___   ___  ___  / /  ___ _ ____
2021-03-29T17:36:11+0200 [Controller  27287]     :::::::   :::::::  / /__ / __// _ \ (_-< (_-< / _ \/ _ `// __/
2021-03-29T17:36:11+0200 [Controller  27287]     :::::   :   :::::  \___//_/   \___//___//___//_.__/\_,_//_/
2021-03-29T17:36:11+0200 [Controller  27287]           :::::
2021-03-29T17:36:11+0200 [Controller  27287]     :::::::::::::::::   Crossbar v20.8.1
2021-03-29T17:36:11+0200 [Controller  27287] 
2021-03-29T17:36:11+0200 [Controller  27287]     Copyright (c) 2013-2021 Crossbar.io Technologies GmbH, licensed under AGPL 3.0.
2021-03-29T17:36:11+0200 [Controller  27287] 
2021-03-29T17:36:11+0200 [Controller  27287] Booting standalone node .. <crossbar.node.main._run_command_start>
2021-03-29T17:36:11+0200 [Controller  27287] Node key files exist and are valid. Node public key is 0x5fa1240efc58b3628afd02b751b3095c5498ae943f5d27f015e8d8ce2b1fbfc6
2021-03-29T17:36:11+0200 [Controller  27287] Node key loaded from /path/to/labgrid/.crossbar/key.priv
2021-03-29T17:36:11+0200 [Controller  27287] Node configuration loaded [config_source=localfile, config_path=/path/to/labgrid/.crossbar/config.yaml]
2021-03-29T17:36:11+0200 [Controller  27287] Entering event reactor ...
2021-03-29T17:36:11+0200 [Controller  27287] Starting node .. [<crossbar.node.node.Node.start>]
2021-03-29T17:36:11+0200 [Controller  27287] Node ID cup-27287 set from hostname/pid
2021-03-29T17:36:11+0200 [Controller  27287] RouterFactory.start_realm: router created for realm "crossbar"
2021-03-29T17:36:11+0200 [Controller  27287] <crossbar.node.node.Node._add_global_roles> node-wide role "controller" added on node management router realm "crossbar"
2021-03-29T17:36:11+0200 [Controller  27287] attached session 4942034006234612 to realm "crossbar" (authid="serviceagent", authrole="trusted") <crossbar.router.router.Router.attach>
2021-03-29T17:36:11+0200 [Controller  27287] <crossbar.router.service.RouterServiceAgent.onJoin>: realm service session attached to realm "crossbar" [session_id=4942034006234612, authid="serviceagent", authrole="trusted", on_ready=None]
2021-03-29T17:36:11+0200 [Controller  27287] <crossbar.node.node.Node.start> router service agent session attached [<crossbar.router.service.RouterServiceAgent>]
2021-03-29T17:36:11+0200 [Controller  27287] attached session 1468631257761243 to realm "crossbar" (authid="nodecontroller", authrole="controller") <crossbar.router.router.Router.attach>
2021-03-29T17:36:11+0200 [Controller  27287] <crossbar.node.controller.NodeController.onJoin>: joined realm="crossbar" on local node management router [authid="nodecontroller", authrole="controller"]
2021-03-29T17:36:11+0200 [Controller  27287] Signal handler installed on process 27287 thread 140063715907392
2021-03-29T17:36:11+0200 [Controller  27287] <crossbar.node.node.Node.start> node controller session attached [<crossbar.node.controller.NodeController>]
2021-03-29T17:36:11+0200 [Controller  27287] Using default node shutdown triggers ['shutdown_on_worker_exit']
2021-03-29T17:36:11+0200 [Controller  27287] <crossbar.node.node.Node.boot>::NODE_BOOT_BEGIN
2021-03-29T17:36:11+0200 [Controller  27287] Booting node <crossbar.node.node.Node.boot>
2021-03-29T17:36:11+0200 [Controller  27287] Booting node from local configuration [parallel_worker_start=False] .. <crossbar.node.node.Node.boot_from_config>
2021-03-29T17:36:11+0200 [Controller  27287] Will start 2 workers ..
2021-03-29T17:36:11+0200 [Controller  27287] Order node to start "Router worker001" ..
2021-03-29T17:36:11+0200 [Controller  27287] Starting router-worker "worker001" .. <crossbar.node.controller.NodeController.start_worker>
2021-03-29T17:36:11+0200 [Controller  27287] Unhandled Error
Traceback (most recent call last):
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/crossbar/node/controller.py", line 377, in start_worker
    return self._start_native_worker(worker_type, worker_id, worker_options, details=details)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/crossbar/node/controller.py", line 725, in _start_native_worker
    d = ep.connect(transport_factory)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/crossbar/common/twisted/processutil.py", line 115, in connect
    self._childFDs)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/posixbase.py", line 406, in spawnProcess
    childFDs,
--- <exception caught here> ---
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 732, in __init__
    self.proto.makeConnection(self)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/endpoints.py", line 365, in makeConnection
    return self.protocol.makeConnection(self.transport)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/protocol.py", line 508, in makeConnection
    self.connectionMade()
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/crossbar/node/native.py", line 49, in connectionMade
    WampWebSocketClientProtocol.connectionMade(self)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/autobahn/twisted/websocket.py", line 244, in connectionMade
    self.peer = peer2str(self.transport.getPeer())
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 946, in getPeer
    raise NotImplementedError()
builtins.NotImplementedError: 

2021-03-29T17:36:11+0200 [Controller  27287] worker-specific role "crossbar.worker.worker001" added on node management router realm "crossbar" <crossbar.node.node.Node._add_worker_role>
2021-03-29T17:36:12+0200 [Router      27295] Starting router-worker "worker001" on node "cup-27287" (personality "standalone") and local node management realm "crossbar" .. <crossbar.worker.router.RouterController>
2021-03-29T17:36:12+0200 [Router      27295] Running as PID 27295 on CPython-EPollReactor
2021-03-29T17:36:12+0200 [Router      27295] Entering event reactor ...
2021-03-29T17:36:17+0200 [Router      27295] dropping connection to peer <pipe> with abort=True: WebSocket opening handshake timeout (peer did not finish the opening handshake in time)
2021-03-29T17:36:17+0200 [Router      27295] Connection to node controller closed cleanly
2021-03-29T17:36:17+0200 [Controller  27287] Native worker connection closed cleanly.
2021-03-29T17:36:17+0200 [Controller  27287] unexpected error in processEnded
Traceback (most recent call last):
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 936, in maybeCallProcessEnded
    self.reapProcess()
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 309, in reapProcess
    self.processEnded(status)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/_baseprocess.py", line 55, in processEnded
    self.maybeCallProcessEnded()
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 938, in maybeCallProcessEnded
    _BaseProcess.maybeCallProcessEnded(self)
--- <exception caught here> ---
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/_baseprocess.py", line 66, in maybeCallProcessEnded
    proto.processEnded(Failure(reason))
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/twisted/internet/endpoints.py", line 395, in processEnded
    return self.protocol.connectionLost(Failure(error.ConnectionDone()))
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/crossbar/node/native.py", line 76, in connectionLost
    WampWebSocketClientProtocol.connectionLost(self, reason)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/autobahn/twisted/websocket.py", line 287, in connectionLost
    self._connectionLost(reason)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/autobahn/websocket/protocol.py", line 3412, in _connectionLost
    WebSocketProtocol._connectionLost(self, reason)
  File "/path/to/labgrid/venv/lib/python3.6/site-packages/autobahn/websocket/protocol.py", line 1087, in _connectionLost
    if not self.factory.isServer and self.serverConnectionDropTimeoutCall is not None:
builtins.AttributeError: 'NativeWorkerClientProtocol' object has no attribute 'serverConnectionDropTimeoutCall'
```

The error is gone after cherry-picking e56f8896. Therefore this commit should likely end up in a new bugfix release.

Emantor, 2021-03-30:
Cherry-picked to `stable-0.3` as 86e3c725cce9232b1931c7b57f807aa0112f5dd7, I'll do the next bugfix release on friday, we probably also want several fixes from @Bastian-Krause present on master.

Emantor, 2021-04-07:
I just released 0.3.3, which should fix this issue.

closed 2021-04-07 by Emantor