labgrid-project/labgrid #761: labgrid-exporter does not start ser2net

kjeldflarup, 2021-03-30:
I'm moving from Scratch to Buster on my RPi units. Sametime I also loaded v0.3.2 (and v0.3.0)

I have this exporter yaml which I used to use with 0.1.0
```
pcm52:
  location: pcm52
  USBSerialPort:
    match:
      '@sys_name': '1-1.2'
  NetworkPowerPort:
    model: simplerest
    host: 'http://$LG_EXPORTERIP:9999/gpio/21/{value}'
    index: 0
  NetworkService:
    address: '172.29.5.42'
    username: 'root'
```

When I do a show on the exporter I get this
```
Matching resource 'USBSerialPort' (172.17.120.27/pcm52/NetworkSerialPort/USBSerialPort):
  {'acquired': None,
   'avail': True,
   'cls': 'NetworkSerialPort',
   'params': {'extra': {'path': '/dev/ttyUSB0',
                        'proxy': 'pi013.deif.com',
                        'proxy_required': False},
              'host': '172.17.120.27',
              'port': None,
              'speed': 115200}}
```

Because it has `/dev/ttyUSB0` it means that there is a match on `1-1.2`, but no port is assigned and `ser2net` is not running. 

Trying to debug `labgrid/remote/exporter.py`, I find that `class SerialPortExport` is initialized, but `_start` is never called. 

I can't figure out where in the code this `start()` should be called.

Emantor, 2021-03-30:
Since we are now able to acquire/release resources, the exporter will start ser2net as soon as the place is locked. When the place is unlocked, the exporter will stop ser2net which also neatly allows us to disconnect the user, even if he forgets to close the console connection.

kjeldflarup, 2021-03-31:
@Emantor Does this mean, that the coordinator and the client also needs to be upgraded?

jluebbe, 2021-03-31:
In the [changelog for 0.2.0](https://labgrid.readthedocs.io/en/latest/changes.html#incompatible-changes) we have:
> When using the coordinator, it must be upgraded together with the clients because of the newly introduced match names.

I'd suggest to keep tall the components in sync.

closed 2021-04-06 by kjeldflarup