labgrid-project/labgrid #791: Crossbar fails to start with: 'NativeWorkerClientProtocol' object has no attribute 'serverConnectionDropTimeoutCall'

egrigore, 2021-06-08:
Hello,

Did anyone met this issue lately when trying to start crossbar ? - I just restarted it today - it worked to restart it 2 weeks ago:
`(venv) labgrid@hostname:/labgrid-tests/labgrid$ crossbar start

2021-06-08T11:40:27+0200 [Controller   4119]
2021-06-08T11:40:27+0200 [Controller   4119]     :::::::::::::::::
2021-06-08T11:40:27+0200 [Controller   4119]           :::::          _____                      __
2021-06-08T11:40:27+0200 [Controller   4119]     :::::   :   :::::   / ___/____ ___   ___  ___  / /  ___ _ ____
2021-06-08T11:40:27+0200 [Controller   4119]     :::::::   :::::::  / /__ / __// _ \ (_-< (_-< / _ \/ _ `// __/
2021-06-08T11:40:27+0200 [Controller   4119]     :::::   :   :::::  \___//_/   \___//___//___//_.__/\_,_//_/
2021-06-08T11:40:27+0200 [Controller   4119]           :::::
2021-06-08T11:40:27+0200 [Controller   4119]     :::::::::::::::::   Crossbar v20.8.1
2021-06-08T11:40:27+0200 [Controller   4119]
2021-06-08T11:40:27+0200 [Controller   4119]     Copyright (c) 2013-2021 Crossbar.io Technologies GmbH, licensed under AGPL 3.0.
2021-06-08T11:40:27+0200 [Controller   4119]
2021-06-08T11:40:27+0200 [Controller   4119] Booting standalone node .. <crossbar.node.main._run_command_start>
2021-06-08T11:40:27+0200 [Controller   4119] New node key pair generated! Public key is 0xca0878cde950461c96bb42d21b81661c197e06689fec5bb925c1bab15d9db86f
2021-06-08T11:40:27+0200 [Controller   4119] File permissions on node public key fixed
2021-06-08T11:40:27+0200 [Controller   4119] File permissions on node private key fixed
2021-06-08T11:40:27+0200 [Controller   4119] Node key loaded from /labgrid-tests/labgrid/.crossbar/key.priv
2021-06-08T11:40:27+0200 [Controller   4119] Node configuration loaded [config_source=localfile, config_path=/labgrid-tests/labgrid/.crossbar/config.yaml]
2021-06-08T11:40:27+0200 [Controller   4119] Entering event reactor ...
2021-06-08T11:40:27+0200 [Controller   4119] Starting node .. [<crossbar.node.node.Node.start>]
2021-06-08T11:40:27+0200 [Controller   4119] Node ID coordinator.hostname.com-4119 set from hostname/pid
2021-06-08T11:40:27+0200 [Controller   4119] RouterFactory.start_realm: router created for realm "crossbar"
2021-06-08T11:40:27+0200 [Controller   4119] <crossbar.node.node.Node._add_global_roles> node-wide role "controller" added on node management router realm "crossbar"
2021-06-08T11:40:27+0200 [Controller   4119] attached session 5297033650120967 to realm "crossbar" (authid="serviceagent", authrole="trusted") <crossbar.router.router.Router.attach>
2021-06-08T11:40:27+0200 [Controller   4119] <crossbar.router.service.RouterServiceAgent.onJoin>: realm service session attached to realm "crossbar" [session_id=5297033650120967, authid="serviceagent", authrole="trusted", on_ready=None]
2021-06-08T11:40:27+0200 [Controller   4119] <crossbar.node.node.Node.start> router service agent session attached [<crossbar.router.service.RouterServiceAgent>]
2021-06-08T11:40:27+0200 [Controller   4119] attached session 2183038241498583 to realm "crossbar" (authid="nodecontroller", authrole="controller") <crossbar.router.router.Router.attach>
2021-06-08T11:40:27+0200 [Controller   4119] <crossbar.node.controller.NodeController.onJoin>: joined realm="crossbar" on local node management router [authid="nodecontroller", authrole="controller"]
2021-06-08T11:40:27+0200 [Controller   4119] Signal handler installed on process 4119 thread 140336183330624
2021-06-08T11:40:27+0200 [Controller   4119] <crossbar.node.node.Node.start> node controller session attached [<crossbar.node.controller.NodeController>]
2021-06-08T11:40:27+0200 [Controller   4119] Using default node shutdown triggers ['shutdown_on_worker_exit']
2021-06-08T11:40:27+0200 [Controller   4119] <crossbar.node.node.Node.boot>::NODE_BOOT_BEGIN
2021-06-08T11:40:27+0200 [Controller   4119] Booting node <crossbar.node.node.Node.boot>
2021-06-08T11:40:27+0200 [Controller   4119] Booting node from local configuration [parallel_worker_start=False] .. <crossbar.node.node.Node.boot_from_config>
2021-06-08T11:40:27+0200 [Controller   4119] Will start 2 workers ..
2021-06-08T11:40:27+0200 [Controller   4119] Order node to start "Router worker001" ..
2021-06-08T11:40:27+0200 [Controller   4119] Starting router-worker "worker001" .. <crossbar.node.controller.NodeController.start_worker>
2021-06-08T11:40:27+0200 [Controller   4119] Unhandled Error
Traceback (most recent call last):
  File "/labgrid-tests/venv/lib/python3.6/site-packages/crossbar/node/controller.py", line 377, in start_worker
    return self._start_native_worker(worker_type, worker_id, worker_options, details=details)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/crossbar/node/controller.py", line 725, in _start_native_worker
    d = ep.connect(transport_factory)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/crossbar/common/twisted/processutil.py", line 115, in connect
    self._childFDs)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/posixbase.py", line 406, in spawnProcess
    childFDs,
--- <exception caught here> ---
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 732, in __init__
    self.proto.makeConnection(self)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/endpoints.py", line 365, in makeConnection
    return self.protocol.makeConnection(self.transport)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/protocol.py", line 508, in makeConnection
    self.connectionMade()
  File "/labgrid-tests/venv/lib/python3.6/site-packages/crossbar/node/native.py", line 49, in connectionMade
    WampWebSocketClientProtocol.connectionMade(self)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/autobahn/twisted/websocket.py", line 244, in connectionMade
    self.peer = peer2str(self.transport.getPeer())
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 946, in getPeer
    raise NotImplementedError()
builtins.NotImplementedError:

2021-06-08T11:40:27+0200 [Controller   4119] worker-specific role "crossbar.worker.worker001" added on node management router realm "crossbar" <crossbar.node.node.Node._add_worker_role>

2021-06-08T11:40:28+0200 [Router       4124] Starting router-worker "worker001" on node "coordinator.hostname.com-4119" (personality "standalone") and local node management realm "crossbar" .. <crossbar.worker.router.RouterController>
2021-06-08T11:40:28+0200 [Router       4124] Running as PID 4124 on CPython-EPollReactor
[... 7 lines cut ...]
    self.reapProcess()
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 309, in reapProcess
    self.processEnded(status)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/_baseprocess.py", line 55, in processEnded
    self.maybeCallProcessEnded()
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/process.py", line 938, in maybeCallProcessEnded
    _BaseProcess.maybeCallProcessEnded(self)
--- <exception caught here> ---
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/_baseprocess.py", line 66, in maybeCallProcessEnded
    proto.processEnded(Failure(reason))
  File "/labgrid-tests/venv/lib/python3.6/site-packages/twisted/internet/endpoints.py", line 395, in processEnded
    return self.protocol.connectionLost(Failure(error.ConnectionDone()))
  File "/labgrid-tests/venv/lib/python3.6/site-packages/crossbar/node/native.py", line 76, in connectionLost
    WampWebSocketClientProtocol.connectionLost(self, reason)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/autobahn/twisted/websocket.py", line 287, in connectionLost
    self._connectionLost(reason)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/autobahn/websocket/protocol.py", line 3412, in _connectionLost
    WebSocketProtocol._connectionLost(self, reason)
  File "/labgrid-tests/venv/lib/python3.6/site-packages/autobahn/websocket/protocol.py", line 1087, in _connectionLost
    if not self.factory.isServer and self.serverConnectionDropTimeoutCall is not None:
builtins.AttributeError: 'NativeWorkerClientProtocol' object has no attribute 'serverConnectionDropTimeoutCall'

^C2021-06-08T11:40:42+0200 [Controller   4119] Controller received SIGINT [signal=2]: shutting down node [shutdown_was_clean=True] ..
2021-06-08T11:40:42+0200 [Controller   4119] Node shutdown requested (restart=False, mode=None, reactor.running=True) ..
`

jluebbe, 2021-06-08:
Did you install labgrid using the requirements-* files from git? Those should fix the versions of all dependencies.

egrigore, 2021-06-08:
Indeed it worked after I have updated all the *requirements.txt files with latest from git. Thank you

closed 2021-06-08 by egrigore