labgrid-project/labgrid #794: Storing passwords in a secure manner

initBasti, 2021-07-07:
Hello,

I just wanted to quickly check if I missed something. Within the getting started section, I set up an initial [local. yaml](https://labgrid.readthedocs.io/en/latest/getting_started.html#running-your-first-test) for the first test and I am expected to change the port, username, and password. I have searched through all search results for the word password in the documentation and within the GitHub issues, but I was not able to find any hint on how to securely manage passwords. I was expecting something like a keyring, gpg encrypted file, or some connection to a password manager, but it seems like I should use a plain text password in the `.yaml`.
Am I wrong or could somebody explain to me the current best practices for storing passwords for different boards?

Greetings,
Sebastian

jluebbe, 2021-07-07:
There is no support for encrypting passwords. The expectation is that images for testing would have well known passwords (and SSH keys and other credentials). labgrid itself also has no authentication for access to the coordinator, so it's intended to be used in a trusted environment.

initBasti, 2021-07-07:
Hey @jluebbe,

Thank you for your response, so in order to secure my test setup, I will have to make sure that the master device, which coordinates the work onto the devices under test, cannot be penetrated, right? I was just wondering because I would like to host a CI system on the master device and connect it to multiple test devices (for camera quality testing) and I thought that if an attacker is able to get into the master device he can then easily access all of the prepared test devices. Maybe I have a use case, which does not fit the design of labgrid.

Thank you.
Sebastian

jluebbe, 2021-07-07:
We're currently using a separate VLAN for the labs and only specific systems have access to that network (specific CI runners and developer hosts). That lab network also contains other network devices (power switches, serial servers, ...) that don't have any authentication, so it needs to be relatively protected in any case.

initBasti, 2021-07-07:
Thank you very much for giving me an example, I believe it would be awesome if you could link to a source describing how to properly set up a secure lab in the documentation or if the documentation would contain an example setup. Do you also think this would help or would that be above the scope of labgrid?

jluebbe, 2021-07-08:
I'd think that configuring the Linux hosts and routers in a test-lab setup would be out of scope for labgrid's documentation, as this is a complex topic and also applies to any lab network (with or without labgrid). If you'd like design support or review for your specific project setup, my employer (Pengutronix) can offer consulting.

initBasti, 2021-07-08:
Thank you for the hint but I am a poor developer using my own time and money to build a test lab for an open-source project, so this is out of my scope :) I'll try to figure it out and maybe I can contribute to your project as well.

closed 2021-07-08 by initBasti

jluebbe, 2021-07-08:
@initBasti You could also join our IRC/Matrix channel. We try to help with questions as time allows, and there are also other community members who may have relevant suggestions.