labgrid-project/labgrid #824: Username and Hostname with / break labgrid usage

MatthiasWeidmann, 2021-09-03:
Hi,

**Setup**
I'm running labgrid-cooridnator with a self build docker image which uses the master branch
https://github.com/labgrid-project/labgrid/commit/41b297faba60f32e4986678554af32dbff83ce98
I checked this by running `git logs` on `/opt/labgrid` within the container.

For connecting to the coordinator I'm using the official image from hub.docker.com:
labgrid/client@sha256:c2808c805d6e14505b5050a1cb493ea1e4ecad81bbde8ce5aaadab0d6cdb8667

**Description**

If one of the environment variables LG_USERNAME or LG_HOSTNAME contain a `/`, I'm not able to unlock a reserved place (`labgrid-client -p + unlock`), nor show the current usage (`labgrid-client who`).

**Procedure**

```
# export LG_USERNAME="user/123"
# export LG_HOSTNAME="host/123"
# labgrid-client places
  p1
# labgrid-client -v places
Place 'p1':
  tags: test=123
  matches:
  acquired: None
  acquired resources:
  created: 2021-08-17 14:54:11.289673
  changed: 2021-09-03 09:15:49.538910
# labgrid-client reserve --shell test=123 > labgrid-token.sh
# source ./labgrid-token.sh
# labgrid-client wait
owner: host/123/user/123
token: SMILMTLDAN
state: allocated
filters:
  main: test=123
allocations:
  main: p1
created: 2021-09-03 09:34:41.672340
timeout: 2021-09-03 09:36:01.812984
# labgrid-client -p + lock
acquired place p1
# labgrid-client -p + show
Place 'p1':
  tags: test=123
  matches:
  acquired: host/123/user/123
  acquired resources:
  created: 2021-08-17 14:54:11.289673
  changed: 2021-09-03 09:37:09.926147
  reservation: SMILMTLDAN
```

**Errors**
```
# labgrid-client who
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.1.dev1+g41b297f-py3.7.egg/labgrid/remote/client.py", line 1843, in main
    args.func(session)
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.1.dev1+g41b297f-py3.7.egg/labgrid/remote/client.py", line 272, in print_who
    host, user = place.acquired.split('/')
ValueError: too many values to unpack (expected 2)

# labgrid-client -p + unlock
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.1.dev1+g41b297f-py3.7.egg/labgrid/remote/client.py", line 1841, in main
    session.loop.run_until_complete(coro)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.7/dist-packages/labgrid-0.1.dev1+g41b297f-py3.7.egg/labgrid/remote/client.py", line 629, in release
    _, user = place.acquired.split('/')
ValueError: too many values to unpack (expected 2)
```

**Personal Conclusion**

From what I understand, a `/` should not be allowed in the username or hostname.

jluebbe, 2021-09-03:
As POSIX host- or usernames are not allowed to contain ``/`` (and many other characters), there are likely more places in labgrid that would fail, so we can't easily change that.
I agree that we should check that more strictly.

MatthiasWeidmann, 2021-09-03:
@jluebbe : I have added a pull request, but I'm not that familiar with python and your code.
I tried checking the tests, but it seems that the test is skipped?
Could you please provide me guidance how to test my pull request?

```
git clone https://github.com/MatthiasWeidmann/labgrid.git
git checkout patch-1
cd labgrid
virtualenv -p python3 venv
source venv/bin/activate
pip install -r requirements.txt
python setup.py install
python -m pytest -vvv tests/test_remote_client.py 
```

Output:
```
====================================================================== test session starts =======================================================================
platform darwin -- Python 3.9.6, pytest-6.1.2, py-1.10.0, pluggy-0.13.1 -- /Users/mw/git/github.com/matthiasweidmann/labgrid/venv/bin/python
cachedir: .pytest_cache
rootdir: /Users/mw/git/github.com/matthiasweidmann/labgrid, configfile: setup.cfg
collected 0 items / 1 skipped                                                                                                                                    

======================================================================= 1 skipped in 0.01s =======================================================================
```

Bastian-Krause, 2022-03-16:
@MatthiasWeidmann Try: `python -m pytest -vvv -ra tests/test_remote_client.py`. This should show you the reason why the tests are skipped. Also please follow the [installation instructions](https://labgrid.readthedocs.io/en/latest/getting_started.html#installation) and use `dev-requirements.txt` instead of `requirements.txt`. You should end up with everything you need to run the tests.

Emantor, 2026-10-10:
With the `MetadataIdentity` this should now work properly, please comment and we will reopen if it does not.

closed 2026-10-10 by Emantor