labgrid-project/labgrid #868: Difference of concepts related to proxy.

cs2002, 2021-11-15:
As title, could you teach me the difference of next 3 concepts:

1. LG_PROXY=xxx
2. labgrid-client -P xxx
3. labgrid-exporter -i

Looks we could use above 3 methods to access the remote USB devices, what's the difference? I'm lost here on which to choose, could you kindly give me the scenario for above 3 methods? Thanks!

Bastian-Krause, 2021-11-15:
`LG_PROXY` and `-P`/`--proxy` are equivalent.

The idea behind proxying is desribed [here](https://labgrid.readthedocs.io/en/latest/overview.html#proxy-mechanism). The [exporter man-page](https://labgrid.readthedocs.io/en/latest/man/exporter.html#i-isolated) holds additional information.

After reading this, is this still unclear?

cs2002, 2021-11-15:
Thanks, but the doc is too abstract to understand. 
In fact, from experiment, I didn't see big difference:

1) It looks Make a resouce isolated just let the client force to use a proxy which specified by exporter, usually the exporter name.
2) But even not use isolated, user still could use `labgrid-client -P proxy` to make the same effect, then why isolated here? 

The small difference I noticed is: use isolated, user looks no need clearly know what the proxy is. Is this the main difference?

But with isolated, all resource in one place will all force use proxy, even some resource need proxy while some resource not need.
E.g. One resource is `RawSerial` which will use ser2net, so labgrid-client no need proxy to access it.
While another resource is `USB` which we have to use ssh proxy to access it, they are in one place.
Will it bring any efficiency issue for `RawSerial` resource when we use `isolated`? Because in fact we no need that proxy.
Or we let user specify `-P` when access `USB related function`, while let user not specify `-P` when access `Serial Port`?

What I need is a suggestion for above scenario, choose isolated or let user specify `-P`?

Bastian-Krause, 2021-11-15:
> Thanks, but the doc is too abstract to understand. In fact, from experiment, I didn't see big difference:
>
> 1. It looks Make a resouce isolated just let the client force to use a proxy which [...]

Running the exporter in "isolated mode is useful when resources [...] are not directly accessible from the clients" [[1]](https://labgrid.readthedocs.io/en/latest/man/exporter.html#i-isolated). If that's not the case in your scenario, do not enable isolated mode. This is only required if you have some kind of network separation set up.

If you cannot access the coordinator/exporter directly, use `--proxy` / `-P` / `LG_PROXY`.

> But with isolated, all resource in one place will all force use proxy, even some resource need proxy while some resource not need. E.g. One resource is `RawSerial` which will use ser2net, so [...]

Why do you need a proxy for USB resources? Did you try that without a proxy? Please post your exporter configuration.

Do not use a proxy unless strictly needed.