labgrid-project/labgrid #887: Check if pysnmp-mib is still required for SNMP MIB conversion

Emantor, 2022-01-27:
The pypi pysnmp-mibs package (https://pypi.org/project/pysnmp-mibs/) indicated that pysnmp can do the conversion using pysmi, check if this works within labgrid and remove pysnmp-mibs if it is no longer required.

jluebbe, 2022-01-27:
My concerns are whether this needs a writable rootfs and how this affects startup time. It seems that pre-converted MIBs would solve both.

Emantor, 2022-01-27:
Thanks for the info. I was a bit worried since pysnmp-mibs hasn't been updated in a while, on the other hand these are just precompiled MIBs. Do you still think its worth to look into this?

jluebbe, 2022-01-27:
These basic mibs don't change often, so as long as it's compatible with pysnmp, I'm not worried.

Emantor, 2022-01-27:
Aye, closing this.

closed 2022-01-27 by Emantor