labgrid-project/labgrid #889: Option for acquiring coordinator / exporter consoles for labgrid-client

vishwa-mke, 2022-02-04:
Is there a way to acquire exporters / coordinator's console from the labgrid-client ? 

If yes how and if not can you give me hints on how to implement it, for e.g. as a strategy where each of the coordinator, exporters and targets are treated as levels or states

This option might be valuable in some situations for example to run nmap tests against the target during the various modes of operation of the target device.

Bastian-Krause, 2022-03-08:
Hi,

you could do something like this, assuming you want to nmap your target from the machine exporting your `NetworkSerialPort` and your environment configuration contains a [NetworkService](https://labgrid.readthedocs.io/en/latest/configuration.html#networkservice).

```python
from labgrid.util.helper import processwrapper

def test_nmap(target):
    serial_resource = target.get_resource("NetworkSerialPort")
    network_resource = target.get_resource("NetworkService")

    cmd = ["nmap", network_resource.address]
    out = processwrapper.check_output(serial_resource.command_prefix + cmd)
```

Maybe @Emantor or @jluebbe have a better idea?

Bastian-Krause, 2023-07-23:
@vishwa-mke Did you test my suggestion?

vishwa-mke, 2023-07-23:
Hello Bastian,

Thank you for informing me, I will ask my colleague to test it in the next week's and let you know.

Best regards,
Vishwavardhan

Bastian-Krause, 2023-10-20:
We haven't heard back. I'll assume that the suggestion worked and close the issue. Feel free to comment if it does not work.

closed 2023-10-20 by Bastian-Krause