labgrid-project/labgrid #908: BUG: possible race condition for place release.

atline, 2022-04-14:
I'm not sure how that happen, but now I get place as next:

```
atline@NXL66538:~$ labgrid-client -p imx8mp-evk-sz01 show
Place 'imx8mp-evk-sz01':
  matches:
    */imx8mp-evk-sz01/*
  acquired: None
  acquired resources:
  created: 2022-03-03 10:38:09.334816
  changed: 2022-04-13 15:22:10.992044
Matching resource 'NetworkSerialPort' (TigerII/imx8mp-evk-sz01/NetworkSerialPort/NetworkSerialPort):
  {'acquired': None,
   'avail': True,
   'cls': 'NetworkSerialPort',
   'params': {'extra': {'proxy': 'sz-farm', 'proxy_required': True},
              'host': '192.168.100.106',
              'port': 10021}}
Matching resource 'PDUDaemonPort' (TigerII/imx8mp-evk-sz01/PDUDaemonPort/PDUDaemonPort):
  {'acquired': 'imx8mp-evk-sz01',
   'avail': True,
   'cls': 'PDUDaemonPort',
   'params': {'extra': {'proxy': 'sz-farm', 'proxy_required': True},
              'host': '192.168.100.127',
              'index': 7,
              'pdu': '192.168.100.111'}}
```

You could see `acquired: None` for place `imx8mp-evk-sz01`, and `NetworkSerialPort` is also `acquired: None`, but `PDUDaemonPort` is set as `acquired': 'imx8mp-evk-sz01`. How could that happen?

Q1: I see there is a error log in exporter:

```
2022-04-13T15:22:27 ExporterSession.onUserError(): "CancelledError: "
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/txaio/aio.py", line 484, in done
    res = f.result()
concurrent.futures._base.CancelledError
```
At which situation will above error pop up?

Q2: When above happen, if possible I could use some method to recover that, better not restart exporter?

Bastian-Krause, 2022-05-16:
Wild guess: maybe you locked the place while your exporter on `TigerII` was still in the process of iterating over the resources being exported?

Having explicit matches of the resources you expect would probably prevent such issues in the future. labgrid would only allow you to lock the place if all resources are exported. This obviously does not work with matches such as `*/imx8mp-evk-sz01/*`.

atline, 2022-07-08:
In fact, the exporter already running there for several weeks, so what your guess is not the case.

Anyway, I didn't see it again after 2 month's observation, so I prefer to close it. And reopen it if I see a second time.

closed 2022-07-08 by atline