labgrid-project/labgrid #919: Can't setup crossbar in virtualenv due to dependency issues

NiklasReisser, 2022-05-19:
Hi,

We're currently trying to get a labgrid instance going. I followed the instructions here:
https://labgrid.readthedocs.io/en/latest/getting_started.html#coordinator

But I run into this error: 
```
ERROR: crossbar 21.3.1 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.26.9 which is incompatible.
```

Explicitly adding 
`urllib3==1.24.3` to `requirements.txt` helps, but only until I try to start crossbar: 

```
$ crossbar start

Traceback (most recent call last):
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/bin/crossbar", line 8, in <module>
    sys.exit(run())
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/crossbar/__init__.py", line 175, in run
    _personalities = personalities()
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/crossbar/__init__.py", line 213, in personalities
    from crossbar import personality as standalone
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/crossbar/personality.py", line 41, in <module>
    from crossbar.worker.router import RouterController
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/crossbar/worker/router.py", line 30, in <module>
    from crossbar.worker.transport import TransportController
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/crossbar/worker/transport.py", line 49, in <module>
    from crossbar.bridge.mqtt.wamp import WampMQTTServerFactory
  File "/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/crossbar/bridge/mqtt/wamp.py", line 52, in <module>
    from autobahn.wamp.serializer import JsonObjectSerializer, MsgPackObjectSerializer, CBORObjectSerializer, UBJSONObjectSerializer
ImportError: cannot import name 'MsgPackObjectSerializer' from 'autobahn.wamp.serializer' (/home/i012914@claas.local/projects/labgrid-crossbar-venv/lib/python3.8/site-packages/autobahn/wamp/serializer.py)
```

For which I had to add `msgpack` to `crossbar-requirements.txt`

Now everything appears to work fine. I'm happy to work on a proper fix, but I wanted to ask first if it could be a problem on my side.

Full log for the first problem: 
```
$ pip install -r crossbar-requirements.txt
Collecting attrs==21.2.0
  Using cached attrs-21.2.0-py2.py3-none-any.whl (53 kB)
Collecting jinja2==3.0.2
  Using cached Jinja2-3.0.2-py3-none-any.whl (133 kB)
Collecting packaging==21.0
  Using cached packaging-21.0-py3-none-any.whl (40 kB)
Collecting pexpect==4.8.0
  Using cached pexpect-4.8.0-py2.py3-none-any.whl (59 kB)
Collecting pyserial
  Using cached https://github.com/labgrid-project/pyserial/archive/v3.4.0.1.zip
Collecting pytest==6.2.5
  Using cached pytest-6.2.5-py3-none-any.whl (280 kB)
Processing /home/i012914@claas.local/.cache/pip/wheels/8b/d1/11/6a9855487e4bba54c63dfb07a09505787d50a5e27f90b0af33/pyudev-0.22.0-py3-none-any.whl
Collecting requests==2.26.0
  Using cached requests-2.26.0-py2.py3-none-any.whl (62 kB)
Processing /home/i012914@claas.local/.cache/pip/wheels/8a/46/14/833413574281b7009c9180fce7c595a7cc1b538e43fcd8b7e7/xmodem-0.4.6-py3-none-any.whl
Collecting autobahn==21.3.1
  Using cached autobahn-21.3.1-py2.py3-none-any.whl (495 kB)
Collecting PyYAML==5.4.1
  Using cached PyYAML-5.4.1-cp38-cp38-manylinux1_x86_64.whl (662 kB)
Collecting ansicolors==1.1.8
  Using cached ansicolors-1.1.8-py2.py3-none-any.whl (13 kB)
Collecting pyusb==1.2.1
  Using cached pyusb-1.2.1-py3-none-any.whl (58 kB)
Collecting six>=1.13.0
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Requirement already satisfied: setuptools>=38.0.0 in /home/i012914@claas.local/projects/labgrid-crossbar-venv-2/lib/python3.8/site-packages (from -r crossbar-requirements.txt (line 2)) (44.0.0)
Collecting crossbar==21.3.1
  Using cached crossbar-21.3.1-py2.py3-none-any.whl (524 kB)
[... 97 lines cut ...]
Collecting Twisted[tls]>=15.5.0
  Using cached Twisted-22.4.0-py3-none-any.whl (3.1 MB)
Collecting automat
  Using cached Automat-20.2.0-py2.py3-none-any.whl (31 kB)
Collecting cbor2>=5.2.0
  Using cached cbor2-5.4.3-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (235 kB)
Collecting numpy>=1.20.1
  Using cached numpy-1.22.3-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.8 MB)
Collecting flatbuffers>=2.0
  Using cached flatbuffers-2.0-py2.py3-none-any.whl (26 kB)
Collecting zipp>=3.1.0; python_version < "3.10"
  Using cached zipp-3.8.0-py3-none-any.whl (5.4 kB)
Collecting pycparser
  Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
Collecting typing-extensions>=3.6.5
  Using cached typing_extensions-4.2.0-py3-none-any.whl (24 kB)
Building wheels for collected packages: pyserial
  Building wheel for pyserial (setup.py) ... done
  Created wheel for pyserial: filename=pyserial-3.4.0.1-py2.py3-none-any.whl size=93262 sha256=bb708890708be996308fb452b087357fabd6599fed39a46b5040ee4c7518b407
  Stored in directory: /tmp/pip-ephem-wheel-cache-txpg071o/wheels/86/ec/74/2eeccd8eb95c8b3f16be81b39f88c4bfd2a4aacfdd126b8b6c
Successfully built pyserial
ERROR: crossbar 21.3.1 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.26.9 which is incompatible.
Installing collected packages: attrs, MarkupSafe, jinja2, pyparsing, packaging, ptyprocess, pexpect, pyserial, toml, pluggy, iniconfig, py, pytest, six, pyudev, charset-normalizer, certifi, idna, urllib3, requests, xmodem, pycparser, cffi, cryptography, hyperlink, txaio, autobahn, PyYAML, ansicolors, pyusb, bitstring, zope.interface, colorama, psutil, sortedcontainers, pytrie, hyperframe, hpack, h2, pyasn1, pyasn1-modules, service-identity, lmdb, u-msgpack-python, pygments, watchdog, constantly, passlib, pyqrcode, pynacl, werkzeug, incremental, typing-extensions, automat, pyopenssl, Twisted, txtorcon, cbor2, numpy, click, flatbuffers, zlmdb, priority, sdnotify, mistune, setproctitle, bcrypt, cbor, treq, netaddr, zipp, importlib-resources, py-ubjson, crossbar
Successfully installed MarkupSafe-2.1.1 PyYAML-5.4.1 Twisted-22.4.0 ansicolors-1.1.8 attrs-21.2.0 autobahn-21.3.1 automat-20.2.0 bcrypt-3.2.2 bitstring-3.1.9 cbor-1.0.0 cbor2-5.4.3 certifi-2022.5.18 cffi-1.15.0 charset-normalizer-2.0.12 click-8.1.3 colorama-0.4.4 constantly-15.1.0 crossbar-21.3.1 cryptography-37.0.2 flatbuffers-2.0 h2-4.1.0 hpack-4.0.0 hyperframe-6.0.1 hyperlink-21.0.0 idna-2.5 importlib-resources-5.7.1 incremental-21.3.0 iniconfig-1.1.1 jinja2-3.0.2 lmdb-1.3.0 mistune-2.0.2 netaddr-0.8.0 numpy-1.22.3 packaging-21.0 passlib-1.7.4 pexpect-4.8.0 pluggy-1.0.0 priority-2.0.0 psutil-5.9.0 ptyprocess-0.7.0 py-1.11.0 py-ubjson-0.16.1 pyasn1-0.4.8 pyasn1-modules-0.2.8 pycparser-2.21 pygments-2.12.0 pynacl-1.5.0 pyopenssl-22.0.0 pyparsing-3.0.9 pyqrcode-1.2.1 pyserial-3.4.0.1 pytest-6.2.5 pytrie-0.4.0 pyudev-0.22.0 pyusb-1.2.1 requests-2.26.0 sdnotify-0.3.2 service-identity-21.1.0 setproctitle-1.2.3 six-1.16.0 sortedcontainers-2.4.0 toml-0.10.2 treq-22.2.0 txaio-22.2.1 txtorcon-22.0.0 typing-extensions-4.2.0 u-msgpack-python-2.7.1 urllib3-1.26.9 watchdog-2.1.8 werkzeug-2.0.3 xmodem-0.4.6 zipp-3.8.0 zlmdb-22.4.1 zope.interface-5.4.0
```

Emantor, 2022-05-19:
Which distribution and pip version is this? Looks like the resolver is not constrained to choose the new version, but selects it due to it being the cached version.

NiklasReisser, 2022-05-20:
Hi, 

thanks for the fast reply!
Adding `--no-cache-dir` doesn't help. 

Here are the versions: 
```
$ lsb_release -s -d
Ubuntu 20.04.4 LTS
$ python --version
Python 3.8.10
$ pip --version
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
```

Bastian-Krause, 2022-05-20:
I can reproduce your error..
```
ERROR: crossbar 21.3.1 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.26.9 which is incompatible.
```
..with `pip==20.0.2`.

Please try upgrading pip in your virtualenv before installing the requirements in `crossbar-requirements.txt`:
```
pip install --upgrade pip
```
I don't run into this with `pip==20.3.4`.

NiklasReisser, 2022-05-20:
You're right, after upgrading pip everything works as expected.

Bastian-Krause, 2022-05-20:
I think we should adjust the documentation, preferably wherever a virtualenv is set up.

We do the same in CI for quite some time:
https://github.com/labgrid-project/labgrid/blob/792fa10e765a15f8cff9af4d0af6201ce15ce4ef/.github/workflows/unit-tests.yml#L40

PR #922 (2022-05-30): Add instructions to update pip when setting up a virtualenv for labgrid

**Description**

Extend Documentation to update pip when creating a new virtualenv. The dependency resolver of older pip versions (at least until 2.0.2, as shipped with Ubuntu 20.04) is not able to install the current set of required versions. 

Fixes #919
closed 2022-05-30 by Emantor