labgrid-project/labgrid #923: AndroidFastboot: exporter never detects device

khilman, 2022-05-26:
I'm following the documentation to add an `AndroidFastboot` match to my exporter config like this:

```
mt8173-evb:
  location: lab-seattle
  USBSerialPort:
    speed: 921600
    match:
      '@sys_name': '1-1.3.3'

  AndroidFastboot:
    match:
      sys_name: '1-1.2.1'
```

On the same host running the exporter, I'm able to see the device using normal `fastboot` commands as a non-root user e.g.
```
$ fastboot devices -l
0123456789ABCDEF       fastboot usb:1-1.2.1
```
however, labgrid does does not think `AndroidFastboot` is available (but it has found the `USBSerialPort`)
```
$ labgrid-client -p mt8173-evb show
Place 'mt8173-evb':
  matches:
    */mt8173-evb/*
  acquired: nuc-1/khilman
  acquired resources:
    nuc-1/mt8173-evb/NetworkAndroidFastboot/AndroidFastboot
    nuc-1/mt8173-evb/NetworkSerialPort/USBSerialPort
  created: 2022-05-04 20:04:44.052861
  changed: 2022-05-26 15:28:45.918156
Acquired resource 'USBSerialPort' (nuc-1/mt8173-evb/NetworkSerialPort/USBSerialPort):
  {'acquired': 'mt8173-evb',
   'avail': True,
   'cls': 'NetworkSerialPort',
   'params': {'extra': {'path': '/dev/ttyUSB16',
                        'proxy': 'nuc-1.lan',
                        'proxy_required': False},
              'host': 'nuc-1',
              'port': 39863,
              'speed': 921600}}
Acquired resource 'AndroidFastboot' (nuc-1/mt8173-evb/NetworkAndroidFastboot/AndroidFastboot):
  {'acquired': 'mt8173-evb',
   'avail': False,
   'cls': 'NetworkAndroidFastboot',
   'params': {'busnum': None,
              'devnum': None,
              'extra': {'proxy': 'nuc-1.lan', 'proxy_required': False},
              'host': 'nuc-1',
              'model_id': None,
              'path': None,
              'vendor_id': None}}
```

The kernel messages on this host show the fastboot USB device appearing normally:
```
May 26 15:44:08 nuc-1 kernel: [4460533.673582] usb 1-1.2.1: new high-speed USB device number 41 using xhci_hcd
May 26 15:44:08 nuc-1 kernel: [4460533.774669] usb 1-1.2.1: New USB device found, idVendor=0bb4, idProduct=0c01, bcdDevice= 1.00
May 26 15:44:08 nuc-1 kernel: [4460533.774677] usb 1-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 26 15:44:08 nuc-1 kernel: [4460533.774682] usb 1-1.2.1: Product: Android
May 26 15:44:08 nuc-1 kernel: [4460533.774686] usb 1-1.2.1: Manufacturer: MediaTek
May 26 15:44:08 nuc-1 kernel: [4460533.774690] usb 1-1.2.1: SerialNumber: 0123456789ABCDEF
```

So I'm not sure what's wrong with my match.

khilman, 2022-05-26:
Ah, the `AndroidFastboot` udev class assumes a default vendor and product id.  from udev.py:
```
class AndroidFastboot(USBResource):
    usb_vendor_id = attr.ib(default='1d6b', validator=attr.validators.instance_of(str))
    usb_product_id = attr.ib(default='0104', validator=attr.validators.instance_of(str))
```
I'm not sure how standard it is for fastboot devices to have the same VID/PID.

But, if I update my `AndroidFastboot` config to look like this:
```
  AndroidFastboot:
    match:
      sys_name: 1-1.2.1
    usb_vendor_id: '0bb4'
    usb_product_id: '0c01'
```
then `labgrid-client show` shows my fastboot device as `'avail': True` and I can then use `labgrid-client -p <place> fastboot <cmd>` successfully.

So looks like the docs need an update for `AndroidFastboot` to suggest using the USB vendor/product IDs.

khilman, 2022-05-26:
One addition nit is that `labgrid-client show` shows the vendor/product IDs as decimal rather than the standard hex.

Emantor, 2022-05-27:
> One addition nit is that `labgrid-client show` shows the vendor/product IDs as decimal rather than the standard hex.

There are no additional types attached to the resource parameters while passing it to the show command, so the current architecture does not support displaying those as hex. I can understand why that is confusing, but this requires either reworking the system with additional type information or special casing the various entries which require a hex representation.

PR #924 (2022-05-27): doc: configuration: add missing AndroidFastboot arguments
**Description**
Add the AndroidFastboot arguments `usb_vendor_id` and `usb_product_id` to the docs.

**Checklist**
- [x] The arguments and description in doc/configuration.rst have been updated

Fixes #923
closed 2022-05-27 by Emantor