labgrid-project/labgrid #946: ShellDriver noisy systemctrl command

danielschickmunit, 2022-06-28:
Hi,

I am running command.run("systemctl restart systemd-networkd") which is a rather noisy:

```
root@lab03014:~# MARKER='LVGI''AJRRBK' run 'systemctl restart systemd-networkd'
LVGIAJRRBK         Stopping Network Service...
[FAILED] Failed to start Configure …ooth Modems connected by UART.
See 'systemctl status hciuart.service' for details.
[  OK  ] Stopped Network Service.
         Starting Network Service...
[  OK  ] Started Network Service.
LVGIAJRRBK 0
[  OK  ] Started Wait for Network to be Configured.
[  OK  ] Reached target Network is Online.
[  OK  ] Started deliveryoptimizati…t delivery optimization tasks.
[  OK  ] Started Device Update Agent daemon..
         Starting Docker Application Container Engine...
root@lab03014:~# [  OK  ] Started Docker Application Container Engine.
[  OK  ] Started Azure IoT Edge daemon.
[  OK  ] Reached target Multi-User System.
         Starting Update UTMP about System Runlevel Changes...
         Starting watchdog daemon...
[  OK  ] Started Update UTMP about System Runlevel Changes.
[FAILED] Failed to start watchdog daemon.
See 'systemctl status watchdog.service' for details.
```
the issue here is that the run command fails beacuse the regex 
```regex
LVGIAJRRBK(.*)LVGIAJRRBK\s+(\d+)\s+root@[^ ]+:[^ ]+ 
```
is not being matched, is there some way to alter the regex via the function call to sth like this?

```regex
LVGIAJRRBK(.*)LVGIAJRRBK\s+(\d)(.*)root@[^ ]+:[^ ]+ 
```

I'm a bit unsure what would be best practice here

for now I'll use 
```python
command.console.sendline("systemctl restart systemd-networkd")
```
instead of
```python
command.run("systemctl restart systemd-networkd")
```

Bastian-Krause, 2022-06-29:
Hi,

for testing purposes, systemd's output should not end up in the console. This confuses the ShellDriver. For the same reason, the ShellDriver tries to silence the kernel log.

You're probably using something like `systemd.journald.forward_to_console=1` on the kernel cmdline. Please disable that.

danielschickmunit, 2022-07-01:
thanks, this fixes the issue

in my case I had to set `ForwardToWall=yes` to `ForwardToWall=no`

closed 2022-07-01 by danielschickmunit

danielschickmunit, 2022-08-23:
if sombody stumbles into the same issue, I had to disable the ForwardToWall in the compilation of journalctl and had to add two kernel command line parameters in order to completly get rid of these messages. Those were `quiet` and `systemd.show_status=0`