labgrid-project/labgrid #963: setup: running `python3 setup install` failed with connection error

smrtos, 2022-08-07:
I am setting up labgrid on Ubuntu in corp net behind a proxy 
I have setup below environment variables:
```
http_proxy=http://xxx:port
https_proxy=http://xxx:port
HTTP_PROXY=http://xxx:port
HTTPS_PROXY=http://xxx:port
```
And run below command according to the [doc](https://labgrid.readthedocs.io/en/latest/getting_started.html#installation):

```
sudo python3 setup.py install
```

But always met below error:
```
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7fd379ecf490>: Failed to establish a new connection: [Errno 101] Network is unreachable')': /simple/setuptools-scm/
```

How to fix this? Thanks!

Bastian-Krause, 2022-08-07:
You could try installing `setuptools_scm` via pip before running `setup.py install` (you shouldn't need to run this with sudo):
```
labgrid-venv $ pip install setuptools_scm
```

smrtos, 2022-08-08:
> 

> You could try installing `setuptools_scm` via pip before running `setup.py install` (you shouldn't need to run this with sudo):
> 
> ```
> labgrid-venv $ pip install setuptools_scm
> ```

I have to use the `sudo` to suppress below error:
```

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/usr/lib/python3.8/site-packages/test-easy-install-2666845.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /usr/lib/python3.8/site-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.
```

And I did try to install the ``setuptools_scm` first. It is ready I think:
```
xxx@xxx-box01:~/github/labgrid$ pip3 install setuptools-scm
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: setuptools-scm in /home/xxx/.local/lib/python3.8/site-packages (6.3.2)
Requirement already satisfied: packaging>=20.0 in /home/xxx/.local/lib/python3.8/site-packages (from setuptools-scm) (21.0)
Requirement already satisfied: setuptools in /home/xxx/.local/lib/python3.8/site-packages (from setuptools-scm) (63.4.1)
Requirement already satisfied: tomli>=1.0.0 in /home/xxx/.local/lib/python3.8/site-packages (from setuptools-scm) (1.2.2)
Requirement already satisfied: pyparsing>=2.0.2 in /home/xxx/.local/lib/python3.8/site-packages (from packaging>=20.0->setuptools-scm) (2.4.7)
```

But still the same error.

Bastian-Krause, 2022-08-09:
> I have to use the `sudo` to suppress below error:
>
> ```
>
> The following error occurred while trying to add or remove files in the
> installation directory:
>
>     [Errno 13] Permission denied: [...]

You are not using the recommended way of installation (virtualenv). Please use htat.

> And I did try to install the ``setuptools_scm` first. It is ready I think:
>
> ```
> xxx@xxx-box01:~/github/labgrid$ pip3 install setuptools-scm
> Defaulting to user installation because normal [...]

You are using a mixture of user installation (`pip`) and system-wide installation (`sudo python3 setup.py install`). You can't expect a "user installed" package to be visible in the system installation. Please stick to the recommended installation via virtualenv, see [our docs](https://labgrid.readthedocs.io/en/latest/getting_started.html#installation).

Bastian-Krause, 2022-08-31:
I assume the issue is solved if you stick to the recommended installation instructions. If not, please reopen.