labgrid-project/labgrid #980: AssertionError in step.py when running multiple SSHDrivers in different threads

S4mw1s3, 2022-09-01:
Hi,

I'm using `labgrid==0.4.1` and I want to query multiple devices in parallel by using the SSHDriver. The following script gives me an AssertionError in step.py though:

```
import logging
import threading
import argparse

from labgrid import Target
from labgrid.resource import NetworkService
from labgrid.driver import SSHDriver

parser = argparse.ArgumentParser()
parser.add_argument('user')
parser.add_argument('passwd')
args = parser.parse_args()

def query_params(ip):
    print(f"Checking {ip}")

    t = Target('DUT')
    nw_service = NetworkService(t, "NS", ip, args.user, args.passwd)
    ssh_d = SSHDriver(t, "SSH Driver")
    t.activate(ssh_d)

    try:
        ssh_d.run_check("ls")
    except Exception as e:
        logging.exception(e)

    t.cleanup()

while True:
    t1 = threading.Thread(target=query_params, args=("10.0.0.152", ))
    t1.start()

    t2 = threading.Thread(target=query_params, args=("10.0.0.153", ))
    t2.start()

    t1.join()
    t2.join()
```

The exception I get is:
```
ERROR:root:
Traceback (most recent call last):
  File "/home/oip/Documents/eoptris2_testing/testing/utils/lg_test.py", line 24, in query_params
    ssh_d.run_check("ls")
  File "/home/oip/Documents/eoptris2_testing/venv/lib/python3.10/site-packages/labgrid/binding.py", line 96, in wrapper
    return func(self, *_args, **_kwargs)
  File "/home/oip/Documents/eoptris2_testing/venv/lib/python3.10/site-packages/labgrid/step.py", line 222, in wrapper
    step.stop()
  File "/home/oip/Documents/eoptris2_testing/venv/lib/python3.10/site-packages/labgrid/step.py", line 187, in stop
    steps.pop(self)
  File "/home/oip/Documents/eoptris2_testing/venv/lib/python3.10/site-packages/labgrid/step.py", line 28, in pop
    assert self._stack[-1] is step
```

Not sure if it is only related to the SSHDriver or not. When only using 1 thread, there are no issues.

Bastian-Krause, 2022-09-01:
This is untested and not the recommended approach. #835 allows running a process on a target asynchronously, so this is what you want to use.

S4mw1s3, 2022-09-02:
Ok. Thanks for the suggestion :)

closed 2022-09-02 by S4mw1s3