labgrid-project/labgrid #1015: Openocddriver multiple interface configs

snerzkristall, 2022-10-03:
I am currently using a custom driver because the default openocddriver cannot handle multiple interface configs.
Would you merge a PR implementing this functionality?

This would change the openocddriver as well as the environment configuration (from single str to a list of strings).

The only problem is time, I would need this to be merged until thursday.
My current solution works without any problems, but I would like to add this functionality as I am sure someone else will need this in the future as well.

Bastian-Krause, 2022-10-09:
Could you explain what's the use case for multiple interface configs?

snerzkristall, 2022-10-10:
Olimex debug adapters use miltiple interface configs, because openocd provides them this way.  
e.g. I need both olimex-arm-usb-ocd-h.cfg and olimex-arm-jtag-swd.cfg for my project.  
The second one is basically a one-liner but if I would like to use the openocd files for future updates and not copy the commands into my own config file.  
I could add a search path for the interface directory on the target machin running openocd and then put the inteface config in the list of my own configs but thats just a bad solution...

Bastian-Krause, 2022-10-10:
Okay, if you're still interested in opening a PR for that, we'd be interested in merging this.

snerzkristall, 2022-10-10:
Okay thank you, will do.

closed 2022-10-10 by snerzkristall