labgrid-project/labgrid #1037: ImportError: cannot import name 'soft_unicode' from 'markupsafe'

kjeldflarup, 2022-11-14:
I'm trying to install the released 0.4.1 coordinator. This fails with this error, which is not the case with master.  
Is it time to have a new release out, or at least a buildable bugfix.

Full error
```
Step 10/12 : RUN crossbar --help
 ---> Running in d71e8469d560
Traceback (most recent call last):
  File "/usr/local/bin/crossbar", line 8, in <module>
    sys.exit(run())
  File "/usr/local/lib/python3.10/dist-packages/crossbar/__init__.py", line 175, in run
    _personalities = personalities()
  File "/usr/local/lib/python3.10/dist-packages/crossbar/__init__.py", line 213, in personalities
    from crossbar import personality as standalone
  File "/usr/local/lib/python3.10/dist-packages/crossbar/personality.py", line 41, in <module>
    from crossbar.worker.router import RouterController
  File "/usr/local/lib/python3.10/dist-packages/crossbar/worker/router.py", line 30, in <module>
    from crossbar.worker.transport import TransportController
  File "/usr/local/lib/python3.10/dist-packages/crossbar/worker/transport.py", line 35, in <module>
    from crossbar.worker.controller import WorkerController
  File "/usr/local/lib/python3.10/dist-packages/crossbar/worker/controller.py", line 34, in <module>
    import jinja2
  File "/usr/local/lib/python3.10/dist-packages/jinja2/__init__.py", line 12, in <module>
    from .environment import Environment
  File "/usr/local/lib/python3.10/dist-packages/jinja2/environment.py", line 25, in <module>
    from .defaults import BLOCK_END_STRING
  File "/usr/local/lib/python3.10/dist-packages/jinja2/defaults.py", line 3, in <module>
    from .filters import FILTERS as DEFAULT_FILTERS  # noqa: F401
  File "/usr/local/lib/python3.10/dist-packages/jinja2/filters.py", line 13, in <module>
    from markupsafe import soft_unicode
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/usr/local/lib/python3.10/dist-packages/markupsafe/__init__.py)

```

Bastian-Krause, 2022-11-14:
Agreed. We should either limit markupsafe < 2.1.0 (see their [CHANGES.rst](https://github.com/pallets/markupsafe/blob/12dae101d29425725ff99755c62259bb4ece5319/CHANGES.rst#version-210)) for a 0.4.2 and/or make a new release from master (containing jinja2 3.0.2 which is not affected by that issue).

Since the majority of dependencies was pinned more than a year ago on master, I'd like to bump them one more time before we create a release from master, though. We had #891 prior to the switch to pyproject.toml in #1001. I'll try to create a more recent version within the next days.

Bastian-Krause, 2023-04-26:
Now that we have new releases, this one is done.