labgrid-project/labgrid #1056: DeprecationWarning: There is no current event loop

kjeldflarup, 2023-01-04:
I get an annoying warning at the end of my pytest session, when running on the latest master

```
test_setup/test_setup.py::test_setup_ip
  /usr/local/lib/python3.10/dist-packages/labgrid/remote/client.py:1238: DeprecationWarning: There is no current event loop
    loop = asyncio.get_event_loop()

test_setup/test_setup.py::test_setup_ip
  /usr/local/lib/python3.10/dist-packages/autobahn/asyncio/websocket.py:300: DeprecationWarning: There is no current event loop
    self.loop = loop or asyncio.get_event_loop()

test_setup/test_setup.py::test_setup_ip
  /usr/local/lib/python3.10/dist-packages/autobahn/asyncio/wamp.py:238: DeprecationWarning: There is no current event loop
    loop = asyncio.get_event_loop()
```
I tried to update autobahn to latest version, but the error stil exists. Anybody knows why this warning comes.

Bastian-Krause, 2023-01-04:
What exact Python version are you using?

See the deprecation notices in https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.get_event_loop. Python 3.10.0–3.10.8 emitted these warnings by mistake in some cases.

The event loop has to be created and set explicitly now before `asyncio.get_event_loop()` is called.. So to get rid of all these warnings, we need to make sure that this is the case. I guess for the library and pytest plugin use cases, this is the job of autobahn. For the client use case, we probably need to do this ourselves, i.e. [here](https://github.com/labgrid-project/labgrid/actions/runs/3828861134/jobs/6514914058#step:10:82).

ozan956, 2026-10-13:
As I see this has fixed by @Bastian-Krause already.

See: https://github.com/labgrid-project/labgrid/commit/0e1e4fba

I verified with a test run against current master, no "There is no current event loop" warning is raised anymore.

Closing as resolved. Please comment if you still see this and let us know about the reproducing steps. Thanks!

closed 2026-10-13 by ozan956