labgrid-project/labgrid #1068: Support for a build server

sjg20, 2023-01-18:
I currently use tbot for doing a git bisect with U-Boot, with a script like this:

git bisect run /vid/software/devel/ubtest/do-try.sh ${board} HEAD

This works because tbot includes a build feature. So it can build U-Boot for each commit. Then it runs a smoke test.

Similarly I have a script which builds and boots my current commit (along with any local changes).

I am interested in moving over to use labgrid, but I wonder how this might be implemented?

At a minimum I suppose I need to know the U-Boot board name (e.g. 'rpi_3'). Is it possible to add that to the metadata for the place and access it from an external tool?

Any other ideas?

sjg20, 2023-01-18:
BTW here a fragment is what I currently have in the yaml, for use with my home-grown 'labman' tool:

  hubc: &hubc
    kernel-phys-port: 1-5
    num-ports: 16
    ports:
      port1: &hubc_1
        port-number: 2.1
      port2: &hubc_2
        port-number: 2.2
      port3: &hubc_3
        port-number: 2.3
      port4: &hubc_4
        port-number: 2.4

uarts:
  usbport1: &ttyusb_port1
    type: usb-uart
    symlink: ttyusb_port1
    hub-port: *hubc_4

duts:
  rpi3:
    desc: Raspberry Pi 3b Fedora 34
    console: *ttyusb_port1
    bootdev: *sdwire0
    power: *ykush0_port1
    flash:
      method: rpi
    network:
      method: ethernet
      mac-address: b8:27:eb:b4:f9:f2
    prompt: U-Boot>
    username: pi
    password: raspberry
    builds:
      32-bit:
        default: yes
        target: rpi_3_32b
        toolchain: armv7-a
      64-bit:
        target: rpi_3
        toolchain: aarch64

jluebbe, 2023-01-18:
> I currently use tbot for doing a git bisect with U-Boot, with a script like this:
>
> git bisect run /vid/software/devel/ubtest/do-try.sh ${board} HEAD
>
> This works because tbot includes a build [...]

Labgrid intentionally doesn't include build functionality, as usually the SW under test already comes with a build system and we want to test the artifacts as built by the "real" build system. Also, this fits CI pipelines using different stages for build and (parallel) testing well. Finally, we didn't want to maintain another build system ;).

> At a minimum I suppose I need to know the U-Boot board name (e.g. 'rpi_3'). Is it possible to add that to the metadata for the place and access it from an external tool?
> Any other ideas?

In labgrid, details like the prompts are configured in the (client-side) [environment YAML file](https://labgrid.readthedocs.io/en/latest/configuration.html#environment-configuration). In several similar cases, we've simply named these files after the names the build system uses, thereby handling the mapping from build system platform names to labgrid places (and configuration). For example, running a labgrid-pytest test-suite would then use `pytest --lg-env ${board}.yaml -v`, perhaps from a small wrapper script that performs the build first.

It is certainly be possible to add additional keys to the environment YAML file (under the global or per-target [`options:` key](https://labgrid.readthedocs.io/en/latest/configuration.html#environment-configuration)), which are not used by labgrid itself, but available to user code. Still, I'm skeptical of mixing the different concerns of building and testing this way.

Bastian-Krause, 2023-07-23:
Closing this, labgrid is intended for automation and testing, but can certainly be combined with external build tools/scripts.

closed 2023-07-23 by Bastian-Krause