labgrid-project/labgrid #1069: Can I statically configure places?

sjg20, 2023-01-18:
I see that places are stored in the crossbar thing and added using labgrid-client. Can I put all of this in a yaml file instead? Then I can keep it in source control. I have about 34 DUTs in my lab, with about 125 total bits attached, so I'd like to see about migrating the items programmatically.

           Name        Type  Description
===============  ==========  ==============================
      at91-usb0     at91usb  usbdev-sn9260, at91usb
            bbb         dut  TI Beaglebone Black
            bob         dut  Chromebook Bob
            bpi         dut  Banana Pi
             c4         dut  Hardkernel Odroid-C4
          coral         dut  Chromebook Coral
           dli1         dli  192.168.4.19, 8 ports
           dli2         dli  192.168.4.27, 8 ports
         edison         dut  Intel Edison
        em100-0       em100  DP139140
        em100-1       em100  DP022781
        em100-2       em100  DP025143
        em100-3       em100  DP022783
        em100-4       em100  DP139156
        em100-5       em100  DP138817
        em100-6       em100  DP033694
        em100-7       em100  DP137244
         ff3399         dut  Firefly RK3399
           huba         hub  4-10, 16 ports
           hubb         hub  6-2, 16 ports
           hubc         hub  1-5, 16 ports
           hubd         hub  1-4, 16 ports
           hube         hub  1-3, 16 ports
           hubf         hub  1-2, 16 ports
        imxusb0      imxusb  usbdev-snappermx6
      intelusb0    intelusb  usbdev-edison, dfu
          jerry         dut  Jerry
          kevin         dut  Chromebook Kevin
           lcdk         dut  OMAPL138_LCDK
           link         dut  Chromebook Pixel
      minnowmax         dut  Minnowboard Max
       nyan-big         dut  Asus Chromebook
     olimex-a20         dut  olimex A20-OLinuXino-MICRO
         opi_pc         dut  Orange Pi PC
        opi_pc2         dut  Orange Pi PC 2
       pcduino3         dut  Linksprite pcDuino 3
  pcduino3_nano         dut  Linksprite pcDuino3 Nano
         pine64         dut  Pine A64 Plus 2GB
    portserver1        uart  192.168.4.23
           puma         dut  Theobroma Systems Puma RK3399
          rock2         dut  Rock 2
   rockchipusb0  rockchipusb  usbdev-kevin
           rpi0         dut  Raspberry Pi 0 RaspiOS Bullseye Lite
           rpi2         dut  Raspberry Pi 2
           rpi3         dut  Raspberry Pi 3b Fedora 34
          rpi3z         dut  Raspberry Pi 3b Fedora 31
           rpi4         dut  Raspberry Pi 4 Ubuntu
         salmon         dut  Snapper MX6
    samsungusb0  samsungusb  usbdev-snow
          samus         dut  Chromebook Pixel 2
      samus_tpl         dut  Chromebook Pixel 2 (TPL)
        sdwire0      sdwire  sdwire-18
        sdwire1      sdwire  sdwire-7
       sdwire10      sdwire  sdwireda3
       sdwire11      sdwire  sdwireda4
       sdwire12      sdwire  sdwireda5
       sdwire13      sdwire  sdwireda6
       sdwire14      sdwire  sdwireda7
       sdwire15      sdwire  da61
       sdwire16      sdwire  da62
       sdwire17      sdwire  da63
       sdwire18      sdwire  da64
       sdwire19      sdwire  da65
        sdwire2      sdwire  202001064004
       sdwire20      sdwire  da66
        sdwire3      sdwire  202001064001
        sdwire4      sdwire  202001064005
        sdwire5      sdwire  sdwireda10
        sdwire6      sdwire  202001064003
        sdwire7      sdwire  202001064002
        sdwire8      sdwire  sdwireda1
        sdwire9      sdwire  sdwireda2
         sn9260         dut  Snapper 9260
           snow         dut  Chromebook Snow
     sunxi-usb0    sunxiusb  usbdev-opi-pc
     sunxi-usb1    sunxiusb  usbdev-pcduino3
     sunxi-usb2    sunxiusb  usbdev-olimex-a20
     sunxi-usb3    sunxiusb  usbdev-opi_pc2
     sunxi-usb4    sunxiusb  usbdev-pinea64
          svbob       servo  911416-00262, port 9904
        svcoral       servo  911416-00840, port 9905
        svjerry       servo  730422-00045, port 9903
        svkevin       servo  911416-00627, port 9907
         svlink       servo  905537-00223, port 9902
     svnyan-big       servo  911416-00900, port 9901
        svsamus       servo  686203-00047, port 9900
         svsnow       servo  911416-00558, port 9906
     tegra-usb0    tegrarcm  usbdev-jetson-tk1
     tegra-usb1    tegrarcm  usbdev-nyan-big
            tk1         dut  Jetson TK1
       usbport1        uart  ttyusb_port1
      usbport10        uart  ttyusb_port10
      usbport12        uart  ttyusb_port12
      usbport13        uart  ttyusb_port13
      usbport14        uart  ttyusb_port14
      usbport15        uart  ttyusb_port15
      usbport16        uart  ttyusb_port16
      usbport17        uart  ttyusb_port17
      usbport18        uart  ttyusb_port18
      usbport19        uart  ttyusb_port19
       usbport2        uart  ttyusb_port2
      usbport20        uart  ttyusb_port20
      usbport21        uart  ttyusb_port21
      usbport22        uart  ttyusb_port22
      usbport23        uart  ttyusb_port23
      usbport24        uart  ttyusb_port24
      usbport25        uart  ttyusb_port25
       usbport3        uart  ttyusb_port3
       usbport4        uart  ttyusb_port4
       usbport5        uart  ttyusb_port5
       usbport6        uart  ttyusb_port6
       usbport7        uart  ttyusb_port7
       usbport8        uart  ttyusb_port8
       usbport9        uart  ttyusb_port9
      usbrelay0    usbrelay  6QMBS, 8 ports
      usbrelay1    usbrelay  6AAMZ, 4 ports
      usbrelay2    usbrelay  7QMBS, 8 ports
      usbrelay3    usbrelay  8QMBS, 8 ports
            xu3         dut  Odroid XU-3
         ykush0       ykush  (internal port error)
         ykush1       ykush  (internal port error)
         ykush2       ykush  (internal port error)
      zynq_zybo         dut  Zynq Zybo

Emantor, 2023-01-18:
There is a contrib script to synchronize places with a YAML file, see commit https://github.com/labgrid-project/labgrid/commit/417d4a071ef035f59e26af4c7b892691bd5dc4cd and https://github.com/labgrid-project/labgrid/blob/master/contrib/sync-places.py.

Emantor, 2023-01-18:
Why are your usbports and sdwires separate places?

Emantor, 2023-02-07:
Hey @sjg20, are there still open questions here?

Bastian-Krause, 2023-07-23:
Should be covered by `sync-places.py`, see above.

closed 2023-07-23 by Bastian-Krause

sjg20, 2024-04-11:
Just to confirm that this handles what I need here.

Still, it would be better if labgrid allowed this to be set up on startup using an cmdline option, e.g. in labgrid-exporter